wmbusmeters/wmbusmeters #271: Very nice real-world S/N decoding for IZAR water meter

pswiatki, 2021-03-10:
Not an issue, but materials can be used to update the Wiki page, or another place which could be helpful to users.
Getting telegrams and seeing in the resulting JSON:

{"media":"water","meter":"izar","name":"wasser/P15_Main","id":"12345678","prefix":"**C19SB**","serial_number":"**053746**","total_m3":251.887,"last_month_total_m3":249.493,"last_month_measure_date":"2021-02-28","remaining_battery_life_y":13.5,"current_alarms":"no_alarm","previous_alarms":"no_alarm","transmit_period_s":8,"manufacture_year":"2019","timestamp":"2021-03-10T16:15:13Z","device":"rtlwmbus[0000616]","rssi_dbm":11}

Perfect match:

[image]

Thank you much for this decent werk.

weetmuts, 2021-04-09:
Thanks I have added your material to the wiki!
https://weetmuts.github.io/wmbusmeterswiki/IZAR.html

closed 2021-04-09 by weetmuts

don-vip, 2021-04-18:
Glad to see it works for other people!

VPMateev18, 2023-08-03:
can someone help me