wmbusmeters/wmbusmeters #272: support for rtl_433 -M level option

lolorc, 2021-03-15:
Hi,

rtl-433 -M level:
  "Use "level" to add Modulation, Frequency, RSSI, SNR, and Noise meta data."

Could it be possible for wmbusmeters to make use of these fields ? (at least rssi)

TIA

lolorc, 2021-04-14:
there's something quite strange about RSSI levels reported by rtl_433 : my rtlsdr is 1m away from the counter, and it reports RSSI=-0.143 SNR=28.022 NOISE=-28.16

weetmuts, 2021-09-09:
I believe the RSSI:s are hard to calculate properly for rtlsdr dongles since the physical characteristics of your particular dongle is unknown.

 @merbanan We do store the rssi in the json, when we have an rssi, but what should we do with the snr and noise?

merbanan, 2021-09-09:
All those parameters are only valid when the hardware has a fixed configuration. By default the gain stages are partly dynamic which makes it hard to look at one value at the time.

@weetmuts you could add the other values also to the json output.

It's a hard issue that I have not spent to much digging into.

merbanan, 2021-09-09:
If you use gqrx or something like that you can see the dynamic range changing during silence and when a signal is present. This mode of operation increases the dynamic range but as the rtl-sdr doesn't inform what gain is used you can not get the true physical signal strength level. You can only calculate it based on the digital samples.

pswiatki, 2021-09-09:
I wonder how it could be done in practice... I believe you have to have some reference in order to be able to calculate the RF level based on a given set of raw samples taken from the device. Is that true?

pswiatki, 2021-09-09:
That reference level could be the "momentary" (i.e. at a given point in time) value of AGC setting. But then you would have to further calibrate for all the potential losses between the RF-IN connector and the RF input of the chip itself. 
Forthermore, I wonder very much how the SNR is calculated for transmissions in 433.92 or 868 bands [MHz]. I imagine there could be several transmissions at exactly the same time and sometimes sources could have similar RF level at the antenna input of the receiver. That would mean low SNR for the source we are interested in, but we don't know which one it is unless we decode such transmission. Maybe it is the sample buffer that is put aside and when the decoding reveals it was possible to get a given source ID from those samples - only then one could retroactively calculate the SNR based on such sample set. Is it done like that?

KrzysztofHajdamowicz, 2023-02-05:
I am interested in those values.
CC1101 reports RSSI in -dBm, rtl_wmbus reports RSSI in +dBm, but rtl_433 returns fixed 999dBm.
This makes for me harder to position antenna properly.

I don't know how much usable signal strength is left until stick can't decode data, while I try to hide antenna trying to gain Wife Acceptance Factor up.

merbanan, 2023-02-05:
The rtl_sdr has a dynamic AGC-stage. RSSI values can only be trusted if you use a fixed gain setting. And rtl_433 do provide statistics on what it can but AFAIK it is not properly added to the wmbusmeters input.

The intended use of the rtl_433 input is if someone wants to monitor wmbus signals and other signals with only one rtl_sdr device. This generic use has its limits. For the best reception rtl_wmbus should be used.

KrzysztofHajdamowicz, 2023-02-06:
In my case, rtl_wmbus was not able to pick `amiplus` meter, where rtl_433 does this without a problem.

weetmuts, 2023-02-06:
Interesting! Do you have the rtlsdr antenna very close to the meter?

KrzysztofHajdamowicz, 2023-02-06:
I can show you video in private communication.  
For public record: Same antenna connected to CC1101+Esp8266 reports -68 to -72dBm

merbanan, 2023-02-06:
> In my case, rtl_wmbus was not able to pick `amiplus` meter, where rtl_433 does this without a problem.

That was unexpected, but glad that there was a second option that could be of use. Try running rtl_wmbus with lower fixed gain. The signal might saturate the receiver.

KrzysztofHajdamowicz, 2023-02-06:
> > In my case, rtl_wmbus was not able to pick `amiplus` meter, where rtl_433 does this without a problem.
> 
> That was unexpected, but glad that there was a second option that could be of use. Try running rtl_wmbus with lower fixed gain. The signal might saturate the receiver.

That's definetely a possibility, I have GSM tower 500m away from my location, bunch of Wi-Fi networks etc.  
Maybe I should run some spectrum analyzing software on this stick to determine if I pick too strong non-interesting signals or wmbus signal is too weak?

merbanan, 2023-02-06:
The 868.8MHz spectra should be clean. As I said, you can try running the rtl-sdr command with a fixed gain to see if that helps.

closed 2023-05-08 by github-actions[bot]