wmbusmeters/wmbusmeters #274: listening on c1 and t1

rattinca, 2021-03-18:
I would like to switch between listening on t1 and c1, because i have meters on both but only one wmbus module. Is there an option for this?
I could write a script and do  oneshot alternately on c1 and t1 but would like to avoid it.
If there is no option, could you point me to where the functionality for  oneshot is implemented, so i can have a go at it? :-)

weetmuts, 2021-03-18:
Yes, I have been thinking about this before! The trickiest part is to come up with a good way to specify how/when it should switch and how this selection should be encoded into conf-file/command line settings. :-)

So, do you want it to switch after a telegram has been received from a certain meter?
Sort of round-robin listening between all defined meters?

Or do you want to switch during certain hours?

Should we add info to each meter on when it is transmitting and how often?
Then auto calculate when the device should be listening on each mode to capture the meters signals.

Or, if you are using an im871a, then there will be a firmware upgrade soon, that enables it
to listen to both c1 and t1. :-)  I am waiting for such a dongle to test it.

weetmuts, 2021-03-18:
Which wmbus module are you using?

rattinca, 2021-03-18:
I think the standard should be, to listen to every configured meter once and then switch. This should be easy using basically the same functionality that oneshot does (with a specified timeout, in case a configured meter is gone), just switching instead of stopping.
Yes, round robin :-)

Yes, info on how often it transmits and some sort of log on when it transmits, would be awesome! I was thinking of listening to each configured meter once and than using some decision system to decide when to switch. If we can listen to every meter on a chanel within 10-15 minutes, it should be fine to just go round robin. If i have a sensor configured, that sends only every few hours, i need to make sure i'm on the right chanel at the right time, so i don't miss it, but i dont want to be stuck on a chanel for a few hours.
so, we are thinking the same thing ;-)

rattinca, 2021-03-18:
i do have a iM871A, so the upgrade would be very welcome. Do you know when it might be comming?

rattinca, 2021-03-18:
How much time does it take to switch from c1 to t1 or back?

weetmuts, 2021-03-18:
I suppose it takes at most a second to switch between modes. 

The firmware 0x14 can listen to c1 and t1 at the same time. I have just added support for this mode, in wmbusmeters. Try emailing them if there is no public download page for the firmware upgrade.

rattinca, 2021-03-19:
then if we dont have bad luck and two sensors are sending at the same time, we should be able to catch everything...

pswiatki, 2021-03-19:
Bear with me, please - as I don't have all the technical details, but I am curious: does that mean the spectra of c1 and t1 do not interfere one with the other?

weetmuts, 2021-03-19:
c1 and t1 uses the same frequency but different modulation techniques. c1 was introduced in combination with frame format b, to reduce the energy consumption. So the builtin battery would last longer.

rattinca, 2021-03-19:
i spent some time thinking about it.

i need every meter to know the time between its messages (according to its current settings).

Then i can do an object thats based on a map, that allways gives you the time for the next switch.
Object would need one message time per meter to initialize.

Then it should be easy to switch at the specified time and get the time for the next switch, right?

If there are two messages to close to each other to switch, i would prioritize the one with more time between messages. But this should not happen, if switching is fast.

rattinca, 2021-03-20:
Or maybe better: objekt will give you the id of the meter you want to get message from before switching to other Chanel. Just need to put the ids in my map.

What to you think? Would you prefer a time to be at other Chanel or an id after witch to switch?

rattinca, 2021-03-20:
In the rare event that two messages are at the same time on different chanels, i will alternate between them.

rattinca, 2021-03-22:
Ok, i have an objekt now.
It needs for each meter we want to listen to: id, Channel, time of message and time between messages.
Then you can do getNextID and it will always give you the id of the message after wich you want to switch.
So you we would we would catch all messages by switching after catching the last message expected on the current Channel.
If two messages are at the same time, it will choose the one that sends less often. In the rare event this should be the same also, it will alternate.

If i give you the code to this object, can you integrate it to do inteligent switching?
Or can you point me to the right spot to try and do it myself?

rattinca, 2021-04-09:
can you tell me how to switch the channel we are listening to?

weetmuts, 2021-04-09:
How about something like this: 
in main.cc in regular_callback add after the line "meter_manager_->pollMeters(bus_manager_);"
add the call: bus_manager_->adjustListeningModes();
This is called every 2 seconds, but the actual adjustment of listening modes (linkmodes+frequency)
should not happen that often.

In BusManager add adjustListeningModes(). Here you have access to the meters through meter_manager_
and you have access to the wmbus devices through bus_devices_. 

You can now calculate a link modes set and then apply device->setLinkModes(new_set);

rattinca, 2021-04-14:
so... setLinkModes() will cause listening to the set linkmodes (if possible)?
and i would want to set it to listen to a linkmodeset containing only c1 and then switch to a linkmode containing only t1?

weetmuts, 2021-06-20:
@rattinca  Do you want to continue work on this problem? The latest im871a firmware can now listen to both c1/t1 at the same time, as can amb8465 and the rtlwmbus software.

rattinca, 2021-06-20:
Im still working on it when I find the time. I have the structure to generate a list that tells me after which message I want to switch to which channel, but I'm having trouble finding how to tell wmbus to switch to a specific channel.
I'm not scaping it yet, since it would work for any combination of channels and more than two.
But it's not very high on my list right now, since my urgend problem was solved by the firmware update...

BIBOLV, 2023-01-01:
Dongles now support both modes at same time - closing issue. Please reopen if you want to continue this tread,

closed 2023-01-01 by BIBOLV