wmbusmeters/wmbusmeters #275: Killall wmbusmeters problem

Arduan77, 2021-03-27:
Hi, I am using wmbusmeters with oneshot option without config files. There is a problem with latest version, with running processes blocking wmbusmeters from starting from the command line.  After reboot there are two wmbusmeters processes which cannot be killed by any command (also sudo killall -9 wmbusmeters), the processes renew with the new PID.  Kill by PID doesn't work either.  Runing the killall command several times quickly will sometimes kill both processes, but not always.  In the old version everything worked fine, there were no problems with blocking processes. Any ideas?

Arduan77, 2021-03-28:
Solved... sudo snap stop wmbusmeters, or better sudo snap disable wmbusmeters

weetmuts, 2021-04-08:
Great the snap is still work in progress. Thanks for reporting, and solving!

closed 2021-04-08 by weetmuts