wmbusmeters/wmbusmeters #276: Support for Diehl

journaldethomas, 2021-04-10:
Hello,

I try to obtain information from my water meter which is a Diehl IZAR RC I G4 and not R4 like we can see usually. IZAR RC i G4 integrates the standardized OMS protocol and I have doubt if I can obtain results with wmbusmeters. But perhaps you could help me and like this add a new supported water meters to the long list :)

Here are the command and results obtained :

`$ wmbusmeters auto:c1
Started auto rtlwmbus[00000001] listening on c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 92604498
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07)
                   ver: 0x7b
                device: rtlwmbus[00000001]
                  rssi: 13 dBm
                driver: unknown!`

and when I listen to this specific meter, even if I do not know the ID Key it seems that unknown driver is a problem :

`$ wmbusmeters auto:c1 MyTapWater auto 92604498 NOKEY
Started auto rtlwmbus[00000001] listening on c1
(meter) MyTapWater: meter detection could not find driver for id: 92604498 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x7b
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
MyTapWater	92604498		2021-04-09 21:06.11`

If you want I can send more information or more logs. Just tell me what I have to send to you :)

Thomas

weetmuts, 2021-04-10:
If it is an OMS compliant meter, then it should be easy to add support for it, try this:
wmbusmeters --debug auto:c1 MyTapWater supercom587 92604498 NOKEY

And paste the log here.

weetmuts, 2021-04-10:
What I want to see is primarily the telegram=|....| log lines but the other debug info will tell us if you need a decryption key as well.

journaldethomas, 2021-04-10:
Thanks a lot for your prompt answer. I have a result now
`wmbusmeters auto:c1 MyTapWater supercom587 92604498 NOKEY
Started auto rtlwmbus[00000001] listening on c1
(meter) MyTapWater: meter detection did not match the selected driver supercom587! correct driver is: unknown!
(meter) Not printing this warning agin for id: 92604498 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x7b
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
MyTapWater	92604498	214.787 m3	2021-04-10 09:22.58`

I have less information that I saw in others case but it should come from the watermeter.
Otherwise, here logs with the debug mode activated, a lot of messages that perhaps could help :

`wmbusmeters --debug auto:c1 MyTapWater supercom587 92604498 NOKEY
(wmbusmeters) version: 1.2.0-21-ga26bac3
(config) number of meters: 1
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(serial) EVENT thread interrupted
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 1443 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(shell) return code 0
(shell) 1445 exited
(serial) EVENT thread interrupted
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus[00000001] listening on c1
(rtlwmbus) using command: rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 1451 fd 9 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) received ascii "T1;0;0;2021-04-10 07:18:04.000;24;18;5190FFFF;0x3effffffffff9051ffff7abbffff05ff0f24ffffffbfffc2ff6462ffffafb522ffffffff50ff89050affffff2890ffbca3ff5a4563ffff40ffffc1ff28ff39<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2021-04-10 07:18:04.000;24;18;5190FFFF;0x3effffffffff9051ffff7abbffff05ff0f24ffffffbfffc2ff6462ffffafb522ffffffff50ff89050affffff2890ffbca3ff5a4563ffff40ffffc1ff28ff39<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;1;1;2021-04-10 07:18:08.000;29;27;92604498;0x1e44a511984460927b077a9f0010059d1c32b86d02067c1048325a9e961b6b<0A>T1;1;1;2021-04-10 07:18:08.000;33;32;92604498;0x1e44a511984460927b077a9f0010059d1c32b86d02067c1048325a9e961b6b<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2021-04-10 07:18:08.000;29;27;92604498;0x1e44a511984460927b077a9f0010059d1c32b86d02067c1048325a9e961b6b<0A>T1;1;1;2021-04-10 07:18:08.000;33;32;92604498;0x1e44a511984460927b077a9f0010059d1c32b86d02067c1048325a9e961b6b<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 1 templates.
[... 303 lines cut ...]
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2021-04-10 07:19:53.000;43;24;FF801762;0x1e44a5ff621780ffff077a1b00100521ffffffa17ee1d5cb5f9e1f59689cc0<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2021-04-10 07:19:54.000;34;27;92604498;0x1e44a511984460927b077a9f0010059d1c32b86d02067c1048325a9e961b6b<0A>T1;1;1;2021-04-10 07:19:54.000;30;31;92604498;0x1e44a511984460927b077a9f0010059d1c32b86d02067c1048325a9e961b6b<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2021-04-10 07:19:54.000;34;27;92604498;0x1e44a511984460927b077a9f0010059d1c32b86d02067c1048325a9e961b6b<0A>T1;1;1;2021-04-10 07:19:54.000;30;31;92604498;0x1e44a511984460927b077a9f0010059d1c32b86d02067c1048325a9e961b6b<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 5 devices
(serialtty) closed /dev/ttyAMA0 (detect amb8465)
(serial) TIMER thread interrupted
(serialtty) closed /dev/ttyAMA0 (detect im871a)
(serialtty) closed /dev/ttyAMA0 (detect rc1180)
(serialtty) closed /dev/ttyAMA0 (detect cul)
(shell) sending SIGTERM to process group 1451
(bgshell) sent sigterm, now waiting for child 1451 to exit.
(serial) EVENT thread interrupted
(serial) event loop stopped!
(bgshell) 1451 terminated due to signal 15
(wmbus) disconnected 00000001 rtlwmbus
(serialcmd) closed /bin/sh pid=1451 fd=9 (rtlwmbus)
(wmbus) deleted rtlwmbus
(serial) closing 5 devices
`

weetmuts, 2021-04-10:
Thanks! It was easy to add this meter. The telegram only contains the total volume and the error codes.
If you run now with auto for the meter type, it will use the new driver dme_07. 
If you run with --debug you will see the bytes decoded as:
(dme_07) 11: 04 dif (32 Bit Integer/Binary Instantaneous value)
(dme_07) 12: 13 vif (Volume l)
(dme_07) 13: * 03470300 total consumption (214.787000 m3)
(dme_07) 17: 02 dif (16 Bit Integer/Binary Instantaneous value)
(dme_07) 18: FD vif (Second extension FD of VIF-codes)
(dme_07) 19: 17 vife (Error flags (binary))
(dme_07) 1a: * 0000 error codes (OK)

However, we do not know how the 16 error code bits map to the potential error conditions:  Leak detection | Mechanical tampering | Meter stopped | Overflow | Underflow | Backflow | Low battery indicator

So for the moment, if the error code bits are all zero, this is reported as OK,
otherwise "ERR 1ab2" or whatever the error code bits are set to.

Pushed to githib. Please test!

don-vip, 2021-04-18:
> However, we do not know how the 16 error code bits map to the potential error conditions: Leak detection | Mechanical tampering | Meter stopped | Overflow | Underflow | Backflow | Low battery [...]

Do you have a Diehl error telegram in your archives? I can help for this.

journaldethomas, 2021-07-08:
Thanks a lot for all your support. Everything works well since several months now !

closed 2021-07-08 by journaldethomas