wmbusmeters/wmbusmeters #277: Support for QWater5.5

mblnk, 2021-04-12:
I have two Qundis QWater5.5 in my appartment and they are configured to send s1 telegrams every 4h or so.
As the driver is identified as unknown, I tried to add support for it.

One telegram I recorded looks like this:

`telegram=|2D4465327663341317077AAA0000000C13044001004C1340620000426C9F2C02BB560000326CFFFF046D180DA924|`

    Even though the meters are definitely Qundis QWater5.5, the meters do not identify with
    manufacturer code QDS but with LSE.

This is what I came up with for parsing its content.
```
    (qwater55) 0f: 0C dif (8 digit BCD Instantaneous value)
    (qwater55) 10: 13 vif (Volume l)
    (qwater55) 11: * 04400100 total consumption (14.004000 m3)
    (qwater55) 15: 4C dif (8 digit BCD Instantaneous value storagenr=1)
    (qwater55) 16: 13 vif (Volume l)
    (qwater55) 17: * 40620000 due date consumption (6.240000 m3)
    (qwater55) 1b: 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
    (qwater55) 1c: 6C vif (Date type G)
    (qwater55) 1d: * 9F2C due date (2020-12-31)
    (qwater55) 1f: 02 dif (16 Bit Integer/Binary Instantaneous value)
    (qwater55) 20: BB vif (Volume flow l/h)
    (qwater55) 21: 56 vife (duration of limit exceed last lower  is 2)
    (qwater55) 22: * 0000 error code (0)
    (qwater55) 24: 32 dif (16 Bit Integer/Binary Value during error state)
    (qwater55) 25: 6C vif (Date type G)
    (qwater55) 26: * FFFF error date (2127-15-31)
    (qwater55) 28: 04 dif (32 Bit Integer/Binary Instantaneous value)
    (qwater55) 29: 6D vif (Date and time type)
    (qwater55) 2a: * 180DA924 device datetime (2021-04-09 13:24)
```

What I am not so sure about is the error code. From the value it reads and from the messages that can appear on the display according to the [datasheet](https://qundis.com/wp-content/uploads/2017/12/DST1_QWA5_EN0_WMEUP_V210_030317.pdf), I think some error code is most likely. 0 then means no error.

weetmuts, 2021-04-12:
Great work! Yes, the manufacturer codes change over time when companies are bought or the company changes their name for any other reason. Its a mess. Also sometimes a telegram has one manufacturer for the transport layer (aka the meter) and another manufacturer for the data link layer (for example a wmbus addon).

In fact, it is sometimes impossible to figure out a suitable short name for the driver that will match all the brands the meter will be sold under. So, I decided a few weeks ago that new drivers should be named after their mfct_media_version numbers. Since the meter 'auto' now works, it does not matter that much that the driver name is foobar.

I will add a mapping from known brand names to the driver name, just for documentation, 
so that "wmbusmeters --listmeters" also will print known brand names that the meter is sold under. 

So, can you name this new driver  lse_07_17 (instead of qwater55)?

In the future, if they change name from lse to xxy then, instead of creating a new driver xxy_07_17, we will
just map xxy,07,17 to the lse_07_17 driver in meter_detection.h.

Yes, the error codes are a problem. The wmbus standard does not give us a hint on what the bits mean.
Lacking more information, printing OK when error_code = 0, is how I hae done it so far.
When error_code <> 0, then print "ERR 6f24" ie ERR + hex digits of the error code.

Thanks!

closed 2021-04-13 by weetmuts

mblnk, 2021-04-13:
Thanks!

I modified the name and the error code representation.

Related to the error codes, I found the following page in the manual:
[image]
Hopefully, these correspond directly to the error codes in the telegram.

Also the device can send at least one other type of S Telegram.

[image]

weetmuts, 2021-04-13:
Thanks! It seems like it can transmit using C linkmode as well?
Which wmbus device are you using to receive the telegrams? im871a,rtl_wmbus etc?

mblnk, 2021-04-13:
Yes, it seems like it! I use the rtl_wmbus with a NESDR SMART v4

I actually confirmed that I do get the S telegrams.
On friday, a water meter readout is scheduled for our building and I might be able to observe the other type of S telegram. 
Maybe they will also send some kind of trigger telegram to the meter.

On another note:
At the moment, I dont listen to the telegrams 24/7, because the sdr device heats up quite a bit and because the wmbusmeter requires substantial cpu utilisation (~40 % one core on my laptop). Is this expected? Does it depend on the number of telegrams received? With a quite bad antenna I pick up around ~5k telegrams per hour

Can you recommend a (hopefully cheap) device (C + S) which does not need that much power to run?

Thanks a lot!

weetmuts, 2021-04-13:
Well, its rtl_wmbus that uses a lot more cpu when receiving both S+C+T telegrams.
You can follow this issue: https://github.com/xaelsouth/rtl-wmbus/issues/22 
Also the rtl_sdr dongle can get warm while it is running since it is send 16 million samples per second over the USB.

wmbusmeters by itself uses very little cpu. A few bytes in, a few bytes out.

I do not know of any other dongle that can listen to S and C at the same time. 
But you can buy two im871a dongles and set one to S and the other to C.
Or you can buy a single dongle and help rattinca with this issue: https://github.com/weetmuts/wmbusmeters/issues/274

PR #278 (2021-04-13): Add stuff for QWater5.5
My try at supporting the Qundis  Q Water 5.5

If finished, should close #277