wmbusmeters/wmbusmeters #279: (cul) warning: the hex string is not proper! Ignoring telegram!

cvjensen, 2021-04-13:
I'm trying to get data from my kamstrup multical 21

I have this CUL device:
https://www.smart-home-komponente.de/nano-cul/nano-cul-868-extra/

Flashed following this guide:
https://www.smart-home-komponente.de/2018/03/09/nanocul-und-jeelink-am-pc-flashen/

Then I have the following conf files:

```
loglevel=normal
device=/dev/ttyUSB0:t1
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
```

```
name=MyTapWater
id=63015340
key=12B9C058EF8E3B35E719F085B1123456
```

In the log file /var/log/wmbusmeters/wmbusmeters.log I just get:
```
(wmbusmeters) logging started 2021-04-13 16:17:42 using _
[2021-04-13_16:17:44] Started config cul on /dev/ttyUSB0 listening on t1
(wmbusmeters) waiting for telegrams
[2021-04-13_16:17:46] (memory) rss 2134016 peak 2.03 MiB
[2021-04-13_16:18:30] (cul) warning: the hex string is not proper! Ignoring telegram!
[2021-04-13_16:19:20] (cul) warning: the hex string is not proper! Ignoring telegram!
[2021-04-13_16:26:20] (cul) warning: the hex string is not proper! Ignoring telegram!
```

What does that mean?

weetmuts, 2021-04-13:
Use loglevel=debug so that we can see the hex strings.

weetmuts, 2021-04-13:
Do you get any readings at all?

cvjensen, 2021-04-14:
No readings 
```
root@raspberrypi:/var/log/wmbusmeters/meter_readings# ls
root@raspberrypi:/var/log/wmbusmeters/meter_readings#
```

Here's the debug - please tell If you want more - this is just a few seconds ...
```

root@raspberrypi:/var/log/wmbusmeters# tail -f wmbusmeters.log
(serial) received binary "374432453638423943304232314136323441304139304144463834334344383631344232303137393341424542303746324632363246464332454537383031424644383134333441393342313036464637424230394336363230464445393739313746353342364230433643343235373839373630323234"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43699201655CD4CFF3BDAFDAA82ABAC9E3F6B9570A1D9CE8FF777D2E68B9C0B21A624A0A90ADF843CD8614B201793ABEB07F2F262FFC2EE7801BFD81434A93B106FF7BB09C6620FDE97917F53B6B0C6C425789760224"
(cul) no eol found yet, partial frame
(serial) received binary "303843463035343136363236450D0A"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43699201655CD4CFF3BDAFDAA82ABAC9E3F6B9570A1D9CE8FF777D2E68B9C0B21A624A0A90ADF843CD8614B201793ABEB07F2F262FFC2EE7801BFD81434A93B106FF7BB09C6620FDE97917F53B6B0C6C42578976022408CF05416626E<0D><0A>"
(cul) checkCULFrame RSSI_RAW=110
(cul) checkCULFrame LQI=49
(cul) bad hex "7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43699201655CD4CFF3BDAFDAA82ABAC9E3F6B9570A1D9CE8FF777D2E68B9C0B21A624A0A90ADF843CD8614B201793ABEB07F2F262FFC2EE7801BFD81434A93B106FF7BB09C6620FDE97917F53B6B0C6C42578976022408CF05416"
[2021-04-14_07:48:48] (cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.
(serial) received binary "62"
(cul) checkCULFrame "b"
(cul) no eol found yet, partial frame
(serial) received binary "3745343432343233373034393131353632303043374230463741463830303730303541384338394442424332443044344346324342353435323646393546463742333242343344304231453344303446414443313839323539433935463430334546394441464437373136374546433931383535423541"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43D0B1E3D04FADC189259C95F403EF9DAFD77167EFC91855B5A"
(cul) no eol found yet, partial frame
(serial) received binary "354646304441364435394342394233383133434144313238443531364542313443433646323443304438423332343332323431393133333334323544344530433242413030454439353532343339343639313434413032464146464436443536343330454638304543353033363032444546303841354534"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43D0B1E3D04FADC189259C95F403EF9DAFD77167EFC91855B5A5FF0DA6D59CB9B3813CAD128D516EB14CC6F24C0D8B3243224191333425D4E0C2BA00ED9552439469144A02FAFFD6D56430EF80EC503602DEF08A5E4"
(cul) no eol found yet, partial frame
(serial) received binary "32343043353844413239433444410D0A"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43D0B1E3D04FADC189259C95F403EF9DAFD77167EFC91855B5A5FF0DA6D59CB9B3813CAD128D516EB14CC6F24C0D8B3243224191333425D4E0C2BA00ED9552439469144A02FAFFD6D56430EF80EC503602DEF08A5E4240C58DA29C4DA<0D><0A>"
(cul) checkCULFrame RSSI_RAW=-38
(cul) checkCULFrame LQI=98
(cul) bad hex "7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43D0B1E3D04FADC189259C95F403EF9DAFD77167EFC91855B5A5FF0DA6D59CB9B3813CAD128D516EB14CC6F24C0D8B3243224191333425D4E0C2BA00ED9552439469144A02FAFFD6D56430EF80EC503602DEF08A5E4240C58DA29"
[2021-04-14_07:49:00] (cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.
(serial) received binary "62374534343234323337303439313135363230304337423046374146383030373030354636323433443437"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005F6243D47"
(cul) no eol found yet, partial frame
(serial) received binary "43333835444338423045363032344532354539454436433837423332374237433742453537434633434539343439384439303339423444463444443233"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005F6243D47C385DC8B0E6024E25E9ED6C87B327B7C7BE57CF3CE94498D9039B4DF4DD23"
(cul) no eol found yet, partial frame
(serial) received binary "443630454333363030344233383039463432444642333141353835413943363635314544384644363442444331333143303533364534364134454636423233413332453844354631454543423633303742424636433930454335463734454133373036414639414236444543304146384346433331443641"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005F6243D47C385DC8B0E6024E25E9ED6C87B327B7C7BE57CF3CE94498D9039B4DF4DD23D60EC36004B3809F42DFB31A585A9C6651ED8FD64BDC131C0536E46A4EF6B23A32E8D5F1EECB6307BBF6C90EC5F74EA3706AF9AB6DEC0AF8CFC31D6A"
(cul) no eol found yet, partial frame
(serial) received binary "3142353235333237343031313030464434413630343634363845313441330D0A"
(cul) checkCULFrame "b7E44242370491156200C7B0F7AF8007005F6243D47C385DC8B0E6024E25E9ED6C87B327B7C7BE57CF3CE94498D9039B4DF4DD23D60EC36004B3809F42DFB31A585A9C6651ED8FD64BDC131C0536E46A4EF6B23A32E8D5F1EECB6307BBF6C90EC5F74EA3706AF9AB6DEC0AF8CFC31D6A1B525327401100FD4A6046468E14A3<0D><0A>"
(cul) checkCULFrame RSSI_RAW=-93
(cul) checkCULFrame LQI=10
(cul) bad hex "7E44242370491156200C7B0F7AF8007005F6243D47C385DC8B0E6024E25E9ED6C87B327B7C7BE57CF3CE94498D9039B4DF4DD23D60EC36004B3809F42DFB31A585A9C6651ED8FD64BDC131C0536E46A4EF6B23A32E8D5F1EECB6307BBF6C90EC5F74EA3706AF9AB6DEC0AF8CFC31D6A1B525327401100FD4A6046468E"
[2021-04-14_07:49:14] (cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.
```

weetmuts, 2021-04-14:
Thanks, your CUL sends this string of bytes over the serial port:
b7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43D0B1E3D04FADC189259C95F403EF9DAFD77167EFC91855B5A5FF0DA6D59CB9B3813CAD128D516EB14CC6F24C0D8B3243224191333425D4E0C2BA00ED9552439469144A02FAFFD6D56430EF80EC503602DEF08A5E4240C58DA29C4DA<0D><0A>
Remove the EOL <0D><0A> and the the prefix b (which indicates a T1 message) you have:
7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43D0B1E3D04FADC189259C95F403EF9DAFD77167EFC91855B5A5FF0DA6D59CB9B3813CAD128D516EB14CC6F24C0D8B3243224191333425D4E0C2BA00ED9552439469144A02FAFFD6D56430EF80EC503602DEF08A5E4240C58DA29C4DA

Now count the number of characters in the string:
echo -n "7E44242370491156200C7B0F7AF8007005A8C89DBBC2D0D4CF2CB54526F95FF7B32B43D0B1E3D04FADC189259C95F403EF9DAFD77167EFC91855B5A5FF0DA6D59CB9B3813CAD128D516EB14CC6F24C0D8B3243224191333425D4E0C2BA00ED9552439469144A02FAFFD6D56430EF80EC503602DEF08A5E4240C58DA29C4DA" | wc -c

And it says 253 characters, but a proper hex string must be a multiple of 2. Two hex = one byte.
So why is it an odd number of hex chars being sent from the CUL. is something wrong with our understanding of the CUL protocol? 

Can you confirm if this happens for all your telegrams? Or if it is spurious for a few telegrams?
Can you then investigate the CUL protocol to see if the wmbus_cul.cc is doing the right thing?

cvjensen, 2021-04-14:
How do you want me to confirm if it happens for all telegrams ? I'ts been running all day and nothing is in the meter readings folder , so I guess all of them is causing an error ?

I'm all new in this CUL world , so I simply do not understand what you want me to do with the wmbus_cul.cc???

weetmuts, 2021-04-14:
You need to load your CUL device with the right firmware to listen to C1 telegrams.
Wmbusmeters correctly probes the cul device and reports that it cannot set the c1 mode.

Unfortunately my experience with the nano-cul is that it cannot receive C1 telegrams unless you are 
very very close to the meter. I recommend that you buy an im871a dongle instead
or try the rtlsdr solution.

weetmuts, 2021-04-14:
> I'm all new in this CUL world , so I simply do not understand what you want me to do with the wmbus_cul.cc???

Well, if the cul is expected to send the bytes as ascii hex, then why does it send an odd number of hex characters?
Either your cul is broken or we have misunderstood the protocol.

I am not an expert on CUL either....but the serial protocol looks pretty straightforward. So if something as weird
as a hex character is missing, then I am suspecting the CUL device.

cvjensen, 2021-04-14:
I can see that others in here successfully extracted data from their multical 21 - do you know which devices they used?
If I should give it a shot with the im871a dongle, which one do you recommend then?

weetmuts, 2021-04-14:
I am using the im871a to listen to my Multical21 on C1 mode. 
https://wireless-solutions.de/products/m-bus/im871a-usb/

cvjensen, 2021-04-14:
Well then it should work for me too if I buy an im871a , right ? . I'll try to buy one , then we can keep in touch .

weetmuts, 2021-04-14:
Yes, it is a good dongle. What is your distance to your meter and how many walls are there in between?

cvjensen, 2021-04-14:
Here's a drawing . 1. is my pi + dongle 2. is the watermeter. Around 6-7meters straight line . The walls are made of bricks (around 10cm thick)

[image]

cvjensen, 2021-04-23:
Hi again , I've now received the dongle you suggested - and it does not give the same errors . However I still don't get any readings - how often can I expect my multical 21 to send telegrams ?

weetmuts, 2021-04-23:
If it is a c1 meter then it transmits once every 16 seconds or so. For t1 meters I do not now.

Run: wmbusmeters auto:c1,t1
if you have the latest version of wmbusmeters and the latest version of the im871a dongle, then it should work.
If not, then try just: wmbusmeters auto:c1

and stand close to the meter, to see if it works...

cvjensen, 2021-04-23:
I dont know how to answer if its a C1 or T1 meter . 

I've now moved my raspberry out in the same room so the usb device is literally 30cm away from the water meter. 

This is what I get now:

```
root@raspberrypi:~# wmbusmeters auto:c1,t1
Started auto im871a[00102807] on /dev/ttyUSB0 listening on c1,t1
Warning! Desired link modes c1,t1 cannot be set for device /dev/ttyUSB0:im871a[00102807]
No meters configured. Printing id:s of all telegrams heard!
^Croot@raspberrypi:~wmbusmeters auto:c1
Started auto im871a[00102807] on /dev/ttyUSB0 listening on c1
No meters configured. Printing id:s of all telegrams heard!
^Croot@raspberrypi:~# wmbusmeters auto:t1
Started auto im871a[00102807] on /dev/ttyUSB0 listening on t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 56114970
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Heat volume at inlet meter (0x0c)
                   ver: 0x20
                device: im871a[00102807]
                  rssi: -49 dBm
                driver: unknown!
Received telegram from: 56114970
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Heat volume at inlet meter (0x0c)
                   ver: 0x20
                device: im871a[00102807]
                  rssi: -49 dBm
                driver: unknown!
```

56114970 is my heating company meter - the one that I'm not allowed to have the key for.
So ... still no water meter ...

weetmuts, 2021-04-23:
Go download the datasheet for the multical21 and see if you can translate the type and configuration numbers.
These might give you info on what/when/how it is sending telegrams.

cvjensen, 2021-04-23:
Found it...

```
It is possible to choose between different protocols (C1, T1),
and various reading intervals, by choosing a specific module.
Each module contains the option of choosing between up
to 10 different data packages. You MUST choose one data
package. 

A Wireless M-Bus data package is transmitted every
16 seconds (‘drive-by’) or 96 seconds (‘fixed network’).
When sending a data package every 16 seconds the
package is kept short and compressed to achieve a
long battery life.
At 96 second intervals, a longer and intelligent radio
package with built-in ‘repair coding‘ is sent – the long
battery life is still guara
```

I just wonder if the radio is disabled .. but then again .. how should my water company then know my water consumption . Cant remember when I last time looked at the meter display .

cvjensen, 2021-04-23:
btw why does it say:
`Warning! Desired link modes c1,t1 cannot be set for device /dev/ttyUSB0:im871a[00102807]
and then nothing comes up on the screen .

but if I only write t1 then all my heating telegrams shows up .

weetmuts, 2021-04-23:
c1 and t1 are different radio modes. Some dongles can listen to both at the same time. This is only necessary if you have a c1 and t1 meter and you want to listen to them both at the same time.

weetmuts, 2021-04-23:
otherwise you pick the linkmode c1,t1 or s1 that your meter uses.

weetmuts, 2021-04-23:
try wmbusmeters auto:c1 and wait for 30 minutes.

cvjensen, 2021-04-23:
> try wmbusmeters auto:c1 and wait for 30 minutes.

Alright I'll give it a try - I've moved the USB so it is almost becoming a part of the water meter . Distance cannot be a problem from now on .

weetmuts, 2021-04-23:
Too close is not good either. Being in the same room is guaranteed to work. :-)

cvjensen, 2021-04-23:
I think I've tested from 7 meters (it could see my heating meter) to 1 cm - nothing . 

Still nothing: 

```
root@raspberrypi:~# wmbusmeters auto:c1
Started auto im871a[00102807] on /dev/ttyUSB0 listening on c1
No meters configured. Printing id:s of all telegrams heard!

```

want me to test another half hour from 1 meters distance ? or do you have any other suggestions . If you ask me I dont think its anything to do with the distance .

weetmuts, 2021-04-23:
Then it is for sure silent on the c1. Try half an hour with t1. 

Also can you post your type and con strings here from the meter face?

cvjensen, 2021-04-23:
I'll try t1 just have so filter out my own and my neighbours heating meter .. else I just get spammed on the T channel :)

you mean a picture of the meter face itself ? physically ?

weetmuts, 2021-04-23:
Sure post or email me a picture of the meter face. The strings on the face
has information.

cvjensen, 2021-04-23:
Just sent you a mail :)

cvjensen, 2021-04-23:
Something strange is happening when I try the C1 command.

I found out that I had a process hanging:

```
root@raspberrypi:~# ps -ef | grep -i wmbus
root      9001  7992 99 14:07 ?        01:20:59 wmbusmeters auto:c1
root     13828  9400  0 15:29 pts/1    00:00:00 grep -i wmbus
```

Tried to kill it and start a C1 scan again but it just hangs:

```
root@raspberrypi:~#  wmbusmeters auto:c1
^C

```

... this is like 10 minutes just hanging there before I tried to Ctrl C .

cvjensen, 2021-04-23:
Just tried to restart my pi - it also happens for the T1 command:

```
root@raspberrypi:~# wmbusmeters auto:t1
^C

^C

```

cvjensen, 2021-04-23:
If I start a process that hangs then the next process will start searching for telegrams but the first will hang forever . 

Can you help me with a command to filter out my water meter on the ID ?
Then I can try the T1 for 30 minutes or so .

weetmuts, 2021-04-23:
First make sure no other wmbusmeters process is running. Kill them with
"sudo killall -9 wmbusmeters" and make sure no daemon is running either:
"sudo systemctl start wmbusmeters" and "sudo systemctl disable wmbusmeters"

Then you can do:
"wmbusmeters auto:t1 | grep 12345678"
Put in your meter id instead of 12345678. Wait. :-)

weetmuts, 2021-04-23:
Then you can try s1 as well.

pswiatki, 2021-04-23:
Fredrik,

`"sudo systemctl start wmbusmeters" and "sudo systemctl disable wmbusmeters"`

did you mean: 

`"sudo systemctl stop wmbusmeters"`

weetmuts, 2021-04-23:
yeah stop and disable.

pswiatki, 2021-04-23:
That's what I thought. Thank you!

weetmuts, 2021-05-22:
Closing this after we tried to resolve the problem. It seems like the multical21 meters is transmitting (or not properly transmitting) in a way that cannot be heard by any of the im871a,rtlwmbus or cul..... but can be heard by the water company.
We will file this under the X-files....

closed 2021-05-22 by weetmuts

zibous, 2021-05-27:
I use the same hardware but the version version: **0.9.34** and all works w/o problems
Installed on a Raspberry Pi 3 Model B.

[image]

No distance problems, the NanoCul not only sees my watermeter, but also everyone in my area 👍 ....

Project see: https://github.com/zibous/ha-watermeter

trsqr, 2021-11-29:
I'll link this issue with https://github.com/weetmuts/wmbusmeters/issues/390 as I had the same problem and managed to solve it.