wmbusmeters/wmbusmeters #280: Support for Qundis Heat Cost Allocator (ver: 0x34)

marioli, 2021-04-14:
Ladies and gentlemen,

great project :)

I would like to add support for another meter type. I read the Howto several times but I was unable to recognise the content of the telegram. It would be great if you can help me.

This is, what I have done (in current docker image):

```
/wmbusmeters # ./wmbusmeters --debug simulation_qds.txt
(wmbusmeters) version: 1.3.0-11-g9a6738e
(config) using device: simulation_qds.txt:any
(config) number of meters: 0
(lookup) with file "simulation_qds.txt"
(lookup) driver: simulation file
(main) opening simulation_qds.txt:any
Started config simulation on simulation_qds.txt listening on any
(simulation) in simulation_qds.txt
(main) regular reset of simulation  on simulation_qds.txt will happen every 82800 seconds
(main) added simulation_qds.txt to files
(wmbus) no alarm (expected activity) for simulation
(main) simulation_qds.txt already configured as simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(simulation) from file "49449344131760903408780dff5f350082c800000f0007b06effff720200009f2c36030000bf235102000007000100020000000000070031003d00650020ff450046002f046d100dad24"
(wmbus) parseDLL @0 74
(telegram) DLL L=49 C=44 (from meter SND_NR) M=4493 (QDS) A=90601713 VER=34 TYPE=08 (Heat Cost Allocator) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 64
(wmbus) parseAFL @10 64
(wmbus) parseTPL @10 64
(dvparser) found new format "0DFF5F046D" with hash 3cd3, remembering!
(telegram) TPL CI=78
Received telegram from: 90601713
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                driver: unknown!
(wmbus) 00: 49 length (73 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 9344 dll-mfct (QDS)
(wmbus) 04: 13176090 dll-id (90601713)
(wmbus) 08: 34 dll-version
(wmbus) 09: 08 dll-type (Heat Cost Allocator)
(wmbus) 0a: 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(wmbus) 0b: 0D dif (variable length Instantaneous value)
(wmbus) 0c: FF vif (Vendor extension)
(wmbus) 0d: 5F vife (duration of limit exceed last upper  is 3)
(wmbus) 0e: 35 varlen=53
(wmbus) 0f: 0082C800000F0007B06EFFFF720200009F2C36030000BF235102000007000100020000000000070031003D00650020FF450046002F
(wmbus) 44: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 45: 6D vif (Date and time type)
(wmbus) 46: 100DAD24
telegram=||49449344131760903408780DFF5F350082C800000F0007B06EFFFF720200009F2C36030000BF235102000007000100020000000000070031003D00650020FF450046002F046D100DAD24|+0
(serial) stopping manager
(serial) waiting for stop
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted simulation
```

Any ideas how to parse the values of the device?

Thank you very much.

weetmuts, 2021-04-15:
The Qundis HCA:s are a bit special, they send two kind of telegrams.
Can you listen for a while to see if you see two types of telegrams?
This particular telegram, looks at first glance, to be a mostly proprietary protocol, which might be hard to decode.

CWempe, 2021-07-02:
I think I have the same device.

```
Received telegram from: 94310412
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 12 dBm
                driver: unknown!
```

Did you get any useful data, @marioli ?

I will try to see if I get any non-proprietary data, like weetmuts said.

also see: https://github.com/weetmuts/wmbusmeters/issues/300#issuecomment-873080845

weetmuts, 2021-07-04:
This is solved now! The correct driver is qcaloric and I just pushed a fix to wmbusmeters to auto detect this.
Thanks @CWempe for providing the necessary telegrams!
Closing this now.

closed 2021-07-04 by weetmuts