wmbusmeters/wmbusmeters #281: Are here users from Germany who got the AES-Key for Diehl Hydrus watermeter? Please post the company name

JsBergbau, 2021-04-16:
Hallo,

are here users from Germany that requested the AES-key to decrypt Diehl Hydrus watermeter messages? If so please post the name of your water company here. I'm trying to get if from my company, but it seems I have to collect some examples from other companies.

Thanks a lot in advance.

drose030, 2021-05-26:
Maybe this helps https://www.loxforum.com/forum/faqs-tutorials-howto-s/190604-howto-diehl-hydrus-wasserz%C3%A4hler-werte-an-loxone-mit-wmbus

JsBergbau, 2021-05-26:
Saidly no. This company has a individual key for every meter. 
I've already tried these keys
```
key=39BC8A10E66D83F839BC8A10E66D83F8
#key=51728910e66d83f851728910e66d83f8
#key=NOKEY
#key=00000000000000000000000000000000
#key=11111111111111111111111111111111
```
Don't know where exactly I had the other keys from.

drose030, 2021-05-26:
Okay, sad to hear
I guess I got lucky then

weetmuts, 2021-06-20:
Well, different water companies, different local municipalities etc have different rules.
Some hand out the keys to you. Some don't.  

A lot of companies/local municipalities in Sweden hand out your key (though not all).
Sometimes you have to pay, sometimes you get it free. 

Closing this for now.

JsBergbau, 2021-06-21:
@weetmuts 
Even with paying for the key, I have no chance. So I need examples from companies in Germany which hand over the key. My water company (a quite small one) is concerned, that if I lose the key they are liable for damages that occur because I lost the key. That is a very strange argumentation, I know. So that's why I search for other water companies which hand over the key. 

Since a lot of users that monitor their water usage use your project, chances are quite high, that someone finds this issue here and luckly he is a user from Germany using Diehl Hydrus with the AES key from his company. So it would be very nice, when you re-open this issue, so users find that I'm looking for a water company from Germany that hands over the key to it's customers. Could you please re-open the issue? Thanks a lot.

weetmuts, 2021-06-22:
Of course. :-)

weltenwort, 2021-11-20:
Unfortunately I can only offer a negative example:

My water supplier is **Wasser Nord** in Brandenburg who refuses to hand out the keys for the Diehl Hydrus type 173 water meter they just installed here. They claim they're using the same key for all water meters in the area and can't grant access for that reason.

larsxschneider, 2022-01-06:
@weltenwort I just got the same reply from Wasser Nord :-/ 

Did you happen to try the keys posted above? (https://github.com/weetmuts/wmbusmeters/issues/281#issuecomment-849103756)

I haven't tried them as I don't have the hardware yet.

weltenwort, 2022-01-06:
Yes, I tried those. It's possible I'm making some mistake, but none of the keys I found out in the public worked so far. I can receive the unencrypted headers fine, but decrypting the payload fails.

larsxschneider, 2022-01-06:
@weltenwort Cool. Would you be willing to post a single example telegram of the payload that you received? I would like to play around with the decryption a bit (but I don't want to raise false hopes here... I am pretty sure you have done everything right with the decryption... I just want to give it a second pair of eyes if you want to).

weltenwort, 2022-01-06:
Absolutely, thank you for investing more time into this. I was considering to set up a harness to brute-force it, but didn't have the opportunity yet. Anyway, this is one of the telegrams I captured:

```
3E44A5112655687276077A39003005EBAEEB906AE817B45D3CC6B46A955BAD34DEA47B00F860ACBB6D280069A227B334CEE23006878125BBAD10EDADAD9635
```

I've been decoding it using

```
docker run -t -i --rm \
    --name=wmbusmeters \
    -v /etc/localtime:/etc/localtime:ro \
    weetmuts/wmbusmeters \
    /wmbusmeters/wmbusmeters \
      --debug \
      3E44A5112655687276077A39003005EBAEEB906AE817B45D3CC6B46A955BAD34DEA47B00F860ACBB6D280069A227B334CEE23006878125BBAD10EDADAD9635 \
      hydrus auto 72685526 \
      39BC8A10E66D83F839BC8A10E66D83F8
```

where the last argument is the key mentioned in one of the earlier comments.

Looks like the headers decode fine, which gives me some confidence that the hardware works. It doesn't seem to be using a counter mode with a rotating key, because the payload only changes when the water measurement changes.

I'd be happy to collaborate further on this if you have an idea for how to proceed.

larsxschneider, 2022-01-06:
I think what you are doing is correct as far as I understand it. I tried to decode the encrypted part manually (in Ruby as example):

```rb
require "openssl"

cipher_text = ["EBAEEB906AE817B45D3CC6B46A955BAD34DEA47B00F860ACBB6D280069A227B334CEE23006878125BBAD10EDADAD9635"].pack("H*")
decipher = OpenSSL::Cipher.new("aes-128-cbc")
decipher.decrypt
decipher.padding = 0
decipher.iv = ["A5112655687276073939393939393939"].pack("H*")
decipher.key = ["51728910E66D83F8"*2].pack("H*")
r = decipher.update(cipher_text) + decipher.final
r.unpack("H*")
```

Result:
```
=> ["ce0712f6fd8727d6973f66d5724358fbca9ff0d165047100ad1deea8e7572bb8ca851da81626aff51c39956b26ba19e5"]
```

---

This decoded telegram is not valid as all valid decodigs have to start with `2f2f`. That means the key is wrong. For the fun of it I tried a little loop to "brute force" the key (🤡 I know that likely doesn't work as there are 16^16 combinations):

```rb
require "openssl"
require 'securerandom'

cipher_text = ["EBAEEB906AE817B45D3CC6B46A955BAD34DEA47B00F860ACBB6D280069A227B334CEE23006878125BBAD10EDADAD9635"].pack("H*")
decipher = OpenSSL::Cipher.new("aes-128-cbc")
decipher.decrypt
decipher.padding = 0
decipher.iv = ["A5112655687276073939393939393939"].pack("H*")

loop do
	key = SecureRandom.hex[0,16]*2
	decipher.key = [key].pack("H*")
	r = decipher.update(cipher_text) + decipher.final

	if r.unpack("H*")[0][0,4] == "2f2f"
		puts "possible key = #{key}"
	end
end
```

Results:
```
possible key = b1dd1a924c74fc2fb1dd1a924c74fc2f
possible key = 0ba2db3e5ab70c880ba2db3e5ab70c88
possible key = 7b5f0f2bc688899b7b5f0f2bc688899b
possible key = 6eba351c795498046eba351c79549804
possible key = a5421871711bd1f0a5421871711bd1f0
...
```

This naive approach checks ~2 million keys per minute on my machine. That means it takes ~17 million years to try all of them (16^16/2,000,000/60/24/365) 😂 

As this is symmetric encryption, the key must be in unencrypted from on the metering device. How do we get it from there?

JsBergbau, 2022-01-06:
I wonder if the key is printed on the package when the meter is installed. I think by using an LED which turns / let the display on the watermeter always on will significantly reduce the battery life. Still it will take some years until the battery is empty, but it speeds things up until a new watermeter is installed.
Also putting aluminium foil around the watermeter could possibly shield the device enough that a technican must come to read the data. Perhaps that is a change to get the key by asking the (uninformed?) service personel which are often very willing to help.

pswiatki, 2022-01-06:
I wonder: what about listening to the RF traffic when the service personnel comes over to read the meters (yeah, remotely, but they are nearby, not that far away). Could that produce any useful results?

JsBergbau, 2022-01-06:
I don't think so. The device sends only data, but is not capable to receive data, at least the device that my water company installed. There are versions available that can receive data as far as I remember. But still traffic should be encrypted and not useful at all.

larsxschneider, 2022-01-06:
@weltenwort the serial number on my device is a 8 digit number... maybe they used that 4 times as key?!

larsxschneider, 2022-01-06:
@zivillian Are you with "Wasser Nord" as well? (just found your other issue https://github.com/weetmuts/wmbusmeters/issues/383 )

larsxschneider, 2022-01-07:
🎉 I think I found something interesting: http://www.momzrt.hu/document/HYDRUS_1.2_Product_Specification_en.pdf

Apparently there are 3 modes to configure the meter. I guess WasserNord uses the second mode `Individual key for the variant`:
[image]

For the second mode only the first 6 digits can be changed:
[image]

If you look at the default keys, you can see that:
```
#define PRIOS_DEFAULT_KEY1 "39BC8A10E66D83F8"
#define PRIOS_DEFAULT_KEY2 "51728910E66D83F8"
                            ^^^^^^
```
https://github.com/weetmuts/wmbusmeters/blob/master/src/manufacturer_specificities.cc#L33-L34

That means we have only 16^6 different combinations!!!

larsxschneider, 2022-01-07:
So here is a quick and dirty script to calculate all possibilities:

```rb
require "openssl"
require 'securerandom'

cipher_text = ["EBAEEB906AE817B45D3CC6B46A955BAD34DEA47B00F860ACBB6D280069A227B334CEE23006878125BBAD10EDADAD9635"].pack("H*")
decipher = OpenSSL::Cipher.new("aes-128-cbc")
decipher.decrypt
decipher.padding = 0
decipher.iv = ["A5112655687276073939393939393939"].pack("H*")

for a in 0..15 do
for b in 0..15 do
for c in 0..15 do
for d in 0..15 do
for e in 0..15 do
for f in 0..15 do

	key = (a.to_s(16) + b.to_s(16) + c.to_s(16) + d.to_s(16) + e.to_s(16) + f.to_s(16) + "10E66D83F8") *2
	decipher.key = [key].pack("H*")
	r = decipher.update(cipher_text) + decipher.final

	if r.unpack("H*")[0][0,4] == "2f2f"
		puts "possible key = #{key}"
	end

end
end
end
end
end
end
```

This creates 264 keys ... one of them is the WasserNord key?! 😊

weltenwort, 2022-01-07:
> I wonder if the key is printed on the package when the meter is installed. I think by using an LED which turns / let the display on the watermeter always on will significantly reduce the battery life. Still it will take some years until the battery is empty, but it speeds things up until a new watermeter is installed. And then we should look more carefully / take pictures of the package.
>
> Also putting aluminium foil around the watermeter could possibly shield the device enough that a technican must come to read the data. Perhaps that is a change to get the key by asking the (uninformed?) service personel which are often very willing to help.

I watched the service person very carefully when they installed it. The meter was in an unmarked brown cardboard box. Asked for the available communication interfaces they stated what they're plumbers and no electricians. They didn't even do a functionality check that I noticed. I assume if they're unable to receive anything during their drive-by read-out they'll just send a letter to request a manual read-out as before.

> @weltenwort the serial number on my device is a 8 digit number... maybe they used that 4 times as key?!

Sorry, that doesn't work for me. When I called the technical support person at Wasser Nord they explicitly use it being the same key for all meters in the area as the reason why they couldn't give it out. That doesn't mean it's true, but it shifts the probability.

weltenwort, 2022-01-07:
> For the second mode only the first 6 digits can be changed:

Nice catch, I read that spec but didn't think that the rest might be in the source code. Giving it a try.

larsxschneider, 2022-01-07:
@weltenwort I believe one of those is the right key. Can you confirm?
```
010e4910E66D83F8010e4910E66D83F8
0368d710E66D83F80368d710E66D83F8
040e2d10E66D83F8040e2d10E66D83F8
076a2010E66D83F8076a2010E66D83F8
0d012f10E66D83F80d012f10E66D83F8
1239d310E66D83F81239d310E66D83F8
37257210E66D83F837257210E66D83F8
3d624c10E66D83F83d624c10E66D83F8
421fcc10E66D83F8421fcc10E66D83F8
5464cf10E66D83F85464cf10E66D83F8
59824f10E66D83F859824f10E66D83F8
61d5e210E66D83F861d5e210E66D83F8
62ec0010E66D83F862ec0010E66D83F8
70661c10E66D83F870661c10E66D83F8
763a3410E66D83F8763a3410E66D83F8
80a36110E66D83F880a36110E66D83F8
85edd910E66D83F885edd910E66D83F8
8a5fce10E66D83F88a5fce10E66D83F8
9d52ba10E66D83F89d52ba10E66D83F8
a7429910E66D83F8a7429910E66D83F8
b8b47410E66D83F8b8b47410E66D83F8
bf213610E66D83F8bf213610E66D83F8
c32fcd10E66D83F8c32fcd10E66D83F8
d412d710E66D83F8d412d710E66D83F8
d8619e10E66D83F8d8619e10E66D83F8
dc90be10E66D83F8dc90be10E66D83F8
de263f10E66D83F8de263f10E66D83F8
e592be10E66D83F8e592be10E66D83F8
edad7a10E66D83F8edad7a10E66D83F8
fefb4d10E66D83F8fefb4d10E66D83F8
```

zivillian, 2022-01-07:
@larsxschneider my water supplier is WSE (Wasserverband Strausberg-Erkner) and I was also told they are using one custom key for all their meters.

I tried your script with three different telegrams, but was not able to find a key which works for all of them, but maybe I did something wrong. I've tested the following telegrams:
```
A5C2033E44A5113962487276077A08803005A86940FC8EDEDC61FC258F1CB2DCC38332C9882C6A00F0AF001683AF9184B654AAB607AB01A9E246283EA76E7B21E698537256
A5C2033E44A5113962487276077A078030051F85E52BF9EA9B93B2695D5114024E900469CA24E4AE4297D2C3D15A6FB16F0834A5DFE9677FCED875BDF7E5AD08DB58416312
A5C2033E44A5113962487276077A0780300597635BB6CA696C3D82F4CF9D4631CD7362C8EA72B0DAAF90C38E369F92F094922A4D83C50FD81598DF4FA0C5A32050D9405501
```

drose030, 2022-01-07:
@zivillian I´m with WSE as well 
The device is a Hydrus LNE-14586 (Modelyear is 2016)
For me this one worked: 51728910e66d83f851728910e66d83f8

weltenwort, 2022-01-07:
@larsxschneider I tried your list and as expected they all lead to the encryption "succeeding" (meaning they produce `2f2f` as the first bytes). But none yields sensible data.

Over night I ran a script to brute-force all keys with a variable 6-digit prefix. None yielded success, but my confidence into the "correctness check" of my script is... not too high. :innocent:

zivillian, 2022-01-07:
> For me this one worked: 51728910e66d83f851728910e66d83f8

sadly it's not working for me:
```
(serial) received binary "A5C2033E44A5113962487276077A0D803005269E59BBA69CA173AB842C2E1573267EC2A0A5069EBF32F659C9CE2D2E174D0E1612958057A201FEDFE63DEFF9D8F96354BEEE"
(im871a) checkIM871AFrame "A5C2033E44A5113962487276077A0D803005269E59BBA69CA173AB842C2E1573267EC2A0A5069EBF32F659C9CE2D2E174D0E1612958057A201FEDFE63DEFF9D8F96354BEEE"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 84 (-80 dBm)
(im871a) got crc16 eebe expected eebe
(im871a) received full frame
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) Hauptwasserzaehler: for me? 72486239 in 72486239
[2022-01-07_12:01:56] (meter) Hauptwasserzaehler: meter detection did not match the selected driver hydrus! correct driver is: unknown!
(meter) Not printing this warning again for id: 72486239 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x76
[2022-01-07_12:01:56] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-01-07_12:01:56] (meter) to add support for this unknown mfct,media,version combination
(meter) Hauptwasserzaehler: yes for me
(meter) Hauptwasserzaehler hydrus handling telegram from 72486239
(meter) Hauptwasserzaehler 72486239 "3E44A5113962487276077A0D803005269E59BBA69CA173AB842C2E1573267EC2A0A5069EBF32F659C9CE2D2E174D0E1612958057A201FEDFE63DEFF9D8F963"
(wmbus) parseDLL @0 63
(telegram) DLL L=3e C=44 (from meter SND_NR) M=11a5 (DME) A=72486239 VER=76 TYPE=07 (Water meter) (driver unknown!) DEV=im871a[00102741] RSSI=-80
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(TPL) AES CBC IV decrypting "269E59BBA69CA173AB842C2E1573267EC2A0A5069EBF32F659C9CE2D2E174D0E1612958057A201FEDFE63DEFF9D8F963"
(TPL) num encrypted blocks 3 (48 bytes and remaining unencrypted 0 bytes)
(TPL) IV A5113962487276070D0D0D0D0D0D0D0D
(TPL) decrypted  "179625029BB70E7FE5088831AA1AB610C660F5143E40AD17F1EBE574EFD135E661B3C4F3146644C5373EC4CA9AE6B0F1"
[2022-01-07_12:01:56] (wmbus) WARNING! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 72486239 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x76
(telegram) TPL CI=7a ACC=0d STS=80 CFG=0530 (AES_CBC_IV nb=3 cntn=0 ra=0 hc=0)

```

larsxschneider, 2022-01-07:
@weltenwort Yeah, I tried all these keys last night too and wasn't able to identify a telegram that looked sensible. Would you be willing to share a few more captured telegrams? Plus, would you be willing to share what your total meter volume was be around the time the telegrams where captured?

larsxschneider, 2022-01-07:
@drose030 Can you share a successfully decrypted data package? I would like to know how that looks like 😊

drose030, 2022-01-07:
Probably not what you´re locking for, but in `logs/meter_readings` I find this:
`{"media":"water","meter":"hydrus","name":"Wasser","id":"55524861","total_m3":315.391,"total_tariff1_m3":0,"total_tariff2_m3":0,"max_flow_m3h":0,"flow_temperature_c":127,"external_temperature_c":0,"current_date":"","total_at_date_m3":279.407,"total_tariff1_at_date_m3":0,"total_tariff2_at_date_m3":279.407,"at_date":"2020-12-31 00:00","actuality_duration_s":0,"operating_time_h":0,"remaining_battery_life_y":9.371663,"status":"OK","timestamp":"2021-05-27T10:20:14Z","device":"im871a[00102406]","rssi_dbm":-72}`

I run wmbusmeter as docker in portainer. How do I get what you need?

weltenwort, 2022-01-07:
thanks, that already helps to confirm what a successful decryption looks like :+1:

weltenwort, 2022-01-07:
@larsxschneider I captured

```
3E44A5112655687276077A6ED330055C1B9AB21431F33C04BBE4741DCE827E6849F8407FFCDFB7EFB0262CC350CE8AD13A7B2DE5BE281C5896B6D4E06FDC3A
```

which should correspond to 242.26 m³ and an estimated battery life of 2036-10-27 on the display.

weltenwort, 2022-01-07:
And another one for cross-referencing:

```
3E44A5112655687276077A8DD33005C20A86D323D5AD1A25E3BA4E64427E7A1612693F8CFDAC4B79745E68F41BDC99B936F1A8122449061BB9B1D4BECD9406
```

which should correspond to 243.69 m³.

larsxschneider, 2022-01-07:
Thank you @drose030, that is great!

Do you have a way to get the data telegram *before* encryption? How do you invoke `wmbusmeter` in portainer? Can you start it with `--debug`? That would generate lots of interesting output for each telegram.

larsxschneider, 2022-01-08:
Hmm.. the script 👇  should be able to find the key but it doesn't. So either the assumption is wrong that all keys have the suffix `10E66D83F8` (likely) or we calculate the initialization vector wrong (unlikely).

```rb
require "openssl"

telegrams =	[
	"3E44A5112655687276077A6ED330055C1B9AB21431F33C04BBE4741DCE827E6849F8407FFCDFB7EFB0262CC350CE8AD13A7B2DE5BE281C5896B6D4E06FDC3A",
	"3E44A5112655687276077A8DD33005C20A86D323D5AD1A25E3BA4E64427E7A1612693F8CFDAC4B79745E68F41BDC99B936F1A8122449061BB9B1D4BECD9406",
]

decipher = OpenSSL::Cipher.new("aes-128-cbc")
decipher.decrypt
decipher.padding = 0

for a in 0..15 do
for b in 0..15 do
for c in 0..15 do
	print "."
for d in 0..15 do
for e in 0..15 do
for f in 0..15 do

	key = (a.to_s(16) + b.to_s(16) + c.to_s(16) + d.to_s(16) + e.to_s(16) + f.to_s(16) + "10E66D83F8") * 2

	decipher.key = [key].pack("H*")

	plain_texts = telegrams.map do |telegram|
		manufacturer = telegram[4..7]
		address = telegram[8..19]
		access_number = telegram[22..23]
		cipher_text = telegram[30..]

		decipher.iv = [manufacturer+address+access_number*8].pack("H*")
		plain_text = decipher.update([cipher_text].pack("H*")) + decipher.final
		plain_text.unpack("H*")[0]
	end

	if plain_texts.all? { |i| i[0,4] == "2f2f" }
		puts "\n#{key} (#{plain_texts})"
	end
end
end
end
end
end
end
```

weltenwort, 2022-01-08:
Having searched the 6-digit-prefix key space with the wmbusmeters code as an oracle I tend to agree. Maybe the Hydrus 2 meters are keyed differently - either with a different static suffix or completely unpredictable keys. :thinking:

larsxschneider, 2022-01-08:
@traxanos In [(basierend auf Deinem user namen vermutlich Deinem?!) Post](https://community.symcon.de/t/wasserzaehler-mit-oms-wireless-mbus/125310) schreibst Du, dass Du den AES Schlüssel von Deinem Wasserversorger bekommen hast. Unser Wasserversorger hier im Thread macht das leider nicht. Könntest Du uns folgende Fragen beantworten?

1. Hat Dein Schlüssel das gleiche Format wie die Default Schlüssel? Also 2x the 16 byte z.B.: 
```
39BC8A10E66D83F839BC8A10E66D83F8
11111111111111112222222222222222
```

2. Ist Dein Schlüssel induviduell für Dich oder kannst Du die Werte Deiner Nachbarn damit lesen? Siehe [hier](https://github.com/weetmuts/wmbusmeters/issues/281#issuecomment-1007035774).

3. Wenn Dein Schlüssel ein allgemeiner Schlüssel Deines Wasserversorgers ist, wärst Du bereit uns die letzten 10 Zeichen von Deinem Schlüssel zu geben?

---

I am writing in german because the referenced post was german too. Feel free to switch back to english to ensure everyone in this existing thread can follow.

traxanos, 2022-01-08:
hi, the key has same length. and no i'm not possible to read the counter of my neighbour. it was my first try :)
if i remember correctly, the individual key is mandatory from a certain protocol version.

drose030, 2022-01-09:
@larsxschneider 
I switched the logleve to debug, attached is one telegram with all the output
[hydrus.log](https://github.com/weetmuts/wmbusmeters/files/7835048/hydrus.log)

larsxschneider, 2022-01-10:
@drose030 Thank you! I can confirm that the decryption works with the method we used above:

```rb
require "openssl"

# This is the captured telegram
telegram = "4E44A5116148525570077A2C00400557EE146F0597FFE0724EF417594EDC17"

decipher = OpenSSL::Cipher.new("aes-128-cbc")
decipher.decrypt
decipher.padding = 0

manufacturer = telegram[4..7]
address = telegram[8..19]
access_number = telegram[22..23]
cipher_text = telegram[30..]

# This is the default key
decipher.key = ["51728910e66d83f8" * 2].pack("H*")
decipher.iv = [manufacturer+address+access_number*8].pack("H*")
plain_text = decipher.update([cipher_text].pack("H*")) + decipher.final

puts (plain_text.unpack("H*")[0])
```

Looking at the telegrams I noticed some things:
```
@drose      4E 44 A511 61485255 70 07 7A 2C 00 4005 57EE146F0597FFE0724EF417594EDC17
@weltenwort 3E 44 A511 26556872 76 07 7A 6E D3 3005 5C1B9AB21431F33C04BBE4741DCE827E
@weltenwort 3E 44 A511 26556872 76 07 7A 8D D3 3005 C20A86D323D5AD1A25E3BA4E64427E7A
@zivillian  3E 44 A511 39624872 76 07 7A 08 80 3005 A86940FC8EDEDC61FC258F1CB2DCC383
@zivillian  3E 44 A511 39624872 76 07 7A 07 80 3005 1F85E52BF9EA9B93B2695D5114024E90
            ^^                  
        length                  ^^
                     generation ID          ^^
                                  module state ^^^^
                                         encryption
 
```

@drose has a telegram length of 78 bytes. Everyone else just 62 bytes.

@drose has a device generation ID of 70. Everyone else 76. Higher device generation ID might mean newer version with different encryption keys?

@drose has a module state of `00`.  @weltenwort has `D3` and @zivillian `80`. I don't know what those mean but my gut feeling is that `00` means "no error" and the others might indicate an error code?

@drose has an encryption mode of `4005`. Everyone else has `3005`. I don't know what these values mean but the encryption parameter might be different.

JsBergbau, 2022-01-10:
@larsxschneider 
Perhaps this screenshot helps you, too. 

[image]

Sorry have the data only as screenshot.

weltenwort, 2022-01-10:
> [...] has a device generation ID of 70. Everyone else 76. Higher device generation ID might mean newer version with different encryption keys?

Yes, although the markings on the device (Hydrus 173) don't make it very clear, the 0x76 seems to be what is sometimes marketed as the Hydrus 2. So we might be looking at a different payload structure. If we manage to decrypt one I have no doubt @weetmuts would be able to help us add support.

@larsxschneider regarding the encryption scheme, I might be wrong, but the `30 05` and `40 05` would be consistent with encryption mode 5 with 3 encrypted block and mode 5 with 4 encrypted blocks (little endian):

[image]

Which would be partly good news, because it's still CBC mode with a known IV. Just need the key... if we assume the key is still a duplicated 64 bit key, could we consider brute-forcing to be feasible? CPUs in the cloud are pretty cheap, after all. :shrug:

larsxschneider, 2022-01-11:
> could we consider brute-forcing to be feasible? CPUs in the cloud are pretty cheap, after all.

I am afraid that is still not feasible. I think there are 16^16 different possibilities. My laptop can do ~`12,317,749` AES 128 CBC calculations per second (you can check yours by running `openssl speed -elapsed aes-128-cbc`). That's ~`1,064,253,542,400` per day. That means it would run `17,333,035` days to check everything. On average we should find a result in half the time ... so `8,666,517` days. However, that's still `23,743` years 😢 

If we run the decryption on GPUs we might be able to do it 1000x faster than on my laptop (don't take me too serious here... I haven't measured that... I am just speculating). That means a super fast GPU still calculates 23 years.

If we are lucky we can get a GPU machine on AWS for 30 ct per hour. That's 7.20 EUR per day. 2,600 EUR per year. 63k for 23 years. So on average we would need to spend 63k to crack the key 😄 

Alternatively you could add an additional watermeter that you have full control over. The Hydrus cost ~80 EUR AFAIK plus installtion. Might be the cheaper option 🤣

starkeeper82, 2022-01-16:
Hi there,
I'm currently in the state of evaluating the possibilites with my newly installed water meter. It is a Diehl Hydrus type 173.

Right now I do not own any hardware that would be able to receive the water meter signal. But even without knowing if my water meter has a custom encryption key, I was reading your discussion with huge interest.

Have you ever thought about the problem getting a AES key stored into the device? 
Which way is known to progam the device? 
Is this done by the water supplier or is it done by the manufacturer?

larsxschneider, 2022-01-16:
> Have you ever thought about the problem getting a AES key stored into the device?

That might be a solution. However, I don't own the device and therefore I can't/don't want to open it. Plus, honestly I wouldn't have a clue how to access the device memory/storage anyways 😄 .

> Which way is known to progam the device?

No idea. I think I read somewhere they are programmed by the manufacturer?!

weltenwort, 2022-01-17:
> I think I read somewhere they are programmed by the manufacturer?!

Yes, the order forms I've seen suggests you have to choose your encryption type and keys at order time so the manufacturer can program it. Opening the device would certainly void the official verification you can only do that if you install a meter in addition to the official one.

larsxschneider, 2022-01-20:
@weltenwort Are the order forms accessible on the Internet somewhere? Do they suggest any type of characters or length? Does this mean "WasserNord" defined their own key?

larsxschneider, 2022-01-20:
I wonder what they mean by "double AES 128" ... that's likely nonsense... but they also talk about induvidial keys... nonsense too?! 🤣  [Source](https://github.com/weetmuts/wmbusmeters/files/7908792/gl_kurier0117_internet.1.pdf)

[image]

larsxschneider, 2022-01-21:
@weltenwort I had another thought... let's assume the newspaper article is right and they have indeed individual keys (which would be sane/plausible to me). If someone not too paranoid has setup the system, then they might have used the customer ID / kundenummer as part of the key without salting. My customer ID has 9 digits. That means 7 unknown characters in the key. We can easily check all of them in a reasonable amount of time. It's a long shot but if you want to try it, then you can send me your customer ID via email to "<my-github-username>@ gmail.com" run it.

weltenwort, 2022-01-23:
> Are the order forms accessible on the Internet somewhere? Do they suggest any type of characters or length? Does this mean "WasserNord" defined their own key?

Unfortunately I apparently have failed to save the order forms and am unable to find them anywhere. But I don't remember and further information about the key structure being given there.

> I wonder what they mean by "double AES 128"

Might be a reference to CBC-MAC message authentication?

> let's assume the newspaper article is right and they have indeed individual keys

That's possible but then their argument why they can't give out the key would be invalid.

> they might have used the customer ID / kundenummer as part of the key

That sounds highly speculative to me. I don't want to sound too pessimistic, but even if they used individual keys I wouldn't expect them to order the meters tailored to each household and customer id. That would make replacements way too complicated. Instead they would probably order individually-keyed meters in bulk and just keep record of which meter uses which key in their infrastructure management system. But I appreciate your persistence :wink: I'll send you the id via mail.

weltenwort, 2022-01-23:
In case it helps I have dumped some experimental code I wrote (such as a key candidate generator) to test keys into a repo at https://github.com/weltenwort/hydrus-key-search. This is the first rust code I've written, so use at your own risk :wink:

larsxschneider, 2022-01-23:
I tried `200012345` (`12345` being the uniqe part of the customer ID)  as prefix and suffix. I even found a key but it seems not to be valid. Well, it was a fun exercise 😆 . I've added the code here for reference: https://github.com/weltenwort/hydrus-key-search/pull/1/files

BTW: [Here](https://www.youtube.com/watch?v=O4xNJsjtN6E) is a great explanation of AES. I have a hunch that taking the the 64 bit key twice makes the encryption weaker (besides the obviousily smaller number of possible combinations). This duplication must have a negative impact on the clever s-box design mentioned in the video.

kopierschnitte, 2022-01-29:
Great discussion activity here ;-)
I'm also in Wasser Nord's area and almost gave up in getting the key. Sad to hear that they might did their "homework" correctly and indeed use individual keys. At the moment, I'm in contact with a lawyer in order to discuss possibilities to force WN to hand out the keys to their customers. My lawyer is quite confident but maybe someone wants to join a possible lawsuit against Wasser Nord...

larsxschneider, 2022-01-29:
Well, good luck! 😄 

In case it is useful for your lawyer... while reading about the water meter I found a possible reason for my unreasonable high water consumption: 

My water meter is installed wrong. According to the documentation that can lead to wrong readings:
[image]

Maybe it's the same for you?

Source: https://www.diehl.com/cms/files/542418-DE-DE-20201103_Benutzerhandbuch_HYDRUS_2.0_HWZ_DE.pdf

kopierschnitte, 2022-01-29:
Thanks for this information. Unfortunately, they also did this correctly 😕 
Anyway, I wondered why they reacted so extremely embarassed to my gentle request but after finding this thread I understand that there were indeed some more customers asking the same.

What kind of arguments told they to you why handing out the key is such a great problem?

JsBergbau, 2022-01-29:
@kopierschnitte Please keep us informed what your lawer can do. If he has success maybe it is possible to share his letter (personal data redacted) here to help other users with their water company to get the key. If it goes to court the court decision might be even public and can be shown to our water works.

kopierschnitte, 2022-01-29:
@JsBergbau Yes, of course. But first we would try any other possibilites in order to achieve some kind of "general agreement". Therefore I'm gathering information and trying to get supporters.

I've recently got an answer from our local "Umweltbeirat" and he is quite interested to put this topic on an official agenda for the next meeting in order to "enhance" the relationship between the residents and the local water works :-D

JsBergbau, 2022-01-29:
@kopierschnitte: That would be great. Whats the argumentation refusing giving you the key? My company claims they would be liable if I lose the key for any damages that occur because of that fact.

The "umweltbeirat" is also a good idea, because knowing your water usage live helps to safe water. A lot of laundry machines waste a lot of water with liquid detergents that could be easily detected and thus water saved.

PS: Did you know that Diehl also produces really bad weapons which are by a language trick not forbidden by international treaties. Read on if you want to know more https://github.com/JsBergbau/Verschiedenes/issues/4

weltenwort, 2022-01-29:
> What kind of arguments told they to you why handing out the key is such a great problem?

The answer I got last time is that the key is the same for all devices in the area. Therefore giving it out would violate the privacy of everyone else. That contradicts the announcements in the local press where they tried to preempt privacy concerns with the argument that every meter has an individual key.

I've started another inquiry about it last week with the hope that either (a) their attitude or arguments have changed or (b) I get to talk to a different employee this time. I'll report as soon as I receive any response.

> join a possible lawsuit against Wasser Nord

I'd be willing to participate in a legal exchange of arguments in the hope that no court case would become necessary. Within reasonable financial bounds, of course. :wink: 

> My water meter is installed wrong.

It's installed in the incorrect configuration for me too. In addition it is installed in front of the particulate filter so that's another installation requirement that's not met.

Regarding wrong readings, there is an error code (E07) for air bubbles being detected in the pipes. Have you checked that via the display?

kopierschnitte, 2022-01-29:
I am quite sure that I've read somewhere (maybe in the OMS specification or BSI requirements for smart meters) that it's absolutely mandatory to have unique keys for each meter.
But maybe they really use the same key and therefore refuse to hand it out in order to avoid that this mistake will be uncovered ;-)

You won't get any different contact at WN. It seems that there is only one person responsible for this and he will send you an answer quite soon, trust me...

I guess the time is perfect to further enlarge the public pressure as WN has to deal with bad PR at the moment: https://www.maz-online.de/Lokales/Oberhavel/Hohen-Neuendorf/Eichfrist-fuer-Gartenwasserzaehler-laeuft-unbemerkt-ab-Hunderte-erboste-Anfragen-in-Hohen-Neuendorf

JsBergbau, 2022-01-29:
The article https://www.maz-online.de/Lokales/Oberhavel/Hohen-Neuendorf/Eichfrist-fuer-Gartenwasserzaehler-laeuft-unbemerkt-ab-Hunderte-erboste-Anfragen-in-Hohen-Neuendorf is behind a Paywall. Can you tell us why they think people have to pay more because the watermeter for the garden expired without anyone noticing?

[115 more replies]