wmbusmeters/wmbusmeters #282: is it wmbusmeters working with RPi Zero?

mihai-catalin, 2021-04-18:
Hi,
I have a RTL-SDR v3 dongle connected to a RPi Zero as PoC to decode my water meters as well heat cost allocators.
I installed latest versions of rtl_sdr, rtl_433, rtl_wmbus and wmbusmeters.
When using rtl_433 I get below results with a "fairly" CPU load of 40%:
> pi@raspberrypizero:~ $ rtl_433 -f 868.95M
> rtl_433 version 20.11-114-g4881db38 branch master at 202104141213 inputs file rtl_tcp RTL-SDR
> Use -h for usage help and see https://triq.org/ for documentation.
> Trying conf file at "rtl_433.conf"...
> Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
> Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
> Reading conf from "/usr/local/etc/rtl_433/rtl_433.conf".
> 
> New defaults active, use "-Y classic -s 250k" for the old defaults!
> 
> Registered 27 out of 186 device decoding protocols [ 12 16 19-20 32 34 40 69 73-76 78-79 104-105 112-113 119 124 141 170-173 175 ]
> Found Rafael Micro R820T tuner
> Exact sample rate is: 1000000.026491 Hz
> [R82XX] PLL not locked!
> Sample rate set to 1000000 S/s.
> Tuner gain set to Auto.
> Tuned to 868.950MHz.
> Allocating 15 zero-copy buffers
> baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
> time      : 2021-04-18 13:32:52.421541             Protocol  : 104
> model     : Wireless-MBus Mode      : C            Manufacturer: ITW         ID        : 20170990
> Version   : 19           Device Type: 0x08         Device Type String: Heat Cost Allocator             Control   : 0x44
> Data Length: 45          Data      : 2e4497269009172013087a080020a52c4f07f75e4c890dcff61843c9aa7003fc271e87380a5928bff96f02af793ee9
> Integrity : CRC
> Err Volume flow[0]: 0.000 m3/min                   Max Volume of month -5: 2305852329428197.500 mm3    Min Volume flow[24]: 0.000 Ml/s
> Modulation: FSK          Freq1     : 868.9 MHz     Freq2     : 869.0 MHz
> RSSI      : -0.1 dB      SNR       : 10.4 dB       Noise     : -10.6 dB
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
> time      : 2021-04-18 13:32:52.925035             Protocol  : 104
> model     : Wireless-MBus Mode      : C            Manufacturer: ITW         ID        : 20112714
> Version   : 19           Device Type: 0x07         Device Type String: Water Control   : 0x44          Data Length: 57
> Data      : 3a4497a61427112013077a3b0020a555318eb3fe66654df59fdd72f29f8a387e130d633b062983745b960177d44dcd027f862a0e79332651190000
> Integrity : CRC
> Err Volume flow[0]: 27150.697 m3/min               Max OnTime[0]: 19.000 s   Err Temperature extern 24h: 15273440655.530 C
> Unknown   : none
> Modulation: FSK          Freq1     : 868.9 MHz     Freq2     : 869.0 MHz
> RSSI      : -0.1 dB      SNR       : 11.2 dB       Noise     : -11.3 dB

But when trying below commands I get 100% CPU load and no results even if I am waiting for long time.
> wmbusmeters auto:c1
> wmbusmeters rtlwmbus:868.95M

Is it a limitation due to RPi zero or I am doing something wrong?

Thanks

lolorc, 2021-04-18:
I'm running it on a "Raspberry Pi Model B Rev 2", I run it with this command : "wmbusmeters --format=json --t1 'rtl433:CMD(/home/pi/.local/bin/rtl_433 -s 1.6e6 -M level -F http -F csv -f 868.91M)' diehl izar '*' NOKEY"
cpu usage is about 54%
but the sample rate is far too high.
I've been able to receive datagrams with a better rtlsdr and a sample rate as low as 250k.

mihai-catalin, 2021-04-18:
Thank you!
I run:
`wmbusmeters --format=json --logtelegrams --c1 'rtl433:CMD(/home/pi/rtl_433/build/src/rtl_433 -s 250K -M level -F http -F csv -f 868.95M)' itron auto '*' NOKEY`

and I got:
```
Received telegram from: 20225749
          manufacturer: (ITW) Unknown (0xa697)
                  type: Water meter (0x07)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||3A4497A64957222013077A340020A5DF39C3FE36A28BB3E6D98631D701CF407208E017B4D22640D5F0C8284AF93D35027F8A2A0E79832638200000|+12
Received telegram from: 20226089
          manufacturer: (ITW) Unknown (0xa697)
                  type: Water meter (0x07)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||3A4497A68960222013077A120020A5ACDBD7FD5AC5F34131CB9C108F3C2EF0BED008B88726697A7D554F31B1EB70F5027F8A2A0E79892938200000|+15
Received telegram from: 20170993
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||2E4497269309172013087AE60020A57B50102B19756A36EBB05500B550B8D07821C2253E47812953561AD08E4A259A|+16
Received telegram from: 20170985
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||2E4497268509172013087A020020A518F4F6EFD67C7FA79D7B3538729F3A107990D4F936C4D526DDF6FFC1B90D8561|+17
Received telegram from: 20170992
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||2E4497269209172013087AD60020A54D8469CC58C0E70B99B2D55783381E0DFBBBE2B15860FF02D644549508A56905|+25
Received telegram from: 20159785
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||2E4497268597152013087A030020A5281033EB6F386D227F145ED4F068ECDFA5BB91F7985816FEB1D158163924DC1C|+27
Received telegram from: 20160097
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||2E4497269700162013087A320020A5096E27376FAD0B70709366677161DA1D01AA68B9A3B35483374E9D3240ACF948|+27
Received telegram from: 20226789
          manufacturer: (ITW) Unknown (0xa697)
                  type: Water meter (0x07)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||3A4497A68967222013077A180020A5ABB7B3F0ABB7DF87ED6FF6C41565579FF93747C2279147F44204494388FDF9E5027F8A2A0E79662239200000|+27
Received telegram from: 20226050
          manufacturer: (ITW) Unknown (0xa697)
                  type: Water meter (0x07)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||3A4497A65060222013077A350020A54D3F1051BCBCDF519DFD3714667C3F7C040E0E328A291D1F3A2EF2C1C8694726027F8A2A0E79562538200000|+29
Received telegram from: 20225923
          manufacturer: (ITW) Unknown (0xa697)
                  type: Water meter (0x07)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||3A4497A62359222013077A230020A5622FDBD2CE42E210ADA4048D4BA792FCD0870595EC3C8CA3FBBB1B5EEB450317027F8A2A0E79422038200000|+31
Received telegram from: 20171003
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||2E4497260310172013087AEE0020A5E3235E82F5CF3241FFC0B644A8CB973DB8CA6ECFEF17B14B903D59EA987C2A4A|+33
Received telegram from: 20160115
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||2E4497261501162013087A270020A502AECFA2DB36CD4F696DCE66C5B3E1251423423683D538249F2800EF94D8E92E|+33
Received telegram from: 20225748
          manufacturer: (ITW) Unknown (0xa697)
                  type: Water meter (0x07)
                   ver: 0x13
                device: rtl433[cmd_0]
                  rssi: 999 dBm
                driver: unknown!
telegram=||3A4497A64857222013077A820020A52F9131202CC738893A749E64CF86C0629B19A307C0D12A128F435E95FB0F1031027F8A2A0E79782638200000|+33
```
I suppose I should close this issue and open a new one about supporting my meters. Any clue what to do next? I tried what is written in https://github.com/weetmuts/wmbusmeters/blob/master/HowToAddaNewMeter.txt but no progress so far.

lolorc, 2021-04-19:
you might want to keep this one opened with another title, something like "issue wrt using the rtl_433 source" I also had issues with this source and I managed to get it working with this kind of command.

Can't help you with adding a new meter yet, I'm nowhere there yet ! :-)

BIBOLV, 2021-04-19:
Do you have decryption keys for those meters? Can you provide decrypted telegrams?

mihai-catalin, 2021-04-20:
> Do you have decryption keys for those meters? Can you provide decrypted telegrams?

I'm new to this and taking baby steps...
These meters are purchased for entire apartment building and I doubt the company will share with me this decoding. BTW, they are charging us for reading.
Is it any way to obtain this legally in EU (directive/law)?
If not, what i can technically do for decoding?

PS: maybe is not the place, but i will ask... I'm a bit confused about the scope of different used apps: rtl_433, rtl_wmbus, wmbusmeters; why the decoding cannot be done directly by rtl_433?

weetmuts, 2021-04-20:
Your telegrams from the Itron water meter and hca:s are encrypted. As expected, they should be. But without the decryption keys you cannot listen to them, nor can we add support for the meter or even test if an existing driver already works. The key is an 128-bit AES key so there is no work-around for a missing key.
.
However in a good setup, each meter has its own unique decryption key. Thus if your water company has such a setup,
and is willing, then they can give you your key. A lot of water companies/municipalities does this. Mine for example.
But it differs from company to company and as far as I know, there is no legal requirement to make them to handout your key. 

The wmbus protocol is a huge standard. Its quite a lot of work to implement the standardized parts. Then you need to implement the meter specific implementations because even though a lot is standardized usually the main meter value, there are always several bits or values which understanding require meter specific code. Error codes, historical data etc.

rtl_433 is great for listening to multiple protocols at the same time and auto-decoding them.

rtl_wmbus is great for listening to only wmbus (better range than rtl_433 because it focuses on wmbus only.)

wmbusmeters knows how to decode wmbus telegrams and knows the mappings for a bunch of meters. It relies on either rtl_433/rtl_wmbus or a hardware dongle like im871a/amb8465/rc1800/cul to acquire and decode the radio signal. The analog stuff is not part of wmbusmeters, only the protocol part.

mihai-catalin, 2021-04-21:
Thank you for comprehensive explanation. It is clear now. I will try to get the info and comeback.

closed 2021-04-21 by mihai-catalin