wmbusmeters/wmbusmeters #283: Lansen Pulse Counter - Battery LOW / Battery OK Bit Support?

anonymous-one, 2021-04-19:
I am using a lansen pulse counter with wmbusmeters ( Lansen LAN-WMBUS-G2-OOP ).

I see in their data sheet that they are sending out a Battery LOW / Battery OK bit in the datagram.

http://media1.lansensystems.se/2015/06/LAN-WMBUS-G_Datasheet_1_4.pdf  - Page 5 - "MBUS - Data format"

This PDF is not for the EXACT model I am using, but perhaps the packets they are sending out (at least in relation to this bit) are the same?

Here is a copy paste:

Low battery
Low battery is sent using the power low bit in the status-byte
in EN13757-3 APPLAYER LONG TRANSPORT HEADER (0X72)
Battery low 1 (Bit 2)
Battery OK 0 (Bit 2)

Here is a copy paste of my most recent packets, my battery will be in "OK" state as I just swapped it:

`(lansenpu) log "telegram=|1A443330503702000B027AD7000020|2F2F8E40FD3A700800000000|+4091"`

**Any way possible to get this bit into wmbusmeters?**

Currently I am only getting:

`{"media":"electricity","meter":"lansenpu","name":"MeterPower","id":"00023750","counter_a_int":0,"counter_b_int":909,"timestamp":"2021-04-19T12:06:57Z"}`

---

Note, if the packet format from the PDF above does not match the datagram (0x72?) I will hunt around for one specific to LAN-WMBUS-G2-OOP. Lansen likes to put these behind a 'verified account' wall.

merbanan, 2022-02-06:
The status field is as follows:

0x04 Low battery
0x08 Permanent error/sabotage
0x40 Sabotage enclosure

@weetmuts also has this information.

merbanan, 2022-02-06:
And I don't have the knowledge to implement this.

weetmuts, 2022-09-01:
Howdy! It took a while because I have rewritten the whole driver code.
It is now much easier to write new drivers.
New driver is here: https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_lansenpu.cc

The status value you requested is now in the json as: "status":"OK" if the tpl_sts field is 0x00.

0x04 POWER_LOW
0x08 PERMANENT_ERROR
0x40 SABOTAGE_ENCLOSURE

And a combo like 0x44 will be "status":"POWER_LOW SABOTAGE_ENCLOSURE"

Please test!

BIBOLV, 2023-01-01:
Battery info added to lansenpu driver - closing issue.

closed 2023-01-01 by BIBOLV

anonymous-one, 2023-08-11:
better late than never ;)

i can't quite confirm whether the bit is sent / received / parsed correctly, but i can confirm its popping up OK in the json.

regardless, here is a json dump:

```{"media":"electricity","meter":"lansenpu","name":"MeterPower","id":"00023750","status":"OK","b_counter":939086,"timestamp":"2023-08-11T12:48:39Z","device":"rtlwmbus[cmd_0]","rssi_dbm":113}```

once the battery does run out, which could be X years from now... i will do my best to remember to update here if it does in fact set the bit correctly.