wmbusmeters/wmbusmeters #285: AMB8465 USB stick initialization

makuser, 2021-05-12:
Stopping wmbusmeters and starting it again will not initialize the stick most of the time.

```
(wmbusmeters) version: 1.3.0-11-g9a6738e
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyAMA0) sent "FF0A02008077"
(serial /dev/ttyAMA0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyAMA0 (detect amb8465)
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyAMA0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyAMA0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect rc1180)
(serial /dev/ttyAMA0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyAMA0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyAMA0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect cul)
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyAMA0 (detect cul)
(cul) are you there? no
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyS0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyS0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 7 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "C0FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000"
(serial) received binary ""
(serial) received binary "FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF17"
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(amb8465) are you there? no.
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 7 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 7 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyUSB0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 7 (detect cul)
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyUSB0 (detect cul)
(cul) are you there? no
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
(serial) EVENT thread interrupted
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 1439 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>1434 857
<<<
(shell) return code 0
(shell) 1440 exited
(serial) EVENT thread interrupted
Notice! Other wmbusmeters (pid 857) is running and it might hog any wmbus devices.
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
```

ursf, 2021-05-14:
I've been experiencing the exact same behavior lately, using a Würth Travos-I (AMB8465) and a Würth Tarvos-II (AMB8665 - which is recognized as AMB8465). After some time (usually every 24, 48 or 72 hours when reset-time is default=24h) they will only be initialized when plugged out and in again. For testing purposes I reduced the reset-value to 1h (and additionally, when this didn't help even killed the process every hour with a cronjob) - strangely enough, it re-initializes for some time (sometimes only few hours, sometimes up to around 24-36 hours) and then suddenly refuses to work until re-plugging.

weetmuts, 2021-05-22:
This utterly problematic. The whole alarm code in wmbusmeters was introduced to handle the fact that usb sticks can stop working after running for some time. So was the decicion to have wmbusmeters by default reset the stick after 23hours of running (change this with --resetafter=12h for example). Also the code to track the exact serial port settings, since I was worried that I accidentally set some hardware of software flow control that was tripping.

The problem, ie the usb stick stops working, has been seen in im871a, the amb8465 and two other dongles.
But the effects are different!

The im871a simply silently stopped transmitting data. Here is the reason for the AlarmTimeout and the regular reset.
The amb8465 started sending crap over the serial line. There is code to detect this and reset the dongle.
Another dongle WMB14UE-868, could not communicate for more than a few minutes.... I gave up on that one.
Yet another dongle had to be reset every now and then.

Why on earth can this happen....
The wmbus-chip stops working, as in im871a going silent.....needs a reset to restart.
The communication wmubs-chip to usb2serial stops working. Internal error in dongle. amb8465?
The usb2serial chip goes haywire because of something... WMB14UE-868?
The linux driver for the particular usb2serial chip is borked and will crash after a while....WMB14UE-868?
wmbusmeters accidentally triggers some flow control that blocks further reception?

The wmbusmeters source code was really polite towards the dongle, it just sent the init sequence 
and then sat and listened. So if it worked for a week and then the im871a dongle became silent,
it ought not have been wmbusmeters fault?

Now wmbusmeters is much more assertive, since it reset the dongle every 23h.

weetmuts, 2021-06-09:
@shades66  There is a spurious 00 byte first in the response. I have added some code to skip those and pushed to github.
Can you build and test to see if it works better?

weetmuts, 2021-06-20:
@shades66  Have you had time to test the fix? It should help you!

weetmuts, 2021-08-08:
@shades66 Any update?

weetmuts, 2021-08-08:
@makuser @ursf Do you still experience the dongle hangs that need unplugging/replugging to reset the dongles?

weetmuts, 2021-08-09:
@makuser The problem with amb8465 should now hopefully be fixed. The problem was that the dongle detection code was overwhelmed with valid buffered received telegrams that was sent, before the detection code would get the expected detection response. So the detection code gave up.. :-) Now it looks for the expected response much longer. Please test!

@shades66 Likewise for your problem, it should also be fixed.

I have also found and fixed a race condition that caused wmbusmeters to crash when resetting the dongles. It could be provoked quicker by setting resetafter=21 to reset the dongles every 21 seconds.

weetmuts, 2022-09-03:
Have you experienced this problem recently? Maybe we can close this issue now?

Hadatko, 2022-10-21:
Hello i did experience it today as i enabled debug logs today and install dongle yesterday:
```bash
(serialtty) device /dev/ttyUSB0 is already in use and locked.
(iu880b) could not open tty /dev/ttyUSB0 for detection
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB1 not currently used, detect contents...
(serial) check if /dev/ttyUSB1 can be accessed
(serial) tty /dev/ttyUSB1 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,9,86) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB1 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,160,15,59,236,127,248) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB1 fd 9 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyUSB1) sent "FF0A02008077"
(serial) EVENT thread interrupted
(amb8465/3665) sent 6 bytes OK
(amb8465/3665) reading response... 1
(serial) received binary ""
(amb8465/3665) reading response... 2
(serial) received binary ""
(amb8465/3665) reading response... 3
(serial) received binary ""
(amb8465/3665) are you there? no.
(serialtty) closed /dev/ttyUSB1 (detect amb8465/amb3665)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB1 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,7,86) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,96,15,59,236,127,248) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB1 fd 7 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB1) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001601443C0000C614"
(im871a) checkIM871AFrame "A581100833001601443C0000C614"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 14c6 expected 14c6
(im871a) received full frame
(device info bytes) "33001601443C0000"
(im871a/im170a) info: type=im871a mode=collector firmware=16 hci=01 uid=00003c44
(serial /dev/ttyUSB1) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810616FF000300B32547371000010001FF0702320001000000B635"
(im871a) checkIM871AFrame "A5810616FF000300B32547371000010001FF0702320001000000B635"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 35b6 expected 35b6
(im871a) received full frame
(serialtty) closed /dev/ttyUSB1 (detect im871a)
(serial) EVENT thread interrupted
(device config bytes) "FF000300B32547371000010001FF0702320001000000"
(im871a/im170a) config: other link_mode=t1 id=00103747 media=00 version=01 c_field=00 auto_rssi=01
(im871a/im170a) are you there? yes 00103747 im871a firmware: 16
(main) opening t1
Started auto im871a[00103747] on /dev/ttyUSB1 listening on t1
(im871a) on /dev/ttyUSB1
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,9,86) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,48,7,59,236,127,248) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB1 fd 9 (im871a)
(im871a) get config
(serial /dev/ttyUSB1) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810616FF000300B32547371000010001FF0702320001000000B635"
(im871a) checkIM871AFrame "A5810616FF000300B32547371000010001FF0702320001000000B635"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 35b6 expected 35b6
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00103747
(main) regular reset of im871a  on /dev/ttyUSB1 will happen every 82800 seconds
(im871a) get device info
(serial /dev/ttyUSB1) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001601443C0000C614"
(im871a) checkIM871AFrame "A581100833001601443C0000C614"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 14c6 expected 14c6
(im871a) received full frame
(im871a) device info completed
(im871a) device info: type=im871a mode=collector firmware=16 hci=01 uid=00003c44
(im871a) set config to set link mode 03
(serial /dev/ttyUSB1) sent "A5010306000203300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
```

weetmuts, 2022-10-21:
Actually "waiting for stop" is fine. :-)

This is is the expected printout from the main thread which is waiting after wmbusmeters has setup
the events thread (which listens for incoming data on serial ports) and the alarm thread (which checks dongles and alarms regularly).

Does wmbusmeters not receive any telegrams after this? All looks good in the debug log.
Are you listening to t1 meters?

Hadatko, 2022-10-21:
Oh ok, yeah trying to listen t1 meters. But maybe i should change it to something else. Not sure which type it is. My friend said that i should receive something but he wasn't much specific

weetmuts, 2022-10-21:
With the latest im871a firmware you can do:

wmbusmeters im871a:c1,t1

and then wait until you hear something. Meters are sometimes sending every 16 seconds (multical21) on c1,
or every minute (apator162) on t1 or even less. Sometimes only during business hours and they are silent at night and weekends.

What kind of meter do you want to listen to?

Hadatko, 2022-10-21:
Yeah i changed that to 
  device=/dev/ttyUSB1:im871a:c1,t1 
I am trying to listen Apator devices.

makuser, 2023-03-06:
> @makuser The problem with amb8465 should now hopefully be fixed. The problem was that the dongle detection code was overwhelmed with valid buffered received telegrams that was sent, before the [...]

After extensive testing and usage of the software on different machines for the past few months, I have not been running into this bug at all. There was a hung process once, but that was entirely unrelated to this issue.

Very happy with how this turned out in the past few years :)

@Hadatko do you think I should keep this open for the issue you are having, because from what I see your dongle does not stop working after some time, but instead does not receive anything at all, correct? In that case, you might want to open another issue, again with the appropriate bug description and logs etc.

weetmuts, 2023-03-06:
@makuser Really great to hear! I think we can close this old issue.

closed 2023-03-06 by makuser