wmbusmeters/wmbusmeters #286: is my RTL-SDR dying, or something is wrong with wmbusmeters?

pswiatki, 2021-05-13:
Several days ago I noticed a lot of missing meter readings. Checked `/var/log/wmbusmeters/wmbusmeters.log` and found this:

```
(wmbusmeters) logging started 2021-05-12 00:43:56 using 1.3.0-11-g9a6738e
[2021-05-12_00:43:56] [ALARM SpecifiedDeviceNotFound] the device rtlwmbus[0000616] is not working
No wmbus device detected, waiting for a device to be plugged in.
(wmbusmeters) waiting for telegrams
[2021-05-12_00:43:58] (memory) rss 3858432 peak 3.67 MiB
(wmbusmeters) shutting down
(wmbusmeters) logging started 2021-05-12 00:44:11 using 1.3.0-11-g9a6738e
[2021-05-12_00:44:11] [ALARM SpecifiedDeviceNotFound] the device rtlwmbus[0000616] is not working
No wmbus device detected, waiting for a device to be plugged in.
(wmbusmeters) waiting for telegrams
[2021-05-12_00:44:13] (memory) rss 3727360 peak 3.55 MiB
[2021-05-12_00:45:13] [ALARM SpecifiedDeviceNotFound] the device rtlwmbus[0000616] is not working
[2021-05-12_00:45:47] Started config rtlwmbus[0000616] listening on none
(wmbusmeters) started meter 1 (wasser/P15_Main NNNNNNNN izar)
(wmbusmeters) started meter 2 (prund/P15_Main NNNNNN amiplus)
[2021-05-12_21:51:58] Lost 0000616 closing rtlwmbus[0000616]
No wmbus device detected, waiting for a device to be plugged in.
[2021-05-12_21:51:58] [ALARM SpecifiedDeviceNotFound] the device rtlwmbus[0000616] is not working
[2021-05-12_21:52:00] Started config rtlwmbus[0000616] listening on none
[2021-05-12_21:52:08] Lost 0000616 closing rtlwmbus[0000616]
No wmbus device detected, waiting for a device to be plugged in.
[2021-05-12_21:52:10] Started config rtlwmbus[0000616] listening on none
[2021-05-12_21:52:16] Lost 0000616 closing rtlwmbus[0000616]
No wmbus device detected, waiting for a device to be plugged in.
[2021-05-12_21:52:18] Started config rtlwmbus[0000616] listening on none
[2021-05-13_00:44:15] (memory) rss 4538368 peak 4.32 MiB
```

Looks like `wmbusmeters `no longer believes in my RTL-SDR. By the way: it is the "better" kind - so called "official" [RTL-SDR v.3](https://www.rtl-sdr.com/buy-rtl-sdr-dvb-t-dongles/) 
Should I try another (I have several of those sticks), or perhaps this is a known problem?

pswiatki, 2021-05-17:
It happened again yesterday. I did not analyze the readings deeply, but just by looking at the size of the daily file, I think the stick never actually stopped receiving telegrams. So, this is strange. Or, perhaps not - if the recovery is faster than time interval between telegrams (which is 8 seconds in my case). Still.... Why, what is the reason for those warnings?

weetmuts, 2021-05-17:
What is your setting for alarmtimeout= and
alarmexpectedactivity=
?

weetmuts, 2021-05-17:
I have a lot of questions, but I think you will first have to run with loglevel=debug until you have captured the problem.
With this debug log, then we can proceed further.

pswiatki, 2021-05-17:
I have neither `alarmtimeout=` nor `alarmexpectedactivity=` in use.
Surely, I will switch to `loglevel=debug` and let it run until this issue appears again.

weetmuts, 2021-05-17:
Cool, yes let it run with debug. Which version of wmbusmeters, rtl_sdr and rtl_wmbus are you using?
From the brief description it looks like the child-process running rtl_sdr | rtl_wmbus has died.
rtl_wmbus could have crashed, or rtl_sdr. A glitch in the USB connection (mechanical problem) triggered rtl_sdr to stop. 
etc....

pswiatki, 2021-05-19:
`rtl_sdr` (sadly, no option to get version):

```
[pswiatki@sdr-linux rtl-sdr]$ git log
commit 0847e93e0869feab50fd27c7afeb85d78ca04631 (HEAD -> master, origin/master, origin/HEAD)
Author: Eric Wild <ewild@sysmocom.de>
Date:   Sun Nov 1 00:54:06 2020 +0100

    fix windows build
    
    We really should not have pkgconfig as a build requirement on windows.
```
`rtl_wmbus` version:

```
[root@sdr-linux rtl-sdr]# rtl_wmbus -V
rtl_wmbus: d2be82c with local changes
d2be82cfa2180efb19937ad806dce62b5fe846a0 with local changes
[root@sdr-linux rtl-sdr]#

```
`wmbusmeters` version:

```
[pswiatki@sdr-linux rtl-wmbus]$ wmbusmeters -h
wmbusmeters version: 1.3.0-11-g9a6738e

Usage: wmbusmeters {options} <device> ( [meter_name] [meter_driver]{:<modes>} [meter_id] [meter_key] )*

```
I believe I have an idea why communication with the stick breaks down, but I will investigate a bit further and let everybody know here.

weetmuts, 2021-05-22:
Wmbusmeters reports the rtlwmbus "device" as not working when the child process is no longer running.
See bool SerialDeviceCommand::working() and stillRunning in the the same file.
If you catch the problem with the debug log, there should perhaps be an log message wether
the process exited by itself or died because of a signal or some other reason.

weetmuts, 2021-06-20:
Any update on this? @pswiatki

pswiatki, 2021-06-20:
Yes, sorry I have not updated earlier. I have not seen the issue since late May. However, I have been running with `loglevel=debug` all that time. I am yet to check if reverting back to just the important messages in the log changes the behaviour. By the way: who is in control of and where is the criterion used to rotate `wmbusmeters.log`?

pswiatki, 2021-06-26:
Something seems to be wrong with the resetting of the RTL-SDR dongle in combination with the verbosity of the log. When I keep 'loglevel=debug' in '/etc/wmbusmeters.conf' I see no issues and the dongle works perfectly OK for weeks if not months. As soon as I change to 'loglevel=debug' I start seeing inability to communicate with the dongle which ultimately results in no further attempts and 'wmbusmeters' stays idle not doing anything.
What should I provide from my side to help solving this issue? Excertps from journalctl output, perhaps? If so, what would be of the most interest, specifically?

weetmuts, 2021-06-27:
You wrote loglevel=debug twice. :-) Which setting causes problems and which works?

pswiatki, 2021-06-27:
Ouuugh...... copy-paste mishap :-)  One more time: As soon as I change to `loglevel=normal` I start seeing an inability to communicate with the dongle

pswiatki, 2021-06-27:
I also see some strange behaviour in the system log (`journalctl` produces the following output):

```
Jun 21 17:31:47 sdr-aptv kernel: usb 3-2: USB disconnect, device number 6
Jun 21 17:31:47 sdr-aptv kernel: usb 3-2: new high-speed USB device number 7 using xhci_hcd
Jun 21 17:31:47 sdr-aptv kernel: usb 3-2: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
Jun 21 17:31:47 sdr-aptv kernel: usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 21 17:31:47 sdr-aptv kernel: usb 3-2: Product: RTL2838UHIDIR
Jun 21 17:31:47 sdr-aptv kernel: usb 3-2: Manufacturer: Realtek
Jun 21 17:31:47 sdr-aptv kernel: usb 3-2: SerialNumber: 0000616
Jun 21 17:31:47 sdr-aptv mtp-probe[2675]: checking bus 3, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-2"
Jun 21 17:31:47 sdr-aptv mtp-probe[2675]: bus: 3, device: 7 was not an MTP device
Jun 21 17:31:47 sdr-aptv mtp-probe[2688]: checking bus 3, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-2"
Jun 21 17:31:47 sdr-aptv mtp-probe[2688]: bus: 3, device: 7 was not an MTP device
Jun 21 17:31:48 sdr-aptv mtp-probe[2709]: checking bus 3, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-2"
Jun 21 17:31:48 sdr-aptv mtp-probe[2709]: bus: 3, device: 7 was not an MTP device
Jun 21 17:31:48 sdr-aptv mtp-probe[2722]: checking bus 3, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-2"
Jun 21 17:31:48 sdr-aptv mtp-probe[2722]: bus: 3, device: 7 was not an MTP device
Jun 21 17:31:48 sdr-aptv kernel: usb 3-2: USB disconnect, device number 7
Jun 21 17:31:48 sdr-aptv systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter is at 3.
Jun 21 17:31:48 sdr-aptv audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=wmbusmeters comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jun 21 17:31:48 sdr-aptv audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=wmbusmeters comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jun 21 17:31:48 sdr-aptv systemd[1]: Stopped "wmbusmeters service".
Jun 21 17:31:48 sdr-aptv kernel: audit: type=1130 audit(1624289508.285:145): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=wmbusmeters comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jun 21 17:31:48 sdr-aptv kernel: audit: type=1131 audit(1624289508.285:146): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=wmbusmeters comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jun 21 17:31:48 sdr-aptv systemd[1]: Starting "wmbusmeters service"...
Jun 21 17:31:48 sdr-aptv wmbusmetersd[2739]: (wmbusmeters) started /run/wmbusmeters/wmbusmeters.pid
Jun 21 17:31:48 sdr-aptv systemd[1]: Started "wmbusmeters service".
Jun 21 17:31:48 sdr-aptv audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=wmbusmeters comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jun 21 17:31:48 sdr-aptv kernel: audit: type=1130 audit(1624289508.325:147): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=wmbusmeters comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jun 21 17:31:48 sdr-aptv kernel: usb 3-2: new high-speed USB device number 8 using xhci_hcd
```
This "re-discovery" of the RTL-SDR dongle keeps repeating over and over, but also systemd's `wmbusmeters.service` keeps being started-stopped. Sometimes without any log output between these two actions. Question: how does this dongle-reset operation actually happen? I mean that intentional that supposedly is done once per day?
Also: I wonder if that `mtp-probing` on a newly "discovered" (udev?) RTL-SDR dongle may cause a conflict of some sort and cause the USB device to be "re-connected" again.

weetmuts, 2021-06-27:
What happens if you run with --loglevel=nornal and --resetafter=10000h that should postpone the reset to a year in the future.

pswiatki, 2021-06-27:
I shall try and let you know. Do I reason properly this `--resetafter` CLI option might be "coded" as `resetafter=XXXh` in `/etc/wmbusmeters.conf`?

weetmuts, 2021-06-27:
yes

pswiatki, 2021-06-27:
`resetafter=10000h` added to `/etc/wmbusmeters.conf` We will see what happens after a few days.

Meanwhile, I checked the log and found this:
[journal log extracted](https://drive.google.com/file/d/1ds-jzFmlVQcGPbDxoqyT3RTufksAYONJ/view?usp=sharing)
I provide the link because the log is quite long. It was extracted from the system log using this regexp:
`usb 3-2|wmbusmeters\.service|(SERVICE_(STOP|START).*unit=wmbusmeters)`
It seems `systemd` gets fed up with the naughty `wmbusmeters.service` and finally refuses to restart it any more.
```
Jun 27 21:53:12 sdr-aptv systemd[1]: wmbusmeters.service: Start request repeated too quickly.
Jun 27 21:53:12 sdr-aptv systemd[1]: wmbusmeters.service: Failed with result 'start-limit-hit'.
```
But I am not sure which one of the two: USB device disconnect / service stoppage/restart is the cause and which one the effect of this situation. It doesn't seem likely that my dongle just started failing all of a sudden. If you can extract anything from the log, or more info is needed - please, let me know.

weetmuts, 2021-07-05:
So, from wmbusmeters 1.0 the daemon should always be running.The daemon itself probes any serial ports and rtlsdr devices.

The old solution used udev to start wmbusmeters when a dongle was inserted. Unfortunately this could not work, since only one wmbus dongle actually have a valid usb vendor to be detected by udev. All the others just look like generic serial ports.

Check if you have any remaining udev rule, there should be none. Also check that you have done "sudo systrmctl enable wmbusmeters" so that it restarts on reboots. Then systemctl start wmbusmeters.

weetmuts, 2021-07-05:
The serial ports are probed by other software. Is the serial port a modem? Is the serial port an mtp device.

perhaps these probes write concurrently to the serial port and confuses it?

I know that I lock the serial port when it is in use by wmbusmeters. But perhaps it is not locked when wmbusmeters is probing?

weetmuts, 2021-07-05:
Ah, but this issue is about rtlsdr. Wmbusmeters relies on the rtlsdr library to detect these dongles.... hmmmm.

weetmuts, 2021-07-05:
Do you have some TV/Radio software that wants to configure the rtl-sdr dongle?
Since I did not write the rtlsdr code, I do not know if it locks the dongle when rtl_sdr is running....

weetmuts, 2021-07-05:
Can you check if these files exist in your setup:
/etc/systemd/system/wmbusmeters.service 
/etc/udev/rules.d/99-wmbus-usb-serial.rules 

If so, then can you pull/build/install the latest github master.

pswiatki, 2021-07-05:
It is very much possible that I forgot to remove or disable those old **`udev`** rules. I will check and report back.

pswiatki, 2021-07-05:
OK, checked. I only have:

**`/etc/systemd/system/multi-user.target.wants/wmbusmeters.service`**
which is a symlink to **`/usr/lib/systemd/system/wmbusmeters.service`**.
The latter contains:

weetmuts, 2021-07-05:
service and udev looks good. 
Any tv/radio software that might want to take the rtlsdr dongle?

weetmuts, 2021-07-05:
Here is an indication what could be wrong. There is a "broken" udev rule that will attempt mtp-probe on unknown usb devices
as a catch all, and if it fails disconnect the usb port.

https://superuser.com/questions/1206664/disable-mtp-udev-rules-for-specific-device-so-it-can-be-mount-as-a-usb-mass-stor

Which linux distribution and hardware is this?

 See, if you can do the fix in the first (most upvoted) answer.
Or a similar fix.

wmbusmetersd daemon is constantly running even directly after reboot.
Assuming "sudo systemctl enable wmbusmeters.service"
and "sudo systemctl start wmbusmeters.service"

If it exits, and is restarted by systemctl, then something is seriously wrong.

pswiatki, 2021-07-05:
No, not really. I do not have any such software running on that machine that could claim this dongle.
It is Arch Linux which I massage to a great extent and try to maintain a sane configuration. So that I can at least know what's going on in case of issues. I shall check that offending MTP rule you mentioned in your second message.
By the way: `wmbusmeters` service was brought up successfully and has been working OK since 13:42 today. So, it is a further indication there is some bad blood between the USB resetting and the `wmbusmeters.service` being restarted (log entries indicate that systemd does not like too fast restart of a service and may ground it for good if it decides it's time).

pswiatki, 2021-07-05:
I followed [this answer](https://superuser.com/a/1258799). Will see what happens next time the USB is reset.

pswiatki, 2021-07-05:
Interesting: your [previous comment](https://github.com/weetmuts/wmbusmeters/issues/286#issuecomment-874319634) included something I can now only read in an email notification. This is important:

> wmbusmetersd daemon is constantly running even directly after reboot. Assuming `sudo systemctl enable wmbusmeters.service` and `sudo systemctl start wmbusmeters.service`
> 
> If it exits, and is restarted by `systemctl`, then something is seriously wrong.
> 
I understand you may have opted to remove this part. The question is: why? Does it still hold true, or perhaps not necessarily?
By the way:
```
[root@hist ~]# systemctl status wmbusmeters.service 
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/usr/lib/systemd/system/wmbusmeters.service; enabled; vendor preset: disabled)
     Active: active (running) since Mon 2021-07-05 13:42:01 CEST; 10h ago
       Docs: https://github.com/weetmuts/wmbusmeters
             man:wmbusmeters(1)
    Process: 3517 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 3518 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3519 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3520 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3521 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 3522 (wmbusmetersd)
```

pswiatki, 2021-07-05:
If I may ask - how is that RTL SDR dongle restart done? Via some API call to **`librtlsdr`**? I am wondering if such a restart can trigger some **`systemd`** sentinel that would notice it and perform its own actions (say, mark the service as kind of broken and then trying to restart it from scratch)?

weetmuts, 2021-07-06:
It just sends SIGTERM to the sub-shell that runs the pipe: rtl_sdr .... | rtl_wmbus ....
This shell is is the process group owner so it will shut down the rtl_sdr and rtl_wmbus.

pswiatki, 2021-07-06:
Would it be possible that **`systemd`** can sense this situation (that Exec-s belonging to a service got terminated) and then it will actually trigger the complete service restart?

weetmuts, 2021-07-10:
I do not know for sure, but it would seem odd. Daemons clearly have the ability to fork child processes etc.
As long as the process stored in the pid-file is running then systemd should be happy. I think....

pswiatki, 2021-07-10:
I guess, I will have to learn more about systemd and how it handles things like this. Will come back with more information.

weetmuts, 2021-08-08:
@pswiatki Any update?

pswiatki, 2021-08-08:
Yes, **@weetmuts**. Although it was very confusing and not really indicated in the system journal, I believe the cable somehow degraded, or the code that either controls the power on USB, or data integrity (I guess the USB host driver?) got very picky over time. My setup used to work without any problems, but later stopped and these strange problems started popping up. it ultimately turned out the removal of the extension USB cable helped. It was not damaged (externally) or of really poor quality. I made sure I was getting quality extension cables. One additional thought: could it be the cables had deteriorated internally due to not so low current draw by the RTL SDR dongle (the "original", shiny metal one from RTL-SDR.com)
So far the new setup (RF cable extended, dongle directly in the USB port) has worked for a week without any issues. Hopefully it will stay this way for as long as I need (i.e. until I move to the ARM platform with my monitoring).

weetmuts, 2021-08-08:
Ah! Very interesting and good to know! And it makes sense, the rtlsdr dongle stops working and rtl_sdr will self-terminate, which causes wmbusmeters to declare that it lost the dongle. :-)

If you have power problems, then a dongle like im871a might be better, since they use very little power, if your distance to the meter is not too long.

Can we close this issue now?

pswiatki, 2021-08-08:
Yes, we can close this issue. The distance is very short, in fact. Only 1 to 1.5 meters. Should not be a problem at all. I don't understand it. I have several of these cables, will try another one just to check whether it makes any difference. Also, I should perhaps try this one specimen in some test setup (other than `wmbusmeters`) to see if it is really faulty.

weetmuts, 2021-08-09:
Yes, try some plain rtl_sdr tools and rtl_sdr+rtl_wmbus as well.

Documenting this behaviour could be valuable for the https://github.com/weetmuts/wmbusmeterswiki :-)

I will close this one for now. Many thanks for all the testing.

closed 2021-08-09 by weetmuts