wmbusmeters/wmbusmeters #287: UTC vs. local time

pswiatki, 2021-05-15:
I must be doing something wrong: my daily meter readings (files in `/var/log/wmbusmeters/meter_readings`) start at midnight, but here is what I get in the very first timestamp field:

`{"media":"water","meter":"izar","name":"wasser/P15_Main","id":"zzzzzzz","prefix":"xxxxx","serial_number":"xxxxx","total_m3":273.261,"last_month_total_m3":262.433,"last_month_measure_date":"2021-04-30","remaining_battery_life_y":13.5,"current_alarms":"no_alarm","previous_alarms":"no_alarm","transmit_period_s":8,"manufacture_year":"2019","timestamp":"2021-05-14T22:00:00Z","device":"rtlwmbus[0000616]","rssi_dbm":5} 
`

HAH! Looks like the telegram arrived "precisely" at midnight on my (local) clock, but it was still 22:00 UTC.
I traced this back to `datetimeOfUpdateHumanReadable()` function in `src/meters.cc`:

```
string MeterCommonImplementation::datetimeOfUpdateHumanReadable()
{
    char datetime[40];
    memset(datetime, 0, sizeof(datetime));
    strftime(datetime, 20, "%Y-%m-%d %H:%M.%S", localtime(&datetime_of_update_));
    return string(datetime);
}
```
Somehow `localtime()` does not believe in my timezone (but the clock applet in the corner of my desktop does).
Also, `date` command invoked in the shell returns:

```
$ date
Sat 15 May 02:37:53 CEST 2021
```
Sanity check:

```
$ date -u
Sat 15 May 00:37:56 UTC 2021
```
Is there anything else I can do to make `wmbusmeters` apply local timestamps? By the way: I use systemd service to start it up (and it is enabled, so starts up during the system startup).

weetmuts, 2021-05-17:
For historical reasons the json contains timestamp in UTC timezone, but the human readable data and the fields data uses the local timezone. Since the json is read by code, I assumed the code would have no problem in converting a UTC timestamp to the local timestamp.

Are you reading the json with your software?

pswiatki, 2021-05-17:
Thank you so much for the explanation. Not reading programmatically yet. but I will be. Good to know JSON contains UTC timestamps by design.

closed 2021-05-17 by pswiatki