wmbusmeters/wmbusmeters #290: Cutting down rtl_wmbus CPU utilization from wmbusmeters' level

pswiatki, 2021-05-19:
Recently (see [this issue](https://github.com/xaelsouth/rtl-wmbus/issues/22#issuecomment-833471631)) xaelsouth implemented the following commando line switch in `rtl_wmbus`. This was in reaction to Damian Wróbel's (@dwrobel) [high CPU utilization post](https://github.com/xaelsouth/rtl-wmbus/issues/22#issue-856266170) :

```
# rtl_wmbus -h
rtl_wmbus: invalid option -- 'h'
rtl_wmbus: d2be82c with local changes

Usage rtl_wmbus:
```
[...]
```
	-p [T,S] to disable processing T1/C1 or S1 mode.
# 
```
Is it possible to have this -p switch used by `wmbusmeters ` by a clever specifier in `/etc/wmbusmeters.conf`?
If not - is there any way of specifying the actual command executed for `rtl_wmbus` in the daemon (`systemd`) mode of operation?

pswiatki, 2021-05-20:
Nice try, but a no-go:

**/etc/wmbusmeters.conf:**

```
#loglevel=debug
loglevel=normal
device=rtlwmbus[0000616]:(/usr/local/bin/rtl_sdr -f 868.625M -s 1600000 - 2>/dev/null | /usr/bin/rtl_wmbus -p S)
```
The result from `journalctl` was not so good:

```
May 20 20:57:04 wmbusmetersd[964169]: (wmbusmeters) started /run/wmbusmeters/wmbusmeters.pid
May 20 20:57:04 wmbusmetersd[964170]: Not a valid device "rtlwmbus[0000616]:(/usr/local/bin/rtl_sdr -f 868.625M -s 1600000 - 2>/dev/null | /usr/bin/rtl_wmbus -p S)"
May 20 20:57:04 systemd[1]: wmbusmeters.service: New main PID 964170 does not exist or is a zombie.
```
Oh well, I will have to wait for some better idea from the ones in the know.

weetmuts, 2021-05-22:
So, when the new rtl_wmbus introduced the -s option to listen for s1,t1 and c1 at the same time, 
and it introduce the double decode model, it probably increased the cpu usage significantly.

Perhaps it is time to do what I intended to do at that time, but elected to wait and see, 
ie adjust the parameters to rtl_wmbus depending no which link modes you are interested in.
So if you do not want to hear s1+t1+c1, then wmbusmeters can default to the simpler setting
and avoid the increase in cpu traffic. 

We can also add another option to wmbusmeters that affect how rtl_wmbus is started.
The device string parser can accept extra fields, typically it is the ppm=5 field like this:
rtlwmbus[abc123](ppm=5):c1,t1
the ppm is passed to the rtl_sdr command to tune the dongle.

We could perhaps add:
rtlwmbus[abc123](ppm=5,cpu=minimal):c1,t1

To tweak the rtl_wmbusmeters settings for the minimal cpu usage.
cpu=normal is the default which is max. So you start out with normal,
then you try cpu=minimal to see if you still receive the telegrams.

pswiatki, 2021-05-22:
I sense these **abc123** are placeholders. Perhaps you forgot to replace them (at least the first - with **ppm** example), with the actual, intended string, did you not?
By the way: maybe it would be easier to just select the type of telegrams one is interested in (t1+c1 are always together, I understand, but s1 could be optional). Or something like s1 enable/disable switch?

anonymous-one, 2021-05-22:
This works for me:

`device=rtlwmbus:CMD(/usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s -p S -a)`

```
root@VM-01:~# ps aux | grep rtl_wmbus
wmbusme+   632  0.0  0.0   2384   756 ?        S    May14   0:00 /bin/sh -c /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s -p S -a
```

> Nice try, but a no-go:
>
> **/etc/wmbusmeters.conf:**
>
> ```
> #loglevel=debug
> loglevel=normal
> device=rtlwmbus[0000616]:(/usr/local/bin/rtl_sdr -f 868.625M -s 1600000 - 2>/dev/null | /usr/bin/rtl_wmbus -p [...]

pswiatki, 2021-05-22:
Exactly - good catch. `CMD()` is missing. Thank you so much! I knew I had to fudge up on some mundane detail. I shouldn't work on my project so late at night, either. WIll retry with the correct syntax and report back.

By the way: I find  `-s -P s` a bit confusing:
```
Usage rtl_wmbus:
	-a accelerate (use an inaccurate atan version)
	-r 0 to disable run length algorithm
	-t 0 to disable time2 algorithm
	-d 2 set decimation rate to 2 (defaults to 2 if omitted)
	-v show used algorithm in the output
	-V show version
	-s receive S1 and T1/C1 datagrams simultaneously. rtl_sdr _MUST_ be set to 868.625MHz (-f 868.625M)
	-p [T,S] to disable processing T1/C1 or S1 mode.
```
Does it mean `-s` must be used to get all three types and only then we can decide not to process either the **T1/C1** pair, or **S1** type?

pswiatki, 2021-05-22:
Interesting: `-p S` barely changes CPU usage (as seen in `htop`), but `-a` does cut it more or less in half. Specifically, on my i7 quad-core (eight-thread) CPU I had ~26% before changes - now I have ~14%. Maybe not so important on this system, but when I eventually transition to an ARM-based board (RPi, I guess) CPU usage will become important.

pswiatki, 2021-05-24:
Since CMD() works very well in the config file, I guess it can be closed now.

closed 2021-05-24 by pswiatki

xaelsouth, 2021-06-02:
> Does it mean `-s` must be used to get all three types and only then we can decide (with `-p [T,S]`) not to process either the **T1/C1** pair, or **S1** type?

"-s" option just tells rtl_wmbus that the raw IQ signal received from rtl_sdr must be shifted in the frequency domain _before_ any further signal processing: the signal will be shifted from 868.625M to 868.3M and to 868.95M. Shifting freqs says _nothing_ about how to process the data and decode datagrams.

Without the "-s" option rtl_wmbus knows nothing about the frequency at which the signal is received. rtl_sdr may be configured to receive at 868.3M or at 868.95M. For which kind of data we have to search then? So rtl_wmbus tries it's best and assumes that any data (S1, T1/C1) could be included in the raw signal and applies all decoding techniques as always.

With the "-p" option S1 or T1/C1 data processing path can be disabled which reduce CPU utilization greatly.

"-a" options tells rtl_wmbus to use an inaccurate, but faster atan-technique as proposed by alalons. The inaccuracy _could_ slightly increasy CRC error rate, but it's probably not a big trouble if the meter signal is strong enough.

pswiatki, 2021-06-05:
@xaelsouth Thank you so much for the explanation. Now I fully understand how `rtl_wmbus` works. I have a bit of a problem with comparing floating point values (to zero, for example), but let's talk about it in the proper GitHub space.

anonymous-one, 2021-06-05:
just fyi, a real world example...

when i use `-a` i get 1 to 2 missed readings due to crc errors daily on a meter that advertises every 8 seconds during 'work hours' and much less frequently at night (every 10 minutes or so...)...

mind you i am fairly close to the meter as it lives inside my home so the signal is nice and strong.

in my use case i have no trouble using `-a`.

> > Does it mean `-s` must be used to get all three types and only then we can decide (with `-p [T,S]`) not to process either the **T1/C1** pair, or **S1** type?
> 
> "-s" option just tells rtl_wmbus that the raw IQ signal received from rtl_sdr must be shifted in the frequency domain _before_ any further signal processing: the signal will be shifted from 868.625M to 868.3M and to 868.95M. Shifting freqs says _nothing_ about how to process the data and decode datagrams.
> 
> Without the "-s" option rtl_wmbus knows nothing about the frequency at which the signal is received. rtl_sdr may be configured to receive at 868.3M or at 868.95M. For which kind of data we have to search then? So rtl_wmbus tries it's best and assumes that any data (S1, T1/C1) could be included in the raw signal and applies all decoding techniques as always.
> 
> With the "-p" option S1 or T1/C1 data processing path can be disabled which reduce CPU utilization greatly.
> 
> "-a" options tells rtl_wmbus to use an inaccurate, but faster atan-technique as proposed by alalons. The inaccuracy _could_ slightly increase CRC error rate, but it's probably not a big trouble if the meter signal is strong enough.

weetmuts, 2021-06-09:
I have added a paragraph in the README on how to reduce the CPU usage when listening only to T1/C1 meters:

Here is an example command line that reduces the rtl_wmbus CPU usage if you only need T1/C1 telegrams. It disable S1 decoding (-p s) and trades lower cpu usage for reception performance (-a):

rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)