wmbusmeters/wmbusmeters #292: Is this dongle supported

fsutter, 2021-05-28:
Hi,
I bought this dongle from Alibaba: https://www.alibaba.com/product-detail/868MHz-Wireless-M-Bus-modules-s_62137389990.html?spm=a2700.galleryofferlist.normal_offer.d_image.21302d18P47iCh
Do you know if it is supported?

Many thanks in advance and kind regards
fsutter

weetmuts, 2021-05-28:
It is not supported. I have one, but I never got it working. The serial communication stops working within a few minutes of having it configured, if it even gets that far.

fsutter, 2021-05-28:
Thanks.
I thought it was better that my current SDR dongle.
So I lost 30$ (+26€ custom taxes) for nothing :(

weetmuts, 2021-05-28:
Too bad. Otherwise the im871a is a great dongle.

fsutter, 2021-05-28:
Thanks for the advice.
Do you know if it exists with a SMA connector?

fsutter, 2021-05-28:
Do you recommend me a vendor to buy the im871?

jupiter9595, 2021-05-29:
@fsutter - yes there is im871a adapter with SMA connector. It is proprietary design, Send me a private message to my github account. I will provide you with the details.

fsutter, 2021-05-31:
@jupiter9595 I haven't found a way to send a private message to your github account.

jupiter9595, 2021-06-02:
@fsutter - well, yes for some reason it is still unavailable. I've unchecked [Keep my email addresses private] this in account setting > emails section. But yes it did not work. Any suggestions? To avoid this off topic here.

fsutter, 2021-06-02:
Hi,

are you on twitter with the same nickname?

regards,

fsutter

jupiter9595, 2021-06-02:
Ok, you can now see my email in github front overview page. Send me an email and we'll start from there

weetmuts, 2021-06-09:
Closing this issue for now!

closed 2021-06-09 by weetmuts