wmbusmeters/wmbusmeters #293: Raspberry Zero - DIEHL Metering + rtl-sdr + rtl-wmbus + wmbusmeters - high CPU usage

zibous, 2021-06-02:
Thanks for wmbusmeters - works great, only on my Raspberry Zero it needs a lot of CPU (**90.00 - 94.00 %**):

[image]

```bash
# ----------------------------------------
# root@rbzero: nano /etc/wmbusmeters.conf
# ----------------------------------------
loglevel=normal
device=rtlwmbus
logtelegrams=true
donotprobe=/dev/ttyACM2
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name
meterfilestimestamp=hour
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h localhost -p 1883 -t tele/wasser/verbrauch -m "$METER_JSON"
alarmshell=/usr/bin/mosquitto_pub -h localhost -p 1883 -t tele/wasser/alarm -m "$ALARM_TYPE $ALARM_MESSAGE"
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
ignoreduplicates=true

# open a second console and check mqtt messages
root@rbzero: /usr/bin/mosquitto_sub -h localhost -p 1883 -t tele/wasser/#
# Result:
  {
    "media":"water",
    "meter":"izar",
    "name":"watermeter",
    "id":"18444343",
    "prefix":"",
    "serial_number":"000000",
    "total_m3":258.555,
    "last_month_total_m3":258.437,
    "last_month_measure_date":"2021-06-01",
    "remaining_battery_life_y":12,
    "current_alarms":"no_alarm",
    "previous_alarms":"no_alarm",
    "transmit_period_s":8,
    "manufacture_year":"0",
    "timestamp":"2021-06-02T10:16:47Z",
    "device":"rtlwmbus[00000001]",
    "rssi_dbm":101
 }

```
Installation see: https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-with-rtl-sdr.md

Am I doing something wrong with the settings?

What is the possibility to reduce the CPU consumption for the `wmbusmeters` installation?

BIBOLV, 2021-06-04:
Please check [this](https://github.com/weetmuts/wmbusmeters/issues/290)

zibous, 2021-06-05:
@BIBOLV 

Thank you for pointing this out, but I can't find any setting where rtl_wmbus uses less CPU. 
The data acquisition works, regardless of whether I activate it as a service or via the console. 
No matter what settings I make, rtl_wmbus always needs more than 90% CPU:

```log
Usage /usr/bin/rtl_wmbus:
	-a accelerate (use an inaccurate atan version)
	-r 0 to disable run length algorithm
	-t 0 to disable time2 algorithm
	-d 2 set decimation rate to 2 (defaults to 2 if omitted)
	-v show used algorithm in the output
	-V show version
	-s receive S1 and T1/C1 datagrams simultaneously. rtl_sdr _MUST_ be set to 868.625MHz (-f 868.625M)
	-p [T,S] to disable processing T1/C1 or S1 mode.

root      20   0    2372   1376   1292 R  93,9   0,3   1:29.69 /usr/bin/rtl_wmbus -s -p S
root      20   0    2372   1376   1292 R  91,4   0,3   0:50.21 /usr/bin/rtl_wmbus -s -p S -a
root      20   0    2372   1376   1292 R  95,2   0,3   0:29.57 /usr/bin/rtl_wmbus -s

root@rbzero:~# /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s -p S -a

root      20   0    2372   1372   1288 R  92,9   0,3   0:59.78 /usr/bin/rtl_wmbus -s -p S -a
root      20   0   14864   1868   1664 R   5,2   0,4   0:03.65 /usr/bin/rtl_sdr -d 0 -f 868.625+

T1;0;0;2021-06-05 05:09:52.000;55;67;FFFFFFFF;0x19442423ffffffffffffffddffffdfffffffff54ffffffceffff
T1;0;0;2021-06-05 05:09:54.000;111;115;43410778;0x1944a5117807414344ffa21118ffff3e46d2ff05fc8c25a8f08d
T1;0;0;2021-06-05 05:09:55.000;28;30;15FF0778;0x19ffa5ff7807ff155019ff011bff13ff3a606ebb7a255affa1ff
T1;0;0;2021-06-05 05:09:55.000;111;151;43430778;0x1944a51178074343ff18a261ff0013fff3ffd430ffc011ca9fff
T1;0;0;2021-06-05 05:10:01.000;49;55;FFFF07FF;0x1944ffffff07ffff44ffffffffffff2820d372ff71ff5effffff
T1;0;0;2021-06-05 05:10:01.000;56;55;43890785;0x19442423850789434418a2ff0500132820d3729271a85ea83985

root@rbzero:~# /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s -p S -a -v
rla;T1;1;1;2021-06-05 05:13:52.000;60;59;43890785;0x19442423850789434418a261050013286b411d95d734e01f448d
t2a;T1;0;0;2021-06-05 05:13:54.000;35;27;153207FF;0x1944ff11ff073215ff19ffff1b00ffff66ff4817ffff3dffc5a9
t2a;T1;0;0;2021-06-05 05:13:55.000;112;109;43430778;0x19ffa5117807434344ffa2ffffff133caffff29cd88f76a6fbdc
t2a;T1;0;0;2021-06-05 05:13:56.000;105;152;43410778;0x19ffa51178074143ffffa2ffff00133e1a6f3aa97bc342ff9487
rla;T1;1;1;2021-06-05 05:14:01.000;48;60;43890785;0x19442423850789434418a271050013287c6e543ef6e739c45d8f
t2a;T1;0;0;2021-06-05 05:14:04.000;105;123;FFFF0778;0x19ffa5117807ffffffffa2311800133cb892bb37f95caf7de2de
t2a;T1;0;0;2021-06-05 05:14:04.000;96;159;FFFF0778;0x19ffa5117807ffff44ffa2611800133eff1ee05419b728ffbf80
```

xaelsouth, 2021-06-05:
Raspberry Pi Zero has only one core. It's nearly impossible to do data acqisition (rtl_sdr) and data processing (rtl_wmbus) on this limited  hardware.

I see crc errors in the output attached by you. CRC errors could appear if the CPU does not manage to process the data stream in time.

BIBOLV, 2021-06-05:
Maybe two RasPi's can be used, one for acqisition and one for processing!? 😉

xaelsouth, 2021-06-05:
Will not help to solve the problem - you still have to move the data into the CPU somehow. :P

weetmuts, 2021-06-09:
You can try this device:

rtlwmbus:CMD(/usr/bin/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | /usr/bin/rtl_wmbus -p s -a)

But it might still not be enough to have everything work on a pi zero.
Buy an im871a or amb8465 dongle! They are good and take no cpu power at all.

zibous, 2021-06-10:
@weetmuts 
Thanks, is is better.. From 95% to 86%. 

[image]

> Buy an im871a or amb8465 dongle! They are good and take no cpu power at all.

Yes that's right. I use a NanoCul on a Rasperry PI 3b. Works perfectly .. Raspberry Zero is just a fallback hardware in case something happens to the main device.

CPU Nanao CUL sell: https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-nano_cul.md

Unfortunately I cannot use the latest version of the NanoCUL, only version 0.9.34 works. But has been working reliably for a year. 💯 

Thanks.

weetmuts, 2021-06-20:
Thanks @zibous !

I have updated the README with the workaround that lowers the CPU usage.

closed 2021-06-20 by weetmuts