wmbusmeters/wmbusmeters #295: Hello I have cold and hot meters apator 162.

niklas007, 2021-06-05:
All works fine but  sending mqtt to Ha it reads cold and hot message. From log and shows the meater reading that comes to next. How can I separate the two meter readings ?

weetmuts, 2021-06-05:
Do you have two meters? Each meter should have its own id, you should be able to give diffrent mqtt shell calls for each meter in wmbusmeters, or Ha can perhaps separate the messages based on the id.
You can also json_water=hot to the /etc/wmbusmeters.d/meterhot file and then the json will contain "water"="hot"
likewise for the cold meter.

niklas007, 2021-06-05:
Yes two meters. They have its own IDs.
Yeah I can make different shells. Node red separated them but HA can't so it's shows values hot or cold . Depending on witch meter gives the reading first. So does the Jason reading take the readings from the log

weetmuts, 2021-06-09:
Hmmm, well this seems like a question about HA. Unfortunately that is not my specialty. 
Perhaps you can ask at the HA forums?

weetmuts, 2021-06-20:
Closing this for now.

closed 2021-06-20 by weetmuts