wmbusmeters/wmbusmeters #296: (wmbusmeters) shutting down

Horndrup, 2021-06-18:
After a new installation on a RPI the installation of wmbusmeters stops working after a few hours.

Why does this happen?

(shell) waiting for child 6367 to complete.
(shell) /bin/sh: return code 0
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 im871a
(serialtty) closed /dev/ttyUSB0 (im871a)
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbusmeters) shutting down
(wmbus) deleted im871a
(serial) closing 1 devices

weetmuts, 2021-06-18:
Need more logs to debug this post your syslog and your wmbusmeters log.

Horndrup, 2021-07-05:
I found the problem. I had something in crontab restarting my RPI. Problem solved.

weetmuts, 2021-07-05:
Great!

closed 2021-07-05 by weetmuts