wmbusmeters/wmbusmeters #297: Support for axioma qalcosonic e3

mepetmir, 2021-06-24:
Hello,
I have axioma qalcosonic e3 heating and cooling meter.
I created new driver for it but I can't find second EnergykWh ( 38: EC000000 ). Where is stored cooling energy consumption. 

wmbusmeters --debug simulation_axi3.txt
(wmbusmeters) version: 1.3.0-23-g08b9153 with local changes
(config) using device: simulation_axi3.txt:any
(config) number of meters: 0
(lookup) with file "simulation_axi3.txt"
(lookup) driver: simulation file
(main) opening simulation_axi3.txt:any
Started config simulation on simulation_axi3.txt listening on any
(simulation) in simulation_axi3.txt
(main) regular reset of simulation  on simulation_axi3.txt will happen every 82800 seconds
(main) added simulation_axi3.txt to files
(wmbus) no alarm (expected activity) for simulation
(main) simulation_axi3.txt already configured as simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(simulation) from file "76440907136401030B0D7A78000000046D030FB726346D0000010134FD170000000004207E1BCF0104247E1BCF0104863B0000000004863CEC0000000413644A0800042BCAF1FFFF043BE20200000259B002025D5704026159FE0C781364010384086D3B17BF258408863B000000008408863C28000000"
(wmbus) parseDLL @0 119
(telegram) DLL L=76 C=44 (from meter SND_NR) M=0709 (AXI) A=03016413 VER=0b TYPE=0d (Heat/Cooling load meter) (driver axiq3) DEV= RSSI=0
(wmbus) parseELL @10 109
(wmbus) parseAFL @10 109
(wmbus) parseTPL @10 109
(dvparser) found new format "046D346D34FD170420042404863B04863C0413042B043B0259025D02610C7884086D8408863B8408863C" with hash c4b1, remembering!
(telegram) TPL CI=7a ACC=78 STS=00 CFG=0000 ()
Received telegram from: 03016413
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Heat/Cooling load meter (0x0d)
                   ver: 0x0b
                driver: axiq3
(wmbus) 00: 76 length (118 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 0907 dll-mfct (AXI)
(wmbus) 04: 13640103 dll-id (03016413)
(wmbus) 08: 0b dll-version
(wmbus) 09: 0d dll-type (Heat/Cooling load meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: 78 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field (OK)
(wmbus) 0d: 0000 tpl-cfg 0000 ( )
(wmbus) 0f: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 10: 6D vif (Date and time type)
(wmbus) 11: 030FB726
(wmbus) 15: 34 dif (32 Bit Integer/Binary Value during error state)
(wmbus) 16: 6D vif (Date and time type)
(wmbus) 17: 00000101
(wmbus) 1b: 34 dif (32 Bit Integer/Binary Value during error state)
(wmbus) 1c: FD vif (Second extension FD of VIF-codes)
(wmbus) 1d: 17 vife (Error flags (binary))
(wmbus) 1e: 00000000
(wmbus) 22: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 23: 20 vif (On time seconds)
(wmbus) 24: 7E1BCF01
(wmbus) 28: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 29: 24 vif (Operating time seconds)
(wmbus) 2a: 7E1BCF01
(wmbus) 2e: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 2f: 86 vif (Energy kWh)
(wmbus) 30: 3B vife (forward flow)
(wmbus) 31: 00000000
(wmbus) 35: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 36: 86 vif (Energy kWh)
(wmbus) 37: 3C vife (backward flow)
(wmbus) 38: EC000000
(wmbus) 3c: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 3d: 13 vif (Volume l)
(wmbus) 3e: 644A0800
(wmbus) 42: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 43: 2B vif (Power W)
(wmbus) 44: CAF1FFFF
(wmbus) 48: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 49: 3B vif (Volume flow l/h)
(wmbus) 4a: E2020000
(wmbus) 4e: 02 dif (16 Bit Integer/Binary Instantaneous value)
(wmbus) 4f: 59 vif (Flow temperature 10⁻² °C)
(wmbus) 50: B002
(wmbus) 52: 02 dif (16 Bit Integer/Binary Instantaneous value)
(wmbus) 53: 5D vif (Return temperature 10⁻² °C)
(wmbus) 54: 5704
(wmbus) 56: 02 dif (16 Bit Integer/Binary Instantaneous value)
(wmbus) 57: 61 vif (Temperature difference 10⁻² K)
(wmbus) 58: 59FE
(wmbus) 5a: 0C dif (8 digit BCD Instantaneous value)
(wmbus) 5b: 78 vif (Fabrication no)
(wmbus) 5c: 13640103
(wmbus) 60: 84 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 61: 08 dife (subunit=0 tariff=0 storagenr=16)
(wmbus) 62: 6D vif (Date and time type)
(wmbus) 63: 3B17BF25
(wmbus) 67: 84 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 68: 08 dife (subunit=0 tariff=0 storagenr=16)
(wmbus) 69: 86 vif (Energy kWh)
(wmbus) 6a: 3B vife (forward flow)
(wmbus) 6b: 00000000
(wmbus) 6f: 84 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 70: 08 dife (subunit=0 tariff=0 storagenr=16)
(wmbus) 71: 86 vif (Energy kWh)
(wmbus) 72: 3C vife (backward flow)
(wmbus) 73: 28000000
telegram=||76440907136401030B0D7A78000000046D030FB726346D0000010134FD170000000004207E1BCF0104247E1BCF0104863B0000000004863CEC0000000413644A0800042BCAF1FFFF043BE20200000259B002025D5704026159FE0C781364010384086D3B17BF258408863B000000008408863C28000000|+0
(serial) stopping manager
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted simulation

Here some telegrams:
telegram=|76440907086401030B0D7A6D000000|046D040EB726346D0000010134FD170000000004201505CF0104241505CF0104863B0000000004863CD80000000413B4130600042B78F1FFFF043B0602000002596F02025DD704026198FD0C780864010384086D3B17BF258408863B000000008408863C0B000000|
telegram=|76440907086401030B0D7A6E000000|046D000FB726346D0000010134FD170000000004204212CF0104244212CF0104863B0000000004863CDC000000041398150600042B77F0FFFF043B0202000002597C02025D1405026169FD0C780864010384086D3B17BF258408863B000000008408863C0B000000|
telegram=|76440907086401030B0D7A71000000|046D010FB726346D0000010134FD170000000004208112CF0104248112CF0104863B0000000004863CDC0000000413A1150600042B75F0FFFF043B0202000002595102025DEA04026168FD0C780864010384086D3B17BF258408863B000000008408863C0B000000|
telegram=|76440907086401030B0D7A73000000|046D010FB726346D0000010134FD17000000000420AA12CF010424AA12CF0104863B0000000004863CDC0000000413A7150600042BE9F0FFFF043B0402000002597402025DF304026182FD0C780864010384086D3B17BF258408863B000000008408863C0B000000|
telegram=|76440907086401030B0D7A74000000|046D020FB726346D0000010134FD17000000000420BD12CF010424BD12CF0104863B0000000004863CDC0000000413A9150600042B29F1FFFF043B0302000002599602025DFF04026198FD0C780864010384086D3B17BF258408863B000000008408863C0B000000|
telegram=|76440907086401030B0D7A75000000|046D020FB726346D0000010134FD17000000000420D312CF010424D312CF0104863B0000000004863CDC0000000413AC150600042BA4F1FFFF043B040200000259C502025D290502619DFD0C780864010384086D3B17BF258408863B000000008408863C0B000000|
telegram=|76440907086401030B0D7A76000000|046D020FB726346D0000010134FD17000000000420E612CF010424E612CF0104863B0000000004863CDC0000000413AF150600042BBAF1FFFF043B040200000259C402025D24050261A0FD0C780864010384086D3B17BF258408863B000000008408863C0B000000|
telegram=|76440907086401030B0D7A77000000|046D030FB726346D0000010134FD17000000000420FC12CF010424FC12CF0104863B0000000004863CDC0000000413B2150600042BBAF1FFFF043B040200000259C202025D2A05026198FD0C780864010384086D3B17BF258408863B000000008408863C0B000000|
telegram=|76440907086401030B0D7A78000000|046D030FB726346D0000010134FD170000000004200F13CF0104240F13CF0104863B0000000004863CDC0000000413B5150600042B86F1FFFF043B030200000259C002025D2C05026194FD0C780864010384086D3B17BF258408863B000000008408863C0B000000|

weetmuts, 2021-06-26:
That looks like a nice meter to add a driver for. All the telegram bytes seem to be decoded nicely.
Could it be that the accumulated cooling energy is stored as backward flow and accumulated heating energy is forward flow?

mepetmir, 2021-06-26:
Yes cooling energy is sotred as backward flow. I don't know how to filter it

weetmuts, 2021-06-26:
Ah, yes, you have to hardcode the key for the moment. The findKey currently does not have an api
for searching for VIFEs like backward flow etc. I will add that soon.

Look into meter_amiplus.cc for an example:

    extractDVdouble(&t->values, "0E833C", &offset, &total_energy_returned_kwh_);                                                             
    t->addMoreExplanation(offset, " total energy returned (%f kwh)", total_energy_returned_kwh_);

In your case the key is "04863C".

mepetmir, 2021-06-29:
Thanx it works

weetmuts, 2021-07-01:
Great! Looking forward to your pull request!

weetmuts, 2021-08-08:
@mepetimir Do you have a pull request for us?

weetmuts, 2022-06-18:
Thanks for reporting the telegrams. I have added the driver myself, check out driver_qualcosonic.cc it is much easier to write drivers now.

You can test the decode here:
https://wmbusmeters.org/analyze/76440907086401030B0d7a78000000046d030fB726346d0000010134fd170000000004200f13cf0104240f13cf0104863B0000000004863cdc0000000413B5150600042B86f1ffff043B030200000259c002025d2c05026194fd0c780864010384086d3B17Bf258408863B000000008408863c0B000000

Closing this now.

closed 2022-06-18 by weetmuts