wmbusmeters/wmbusmeters #300: No data after hours of listening

CWempe, 2021-07-01:
Sorry if this is the wrong place to post my issue, but I found no support forum.

I recently got new meters installed in my flat that use wmbus.
* Metrona (Q water 5.5) `DE-11-MI001-PTB005`
* Metrona `T230-A21C-DE08-P`

So I wanted to see if I can read the data with my "blue DVB-T-Stick" (SDR) I had laying around.
I successfully received 433MHz data a while ago. So the dongle should work in general.

I am running wmbusmeters in docker in privleged mode on a PC (Intel, Debian).

My config:

```ini
loglevel=debug
device=rtlwmbus
listento=c1,t1
logtelegrams=true
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
logfile=/wmbusmeters_data/logs/wmbusmeters.log
```

The logs indicate the dongle is detected correctly, I think.

```
(wmbusmeters) version: 1.3.0-29-g3ee42c6
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: rtlwmbus
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)

<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 11 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 12 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
Started config rtlwmbus[00000001] listening on c1,t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 17 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
```

But even after more than 24h there are no new log lines after this.
`meter_readings` is empty.
I would have expected at least some debug messages.

Am I missing something?
Is my configuration wrong?

weetmuts, 2021-07-01:
But wmbusmeters listens by default on 868 MHz. If you try to run exactly this command line instead,
then it will listen to 433 MHz.

wmbusmeters "rtlwmbus:CMD(rtl_sdr -f 433M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)"

weetmuts, 2021-07-01:
You can also experiment with the rtl_sdr rtl_wmbus settings first, without involving wmbusmeters.'
rtl_sdr -f 433M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a
rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a
rtl_sdr -f 868.625M -s 1600000 - 2>/dev/null | rtl_wmbus -s 

Try tweaking the 868.95 to 868.90 869.0 and other nearby frequencies.
Sometimes your rtl_sdr dongle is not perfectly tuned.

weetmuts, 2021-07-01:
You also need to double check your tech specs for your meters for which frequency they are transmitting on.

CWempe, 2021-07-01:
Thanks. I will check and fine tune.

CWempe, 2021-07-02:
It works with `868.90M`. 🥳 

So here is an example with the new frequency:

```
Started config rtlwmbus[00000001] listening on c1,t1 using fq 868.90M
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 94025233
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 45 dBm
                driver: unknown!
telegram=||49449344335202943408780DFF5F350082180000800007B06EFFFF970000009F2C70020000BE26970000000000010018002E001F002E0023FF210008000500020000002F046D220FA227|+2
Received telegram from: 94310409
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 13 dBm
                driver: unknown!
```

The whole log: https://pastebin.com/g2T1u68q

`driver: unknown!` does not sound good.
How do I proceed now?

weetmuts, 2021-07-02:
I put the next line into a file simulation_qundis.txt:
telegram=||49449344335202943408780DFF5F350082180000800007B06EFFFF970000009F2C70020000BE26970000000000010018002E001F002E0023FF210008000500020000002F046D220FA227|

I then run:
wmbusmeters --debug simulation_qundis.txt

(You can also add --debug already when collecting the telegrams.)

The part of the output that is interesting is:

(wmbus) 00: 49 length (73 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 9344 dll-mfct (QDS)
(wmbus) 04: 33520294 dll-id (94025233)
(wmbus) 08: 34 dll-version
(wmbus) 09: 08 dll-type (Heat Cost Allocator)
(wmbus) 0a: 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(wmbus) 0b: 0D dif (variable length Instantaneous value)
(wmbus) 0c: FF vif (Vendor extension)
(wmbus) 0d: 5F vife (duration of limit exceed last upper  is 3)
(wmbus) 0e: 35 varlen=53
(wmbus) 0f: 0082180000800007B06EFFFF970000009F2C70020000BE26970000000000010018002E001F002E0023FF210008000500020000002F
(wmbus) 44: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 45: 6D vif (Date and time type)
(wmbus) 46: 220FA227

From this we can see that HCA sends 54 bytes of proprietary vendor data. Blech.
However these HCA meters often alternate between two different telegrams.
If you listen longer, perhaps you will receive a telegram where the content of the HCA
is not hidden inside a proprietary blob, but instead encoded using the wmbus standard.

If so, the driver whe4x or whe5x could be the right driver for those telegrams.

CWempe, 2021-07-03:
I think I am getting there.

I receive many meters.

```
grep "Received telegram from" wmbusmeters.log | sort -u
Received telegram from: 82087337
Received telegram from: 94025093
Received telegram from: 94025094
Received telegram from: 94025095
Received telegram from: 94025101
Received telegram from: 94025105
Received telegram from: 94025106
Received telegram from: 94025107
Received telegram from: 94025108
Received telegram from: 94025114
Received telegram from: 94025126
Received telegram from: 94025127
Received telegram from: 94025128
Received telegram from: 94025225
Received telegram from: 94025226
Received telegram from: 94025227
Received telegram from: 94025228
Received telegram from: 94025229
Received telegram from: 94025231
Received telegram from: 94025233
Received telegram from: 94025238
Received telegram from: 94025239
Received telegram from: 94025240
Received telegram from: 94025241
Received telegram from: 94025242
Received telegram from: 94025243
Received telegram from: 94025244
Received telegram from: 94025252
Received telegram from: 94025253
Received telegram from: 94251927
Received telegram from: 94310409
Received telegram from: 94310412
Received telegram from: 94310413
Received telegram from: 94310414
Received telegram from: 94310422
Received telegram from: 94310424
Received telegram from: 95440829
```

I created a `wmbusmeters.d` file for each.

```shell
#!/bin/bash

FILE_PATH="../etc/wmbusmeters.d"
FILE_PREFIX="zaehler"

# type or driver
#TYPE="whe46x"
TYPE="whe5x"

for ID in $(grep "Received telegram from" wmbusmeters.log | sort -u | awk '{print $4}')
do
    echo "name=${FILE_PREFIX}_${ID}"  > ${FILE_PATH}/${FILE_PREFIX}_${ID}
    echo "type=${TYPE}"              >> ${FILE_PATH}/${FILE_PREFIX}_${ID}
    echo "id=${ID}"                  >> ${FILE_PATH}/${FILE_PREFIX}_${ID}
    echo "#key="                     >> ${FILE_PATH}/${FILE_PREFIX}_${ID}
done
```

Place in `/wmbusmeters/_data/logs/` and execute there.

I tried both drivers you mentioned.

```json
{
    "media": "heat cost allocation",
    "meter": "whe5x",
    "name": "zaehler_94025114",
    "id": "94025114",
    "current_consumption_hca": 116,
    "set_date": "2020-12-31",
    "consumption_at_set_date_hca": 390,
    "set_date_1": "2020-12-31",
    "consumption_at_set_date_1_hca": 390,
    "error_date": "2127-15-31",
    "device_date_time": "2021-07-02 23:30",
    "timestamp": "2021-07-02T22:22:08Z",
    "device": "rtlwmbus[00000001]",
    "rssi_dbm": 25
}
```

```json
{
    "media": "heat cost allocation",
    "meter": "whe46x",
    "name": "zaehler_94025114",
    "id": "94025114",
    "current_consumption_hca": 116,
    "set_date": "2020-12-31",
    "consumption_at_set_date_hca": 390,
    "set_date_1": "2020-12-31",
    "consumption_at_set_date_1_hca": 390,
    "flow_temperature_c": 127,
    "error_date": "2127-15-31",
    "device_date_time": "2021-07-02 23:23",
    "unknown": "",
    "timestamp": "2021-07-02T22:14:30Z",
    "device": "rtlwmbus[00000001]",
    "rssi_dbm": 26
}
```

The only difference seems to be `"unknown": "",`.
I guess `whe5x` is the better one then.

Next I will have to figure out which one is my meter. 😄

weetmuts, 2021-07-03:
Nice! This means that there are telegrams with proper wmbus data.
Can you post a few of these here? Usually there is a length difference
so it should easy to spot them in the debug log, or using only --logtelegrams

I would like to add the meter version to wmbusmeters so that it can
pick the right driver automatically.

CWempe, 2021-07-03:
Here is about 24h of debug logs. :)

https://cloud.cwempe.ddnss.de/s/x64KMmJZ5ZsxKf6

(link is valid for 3 days)

CWempe, 2021-07-03:
btw: Are these 37 devices all the same types?

I do not know about the other houses around me, but the 4 flats in this house got upgraded last month:

* cold water meter
  * `allmess EVK 3/110 +m`
  *  only 2 flats I think
  * maybe not even with wireless M-Bus
* warm water meter
  * Metrona (Q water 5.5)
  * `DE-11-MI001-PTB005`
* heating water cycle
  * Metrona 
  * `T230-A21C-DE08-P`
  * calculates difference between inflow and outflow  
   I do not know how this is called in english

weetmuts, 2021-07-04:
> Here is about 24h of debug logs. :)

Thanks! The correct driver is actually qcaloric. 
I have now pushed a fix to wmbusmeters so that it will 
recognize this HCA automatically and pick the qcaloric driver.

CWempe, 2021-07-04:
Great!

Some meters now report "0" or "" with the new (automatically detected) driver.

whe5x:

```json
{
    "media": "heat cost allocation",
    "meter": "whe5x",
    "name": "zaehler_94025228",
    "id": "94025228",
    "current_consumption_hca": 251,
    "set_date": "2020-12-31",
    "consumption_at_set_date_hca": 287,
    "set_date_1": "2020-12-31",
    "consumption_at_set_date_1_hca": 287,
    "error_date": "2127-15-31",
    "device_date_time": "2021-07-04 12:07",
    "timestamp": "2021-07-04T10:59:10Z",
    "device": "rtlwmbus[00000001]",
    "rssi_dbm": 15
}
```

qcaloric:

```json
{
    "media": "heat cost allocation",
    "meter": "qcaloric",
    "name": "zaehler_94025228",
    "id": "94025228",
    "current_consumption_hca": 0,
    "set_date": "",
    "consumption_at_set_date_hca": 0,
    "set_date_1": "",
    "consumption_at_set_date_1_hca": 0,
    "set_date_17": "",
    "consumption_at_set_date_17_hca": 0,
    "error_date": "",
    "device_date_time": "2021-07-04 11:50",
    "timestamp": "2021-07-04T10:42:12Z",
    "device": "rtlwmbus[00000001]",
    "rssi_dbm": 16
}
```

I guess that means the new driver is not correct for these meters?!

weetmuts, 2021-07-04:
It is the expected behaviour. The meters send two kinds of telegrams, one
with no data (only proprietary blob)
and the other with actual data. Some meters send multiple telegrams with
some data in on, and other data in others.
wmbusmeters creates a c++ object for each meter and maintains the state.
The full state is then propagated in the json message.

However when a telegram with no data has arrived first, then the c++ object
contains no data,
but a json message will still be sent so it will contain zeroes. However if
you wait for a while,
the state will be updated properly and real values will show.

Den sön 4 juli 2021 kl 13:15 skrev Christoph Wempe ***@***.***

CWempe, 2021-07-04:
Ok, I understand.
But does this not lead to unwanted 0-values when wmbusmeters is restarted?

see:
[image]

Another question:

Am I correct that all my received meters are heat cost allocators? (every json has `current_consumption_hca`)
What about my warm water meter?
The display only shows `XXX m³` (volume), but non of the received telegrams show `m³`.
Is there even a unit for hca?
Maybe I also need to check other nearby frequencies to find more/other meters?

weetmuts, 2021-07-05:
The 37 devices found in the big log file you sent are all Heat Cost Allocators.
Do this: 
cat wmbusmeters.log | grep telegram= | cut -f 1-3 -d '|' > simulation_foof.txt
wmbusmeters simulation_foof.txt 
wmbusmeters simulation_foof.txt 2>&1  | grep driver: | sort | uniq -c

The HCA is dimensionless (no unit),  but I still suffix those values with _hca to easily spot them in the json and in the code.

Your water meter can be sending on s1 which uses a different frequency and linkmode.
But check any docs on your meter.
Also some meters only transmit during office hours (no nights nor weekends) to preserve battery.

CWempe, 2021-07-05:
Thanks.

The warm water meter (Q-water-5.5) seems to support C and S. (if that is the same as c1 and s1):

[image]

[image]

But I do not know if both modes are enabled.

I will add `s1` and try `868.3M` (and `868.25M`).
Let's see what that gets me. 😀

btw:
`"rssi_dbm":17` means `-17 dBm`, right?
So `"rssi_dbm":17` is a stronger signal than `"rssi_dbm":110`.

pswiatki, 2021-07-05:
Sadly, `rtlsdr` does not report RF level in dBm. I would say 17 is a relatively weak signal (but I get proper readings from my water meter when `rssi_dbm` is as low as 4-5). On the other hand, my power meter seems to TX with force :) I get 130+ from that one. Surely, they have all the power they need for that RF transmission right in the box, huh ;-)

weetmuts, 2021-07-05:
Higher rssi is better.

For dbm reporting dongles: -50 is better than -100.
For rtlsdr 100 is better than 5. :-)

weetmuts, 2021-07-10:
> Ok, I understand.
> But does this not lead to unwanted 0-values when wmbusmeters is restarted?

Yes it does. Hmmm, perhaps when we can, we should delay reporting a telegram until the complete set of data has arrived.
Or we should set unknown readings to null?
What do you think?

CWempe, 2021-07-10:
I am still not getting any S1-telegrams. :(

First I tried:

```ini
listento=c1,t1,s1
device=rtlwmbus:868.30M
```

And `868.25M`.
(Remember I needed to offset the frequency by `-0.05`. https://github.com/weetmuts/wmbusmeters/issues/300#issuecomment-873051608)

But then I [read](https://github.com/weetmuts/rtl-wmbus#:~:text=simultaneously) I need to define a frequency between `c1,t1` and `s1`.

So I tried:

```ini
listento=c1,t1,s1
device=rtlwmbus:CMD(rtl_sdr -f 868.575M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s)
```

Now my docker container is restarting every few seconds.

```
[2021-07-10_23:19:19] (wmbusmeters) version: 1.3.0-40-gcf9a22a
[2021-07-10_23:19:19] (config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
[2021-07-10_23:19:19] (config) using device: rtlwmbus:CMD(rtl_sdr -f 868.575M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s)
[2021-07-10_23:19:19] (config) number of meters: 37
[2021-07-10_23:19:19] (serial) expecting devices to work
[2021-07-10_23:19:19] (serial) expecting devices to work
[2021-07-10_23:19:19] (lookup) with cmd "rtlwmbus:CMD(rtl_sdr -f 868.575M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s)"
[2021-07-10_23:19:19] (main) opening rtlwmbus:CMD(rtl_sdr -f 868.575M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s)
[2021-07-10_23:19:19] Started config rtlwmbus listening on c1,s1,t1 using CMD(rtl_sdr -f 868.575M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s)
[2021-07-10_23:19:19] (rtlwmbus) using command: rtl_sdr -f 868.575M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s
[2021-07-10_23:19:19] (rtlwmbus) opening cmd_0
[2021-07-10_23:19:19] (bgshell) exec background "/bin/sh"
[2021-07-10_23:19:19] (serial) EVENT thread interrupted
[2021-07-10_23:19:19] (bgshell) arg "-c"
[2021-07-10_23:19:19] (serial) internal error after select! errno=Invalid argument
[2021-07-10_23:19:19] (bgshell) arg "rtl_sdr -f 868.575M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s"
[2021-07-10_23:19:19] (serialcmd) opened /bin/sh pid 12 fd 3 (rtlwmbus)
[2021-07-10_23:19:19] (main) regular reset of rtlwmbus  using CMD(rtl_sdr -f 868.575M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s) will happen every 82800 seconds
[2021-07-10_23:19:19] (wmbus) no alarm (expected activity) for rtlwmbus
[2021-07-10_23:19:19] (serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
[2021-07-10_23:19:19] (serial) waiting for stop
[2021-07-10_23:19:20] (serial) MAIN thread interrupted
[2021-07-10_23:19:20] (serial) EVENT thread interrupted
[2021-07-10_23:19:20] (serial) internal error after select! errno=Invalid argument
[2021-07-10_23:19:20] (serial) received ascii "/bin/sh: build/rtl_wmbus: not found<0A>"
[2021-07-10_23:19:20] (rtlwmbus) checkRTLWMBusFrame "/bin/sh: build/rtl_wmbus: not found<0A>"
[2021-07-10_23:19:20] (rtlwmbus) only text
[2021-07-10_23:19:20] (bgshell) 12 exited with return code 127
[2021-07-10_23:19:20] (serial) closing non working fd=3 "cmd_0"
[2021-07-10_23:19:20] (wmbus) disconnected cmd_0 rtlwmbus
[2021-07-10_23:19:20] (serialcmd) closed /bin/sh pid=12 fd=3 (rtlwmbus)
[2021-07-10_23:19:20] (serial) no devices working emergency exit!
[2021-07-10_23:19:20] (serial) stopping manager
[2021-07-10_23:19:20] (serial) non working devices found, exiting.
[2021-07-10_23:19:20] (serial) MAIN thread interrupted
[2021-07-10_23:19:20] (serial) event loop stopped!
[2021-07-10_23:19:20] (serial) TIMER thread interrupted
[2021-07-10_23:19:20] (wmbus) deleted rtlwmbus
```

Does the Docker image not contain `rtl_wmbus`?

I will try the following config next.

```ini
listento=s1
device=rtlwmbus:868.25M
```

Maybe I get results by tomorrow.

CWempe, 2021-07-10:
> Yes it does. Hmmm, perhaps when we can, we should delay reporting a telegram until the complete set of data has arrived.

Yes, I think no telegram (or at least no mqtt message) should be sent if there are no correct values.

"No data is better than wrong data!" as my colleagues say.

weetmuts, 2021-07-11:
Assuming that the meters transmit right, but the dongle is somewhat un-tuned, try this with different values for ppm:
`wmbusmeters rtlwmbus(ppm=17)`

From the README:
Use this to tune your dongle and at the same time listen to S1,T1 and C1.

CWempe, 2021-07-11:
Ok, I found out what ppm value I need:

[image]

via: https://www.youtube.com/watch?v=pFVS3-NthKM

I am running now with:

```ini
device=rtlwmbus(ppm=72)
listento=c1,t1,s1
```

I am receiving C1-Telegrams.
So it seems to work. 👍 

We will see if I get S1-telegrams, too. 😃

CWempe, 2021-07-12:
Looks like I did not receive any S1-telegrams.

```shell
# grep -i "s1" wmbusmeters.log
[2021-07-11_17:38:53] Started config rtlwmbus[00000001](ppm=17) listening on c1,s1,t1
[2021-07-11_19:38:04] Started config rtlwmbus[00000001](ppm=72) listening on c1,s1,t1
```

Anything else I can look for in my big 24h debug log?

bensaeba, 2021-07-23:
Hi everyone

I'm writing here because I think I have a similar issue. Please let me know if inapropriate. 

My config file
```
loglevel=debug
device=auto:t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
```
I have a amb8465 dongle and I'm trying to read Izar meter

When doing:
`wmbusmeters auto:t1`

I received  many meters --> 94 different meters

```
Received telegram from: 20b841ae
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: amb8465[00000000]
                  rssi: -31 dBm
                driver: izar
Received telegram from: 20adee42
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: amb8465[00000000]
                  rssi: -80 dBm
                driver: izar
```

```
Received telegram from: 20addec4
Received telegram from: 40952091
Received telegram from: 20b841ae
Received telegram from: 20adee42
Received telegram from: 20adee41
Received telegram from: 20b841cf
Received telegram from: 20adee48
Received telegram from: 20adea12
Received telegram from: 20adea0d
Received telegram from: 20ae33e7
....
```

So, I took the first one to test it: 

`wmbusmeters amb8465:t1 MyTapWater auto 20addec4 NOKEY`

And got the following output:

```
wmbusmeters amb8465:t1 MyTapWater auto 20addec4 NOKEY
Started config amb8465[00000000] on /dev/ttyUSB2 listening on t1
MyTapWater	20addec4	C11UA	094756	164.346 m3	155.172 m3	2020-12-28	5 y	no_alarm	no_alarm	32 s	2011	2021-07-23 23:55.48
MyTapWater	20addec4	C11UA	094756	164.346 m3	155.172 m3	2020-12-28	5 y	no_alarm	no_alarm	32 s	2011	2021-07-23 23:56.22
MyTapWater	20addec4	C11UA	094756	164.346 m3	155.172 m3	2020-12-28	5 y	no_alarm	no_alarm	32 s	2011	2021-07-23 23:56.56
MyTapWater	20addec4	C11UA	094756	164.346 m3	155.172 m3	2020-12-28	5 y	no_alarm	no_alarm	32 s	2011	2021-07-23 23:57.36
MyTapWater	20addec4	C11UA	094756	164.346 m3	155.172 m3	2020-12-28	5 y	no_alarm	no_alarm	32 s	2011	2021-07-23 23:58.12
```
(By the way, I'm not sure of how to interpret some of the results)

So, I created a wmbusmeters.d file for 20addec4:
```
name=MyTapWater
id=20adec4
key=NOKEY
```

But meter_readings directory is still empty.

Am I missing something?
Is my configuration wrong?

Thanks you for your help.

regards
Benoit

PS:

status:
```
 wmbusmeters.service - "wmbusmeters service"
   Loaded: loaded (/lib/systemd/system/wmbusmeters.service; enabled; vendor p
   Active: active (running) since Fri 2021-07-23 23:52:57 CEST; 3min 24s ago
     Docs: https://github.com/weetmuts/wmbusmeters
           man:wmbusmeters(1)
  Process: 10643 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readin
  Process: 10644 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/
  Process: 10649 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, st
  Process: 10651 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbu
  Process: 10652 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeter
 Main PID: 10653 (wmbusmetersd)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/wmbusmeters.service
           └─10653 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid

juil. 23 23:52:57 benpi wmbusmetersd[10653]: (config) "meterfiles" "/var/log/
juil. 23 23:52:57 benpi wmbusmetersd[10653]: (config) "meterfilesaction" "ove
juil. 23 23:52:57 benpi wmbusmetersd[10653]: (config) "logfile" "/var/log/wmb
juil. 23 23:52:57 benpi wmbusmetersd[10653]: (config) "" ""
juil. 23 23:52:57 benpi wmbusmetersd[10653]: (config) loading meter file /etc
juil. 23 23:52:57 benpi wmbusmetersd[10653]: (config) name=MyTapWater
juil. 23 23:52:57 benpi wmbusmetersd[10653]: (config) id=20adec4
juil. 23 23:52:57 benpi wmbusmetersd[10653]: (config) key=<notprinted>
lines 1-23
```

CWempe, 2021-07-24:
I am no expert myself, but I think your config is ok.
Do you have more current logs than the wmbusmeters.service status?
Could there be a permission issue with the meter_readings directory?

btw: I think your issue is completely different than mine.
My first issue was not getting any readings at all, because I had to fine tune the frequency of my radio.
Now I get many readings, but my devices are not part of them.

weetmuts, 2021-07-24:
@bensaeba I think you simply misspelled the id in your config file. It is written: id=20adec4
but it should be id=20addec4

It is a completely different problem though, not related to @CWempe problem which seems
to be more related to the physical problems of decoding the radio waves. :-)

weetmuts, 2021-07-24:
@CWempe wrote:
> Looks like I did not receive any S1-telegrams.
>Anything else I can look for in my big 24h debug log?

Well, since you do get the C1 telegrams, the wmbusmeters setup seems to be working all right.
The trick seems to be to configure rtl_wmbus to decode all the telegrams at once.

You can check in your long 24hour debug log for the rtl_wmbus messages, the usually contain C1
for C1 messages, look for S1 and S0 as well. S1 is successfull received S0 detected but failed to decode.

If there are no S0 or S1 then you have to open an issue with the rtl_wmbus developer to help you.
https://github.com/xaelsouth/rtl-wmbus

bensaeba, 2021-07-25:
Hello @weetmuts , @CWempe 
Thanks for your insight, you were right! 
Everything is working just fine now and I even got my water consumption on domoticz
Sorry about the wrong thread of issue.

weetmuts, 2021-07-25:
To add "idx":"337" to a specific meter, add json_idx=337
to your meterfile where you specify its id and key etc.

You only need one shell command where you should not have the idx since it is added by the meter spec.

bensaeba, 2021-07-25:
@weetmuts ,

Thanks for you quick answer,

I tried this

Meter file:

```
name=MyTapWater
id=20adee41
key=
json_idx=337
```

```
name=Hotbathroom
id=20b841a9
key=
json_idx=338
```

config file

```
loglevel=normal
device=amb8465
listento=c1,t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\",\"svalue\":\"$METER_TOTAL_M3\"}"
```
and this

```
loglevel=normal
device=amb8465
listento=c1,t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"svalue\":\"$METER_TOTAL_M3\"}"
```

None of those 2 config file are working
Domotcz log:
Fist case: 2021-07-25 21:11:34.574 Error: MQTT: MQTT: unknown idx received! (idx 0)
second case: 2021-07-25 21:13:05.285 Error: MQTT: MQTT: Invalid data received!

weetmuts, 2021-07-25:
Ah, you are not sending the full JSON, you are building your own json. Try this shell command instead.
```
shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\":\"$METER_IDX\",\"svalue\":\"$METER_TOTAL_M3\"}"
```

bensaeba, 2021-07-25:
Well, I fol

> Ah, you are not sending the full JSON, you are building your own json. Try this shell command instead.
> 

I followed the tuto on the wiki
https://weetmuts.github.io/wmbusmeterswiki/DOMOTICZ.html

I tried your command:
```
loglevel=normal
device=auto:t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\":\"$METER_IDX\",\"svalue\":\"$METER_TOTAL_M3\"}"
```
still the same error message on domoticz:
`2021-07-25 22:08:28.289 Error: MQTT: MQTT: Invalid data received!`

weetmuts, 2021-07-25:
@bensaeba Please open a new issue for this particular problem, and we can continue this discussion there.

weetmuts, 2021-08-08:
@CWempe Any updates on the S1 non-reception-problem?

CWempe, 2021-08-08:
Sorry.
I am occupied with another project at the moment.

I had the following running for a few days now:

```
rtl_sdr -f 868.625M -p 72 -s 1600000 - 2>/dev/null | rtl_wmbus -s
```

No `S0` or `S1` messages for now.
Only many `C1` messages. (So the command seems to be correct)
So I guess I will open an issue for `rtl-wmbus` when I find the time.

weetmuts, 2021-08-08:
Ok! Lets close this for now then. Please open a new issue if you need! Good luck!

closed 2021-08-08 by weetmuts