wmbusmeters/wmbusmeters #301: nanoCUL

krzbor, 2021-07-01:
I bought nanoCUL and connected to wmbusmeters. Unfortunately, it was not recognized. I decided to change the firmware in nanoCUL. I found the only working one at: https://www.smart-home-komponente.de/support/firmware/ (nanoCUL-MBUS.hex). However, only "auto: t1" is working, c1 is not working.
In auto mode: t1 I have no water meter reading. I receive:

Received telegram from: 21407af0
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: cul
                  rssi: -62 dBm
                driver: izar
Received telegram from: 21407af0
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: cul
                  rssi: -61 dBm
                driver: izar
Received telegram from: 21407af0
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: cul
                  rssi: -61 dBm
                driver: izar
Received telegram from: 21407af0
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: cul
                  rssi: -61 dBm
                driver: izar

How to configure wmbusmeters to get correct readings?

weetmuts, 2021-07-02:
You must specify a meter to listen to. This is just the overview of all meters that it can hear.
Also, which version of wmbusmeters are you using?

Try:
wmbusmeters auto:t1 MyWater auto '*' NOKEY

krzbor, 2021-07-02:
Works great! I wanted to test the device first without creating configuration files. I think it's worth adding
wmbusmeters auto: t1 MyWater auto '*' NOKEY
on the main page (readme). It is the simplest command that reads and displays data.
Thank you very much!

weetmuts, 2021-07-02:
Great! It is the simplest command that reads data from meters that do not need any keys!
Unfortunately most meters nowadays do have unique keys.

Also it is the simplest command that listens to all meters, if you hear your neighbours meters
you might have to restrict listening to your own meter:

wmbusmeters auto:t1 MyWater auto 21407af0 NOKEY

> I think it's worth adding wmbusmeters auto: t1 MyWater auto '*' NOKEY on the main page (readme)
Please make a pull request to the README. :-)

Also if you have the time, it would be very valuable if you could add a page to the wiki,
explaining how to build and configure the nanCUL, which is a bit tricky. 

https://github.com/weetmuts/wmbusmeterswiki

weetmuts, 2021-07-02:
You might want to read this page as well:
https://github.com/weetmuts/wmbusmeterswiki/blob/master/IZAR.md

krzbor, 2021-07-02:
I only read the main page and missed IZAR.md. I have wasted many hours of testing unnecessarily.

closed 2021-07-10 by weetmuts