wmbusmeters/wmbusmeters #302: deamon is stopping just after starting - communication with im871a donlge fails

the78mole, 2021-07-02:
I installed wmbusmetersd from source and installed it using `sudo make install`. 
But now I encounter a problem, using the wmbusmeters service. With `wmbusmeters --useconfig=/`, everything runs fine.

I'm using an im871a and everything works fine with the wmbusmeters executable.
Also when I start `wmbusmetersd /var/run/wmbusmeters/wmbusmeters.pid` as root.

If I use `service wmbusmeters start`, I get the following in `/var/log/wmbusmeters/wmbusmeters.log`:
```
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(im871a) got device id 00000000
[2021-07-02_11:14:12] Not the expected dongle (dongle said 00000000, you said 00101387!
(wmbus) deleted im871a
```

When I start it as root, I get the following:
```
(im871a/im170a) info: im871a other firmware=13 hci=01 uid=00001c2f
(serial /dev/ttyUSB0) sent "A5010500"
(serial) received binary "A5810617FF000300B32587131000010001FF0702320001000000099AE1"
(im871a) checkIM871AFrame "A5810617FF000300B32587131000010001FF0702320001000000099AE1"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
```

User wmbusmeters is in group dialout and `/dev/ttyUSB0` is also owned by the dialout group with proper rights.
If deleted before, also the /var/log/wmbusmeters/wmbusmeters.log does not get created. If touched before, the output is written to it.

Any idea?

weetmuts, 2021-07-02:
Which distribution and which hardware are you using?

weetmuts, 2021-07-03:
Ping?

weetmuts, 2021-07-04:
If you are using raspbian, then read this thread: https://github.com/weetmuts/wmbusmeters/issues/270

the78mole, 2021-07-05:
Sorry, was away for the weekend... 

Yes, I use raspbian. I checked the stuff from your linked thread:
wmbusmeters users groups:
wmbusmeters tty uucp dialout cdrom sudo video plugdev games users input netdev gpio i2c spi pi

Rights:
[image]

When I `sudo su wmbusmeters -` and run `wmbusmeters --verbose --debug --useconfig=/`, 
Also when I run `wmbusmetersd /var/run/wmbusmeters.pid` it runs smoothly.

But I still get the problem when running with `service wmbusmeters start`
[image]

weetmuts, 2021-07-05:
Do you get the same error with "sudo systemctl start wmbusmeters" 
?

the78mole, 2021-07-05:
I'll check thin in the evening, when I'm back from work...

the78mole, 2021-07-05:
It crashes also soon after started with `systemctl start wmbusmeters` :-(

weetmuts, 2021-07-05:
wmbusmeters does not crash, does it? It looks like it fails to detect the im871a though.

What does your /etc/wmbusmeters.conf look like?

Please unplug your im871a dongle.
sudo systemctl stop wmbusmeters

sudo su
echo > /var/log/wmbusmeters/wmbusmeters.log
tail -f /var/log/wmbusmeters

in another window.
sudo systemctl start wmbusmeters
Wait 5 seconds.
Plug in the dongle.
Wait 5 seconds.
Unplug the dongle.
sudo systemctl stop wmbusmeters

Email me, or post the wmbusmeters.log here.

Thanks!

weetmuts, 2021-07-05:
Oh, and when you stopped the service, do "sudo killall wmbusmeters" and "sudo killall wmbusmetersd"
just to be on the safe side.

weetmuts, 2021-07-05:
Also post "sudo tail -f /var/log/syslog" after you started wmbusmeters.service
to verify that the wmbusmeters daemon is properly started and not dying over and over again.
Then "sudo tail -f /var/log/syslog" after you stopped the wmbusmeters.service to verify that it stopped nicely.

the78mole, 2021-07-05:
OK, crash was the wrong wording... It stops, when it can not find the im871a device. After stopping, I can start again as root without issues. I think, re-plugging the device does not help, since it always works as root or when not started as a service.

And yes, I always check, if the process is not running by checking the output of `ps aux | grep wmbus`.

My wmbusmeters.conf:
```
loglevel=debug
device=/dev/ttyUSB0:t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name-id
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=HOME=/home/wmbusmeters/ /usr/bin/mosquitto_pub -h 172.22.2.137 -u mqtt -P xxxxx -t wmbusmeters/$METER_ID -m "$METER_JSON"
json_address=myStreet
json_city=myCity
```

After starting wmbusmeters as root (and killing it again:
[image]

Also starting it as the wmbusmeters user works nicely:
[image]

But I found some other interestigng information when trying to start with `service wmbusmeters start`:
[image]

Maybe this is the root cause of the problem...

weetmuts, 2021-07-05:
Ah, that is an old version of the wmbusmeters.service file. 
Can you delete it please.
The new real version should be here: lib/systemd/system/wmbusmeters.service

It should look like this:

[Unit]
Description="wmbusmeters service"
Documentation=https://github.com/weetmuts/wmbusmeters
Documentation=man:wmbusmeters(1)
After=network.target
StopWhenUnneeded=false
StartLimitIntervalSec=10
StartLimitInterval=10
StartLimitBurst=3

[Service]
Type=forking
PrivateTmp=yes
User=wmbusmeters
Group=wmbusmeters
Restart=always
RestartSec=1

# Run ExecStartPre with root-permissions

PermissionsStartOnly=true
ExecStartPre=-/bin/mkdir -p /var/log/wmbusmeters/meter_readings
ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters
ExecStartPre=-/bin/mkdir -p /run/wmbusmeters
ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters

ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid
ExecReload=/bin/kill -HUP $MAINPID
PIDFile=/run/wmbusmeters/wmbusmeters.pid

[Install]
WantedBy=multi-user.target

the78mole, 2021-07-05:
Does it mean, that the make install did not overwrite the old wmbusmeters.service file? 
I experimented with wmbusmeters some time ago...

weetmuts, 2021-07-05:
I just pushed an improvement to the install script that will detect if there is an /etc/systemd/system/wmbusmeters.service
and move it out of the way.

Hopefully this is the cause of your problems....

weetmuts, 2021-07-05:
Can you check that you do not have any wmbusmeters udev rules as well?
find /etc/udev -name "*wmbus*"

weetmuts, 2021-07-05:
Hehe, you are an early adopter, that has returned. :-)

weetmuts, 2021-07-05:
So I learned that /etc/systemd/system/... is for manually tailored services.
The dir /lib/systemd/system/... is for installed services.

the78mole, 2021-07-05:
You are totally right, I was an early adopter. You are also right, that everything below /etc is protected against updates, e.g. in debian packages. But if you install a package and some etc-file differs from the maintainers one, you get informed and have the option to overwrite, diff,... it. For `make install` this is really a problem...
Nevertheless, everything works now! Thank you so much!
Now I only need to port my M-Bus devices to MQTT (and Home Assistant). They have been also connected by iobroker slave and now need to publish their data differently :-)

the78mole, 2021-07-05:
> Can you check that you do not have any wmbusmeters udev rules as well?
> find /etc/udev -name "_wmbus_"

Actuelly I have, but they do not interfere, as it seems...
[image]

closed 2021-07-05 by the78mole

weetmuts, 2021-07-05:
I am very happy that this was solved and the install script got better.

Yeah, the wmbusmeters.service is already started so the udev rule does not have to do anything.
Except that it might change the group on the device to wmbusmeters instead of dialout.

Please remove it and test that it still works.

weetmuts, 2021-07-05:
I added a udev detector and removal to "make install" as well.

the78mole, 2021-07-05:
> I am very happy that this was solved and the install script got better.
>
> Yeah, the wmbusmeters.service is already started so the udev rule does not have to do anything.
> Except that it might change [...]

Uaaaah, it's not easy for me to test the udev rules, since the Rapi with the wmBus Dongle is in the roof and not easily physically accessible. SSH ist currently my preffered access. I can switch it on and off with my PoE-switch for hard reset, but not much else.

weetmuts, 2021-07-05:
No worries! :-) Let it work in peace now.