wmbusmeters/wmbusmeters #304: wmbusmeters on a Raspberry Pi

krzbor, 2021-07-03:
I installed wmbusmeters on a Raspberry Pi and had correct water meter readings. However, I have noticed that my temperature readings have disappeared. The data was also received by the Raspberry Pi (bluetooth). I restored the previous contents of the SD card and restarted the wmbusmeters installation process. It turned out that the problem was not with wmbusmeters installation, but with running it. I remembered that in debug mode I saw wmbusmeters opening and closing ports. BT in Pi works via UART. There was indeed a problem here. In subsequent tests (after restarting Pi) I changed:
wmbusmeters auto: t1 MyWater auto '*' NOKEY
on
wmbusmeters / dev / ttyUSB0: cul: t1 MyWater auto '*' NOKEY
and now everything works!
Conclusion:
The "auto" mode can be used to determine the port. Then you should restart Pi and call everything with a specific port.

weetmuts, 2021-07-03:
Yes, if you only have a single USB2serial device (the wmbus dongles usually have these)
and it always ends up as /dev/ttyUSB0 then hard-coding the device works great! :-) 

But if you have multiple dongles and they sometimes gets unplugged and replugged
in different ports, or if you use a usb port exteunder. Then they can shift between
the device names.

The auto is pretty cool because you can select a dongle by its
own internal id: rtlwmbus[alfa]:433M im871a[12345678]:c1 rtlwmbus[beta] im871a[332233233]:t1
Here the rtlsdr dongle whose id has been set to alfa (using rtl_eeprom -s alfa)
has a 433M fq tuned antenna where beta defaults to the 868.95M antenna.
The first im871a dongle is set to c1 and the second to t1. (This was useful for im871a with older firmware.)
Now a single im871a can listen to c1 and t1 at the same time.

NanoCul does not seem to have an internal id, unfortunately. 

There is a setting called donotprobe=/dev/ttyUSB123
that will prevent wmbusmeters from disturbing a particular serial port.
You can use it multiple times if you have several sensitive serial ports.
To be on the safe side, you can set donotprobe=all
but then you will have to specify the exact device paths.

Is there a way to detect that the UART/serial port is in use by bluetooth?
If we could detect that, we could prevent wmbusmeters from probing it,
since currently no wmbus-dongle uses bluetooth. They will probably
never use it either...

krzbor, 2021-07-03:
I found a description like this:
https://www.raspberrypi.org/documentation/configuration/uart.md
It shows that this is /dev/ttyAMA0

weetmuts, 2021-08-08:
Thanks! I added a hint to the README for raspberry pi:s to use auto to find the device and then hard-code the device itself using for example /dev/ttyUSBx. 

But it is a generic problem, some embedded devices even have a reboot tty which must not be probed, otherwise the computer will be reboot... :-) So it is hard for wmbusmeters to know the quirks of every single platform.

closed 2021-08-08 by weetmuts