wmbusmeters/wmbusmeters #305: add timestamps to logs

CWempe, 2021-07-03:
I think it would be useful to have timestamps in front of each log line.

now:

```
(wmbusmeters) version: 1.3.0-39-gc30ad88
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: rtlwmbus:868.90M 
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
```

my wish:

```
Jun 28 00:45:01 (wmbusmeters) version: 1.3.0-39-gc30ad88
Jun 28 00:45:01 (config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
Jun 28 00:45:01 (config) using device: rtlwmbus:868.90M 
Jun 28 00:45:01 (config) number of meters: 0
Jun 28 00:45:01 (shell) exec (capture output) "/bin/sh"
Jun 28 00:45:01 (shell) arg "-c"
Jun 28 00:45:01 (shell) arg "rtl_sdr < /dev/null"
Jun 28 00:45:01 (shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
```

weetmuts, 2021-07-03:
Read the readme. ;-) 
logtimestamps

CWempe, 2021-07-03:
Shame on my. 😞

closed 2021-07-03 by CWempe