wmbusmeters/wmbusmeters #307: No meters configured. Printing id:s of all telegrams heard!

ptorrent, 2021-07-07:
Hello there !

I m trying to use wmbusmeters.

I create a directory with meters informations inside
```
name=device1
id=74885803
key=B8DA913940EA648291F7BA97A94D095D
```

then I execute the commande:

```
wmbusmeters --meterfiles=[my directory] auto:c1 /dev/ttyUSB0
```

I receive some data:

[image]

But where can I find the value ?

weetmuts, 2021-07-07:
From commandline:
wmbusmeters  --format=json auto:c1 device1 auto 74885803 B8DA913940EA648291F7BA97A94D095D

ptorrent, 2021-07-07:
oh much better, is there a way to get those values with meter file config ?

weetmuts, 2021-07-07:
Sure is. Read the part of the README that is titled: Build from source and run as a daemon
There are examples of how to write the /etc/wmbusmeters.conf 
and how to add /etc/wmbusmeters.d/MyMeter 
files with meter configs.

weetmuts, 2021-07-07:
When the config is written. You can either run as a daemon or run this command line:
wmbusmeters --useconfig=/
To pickup the settings you have put in place in /etc/....

ptorrent, 2021-07-07:
this is not what I do with meterfiles  options ?

```
wmbusmeters --meterfiles=[my directory] auto:c1 /dev/ttyUSB0
```

weetmuts, 2021-07-07:
Nope, read the README.
--meterfiles=<dir> specifies that wmbusmeters should write each received meter measurement as a file in this <dir>.

ptorrent, 2021-07-07:
Ok thanks ! So if I understand well, there is no way to get metter value juste in stdout or stderr of the command line ? We've to read the file  for each meter ?

weetmuts, 2021-07-07:
Just do not use --meterfiles and it will output all meters on stdout.

ptorrent, 2021-07-07:
Thanks ! Its works. Last question (I hope:)

Is there a way to add meter infomations in the json ?

[image]

weetmuts, 2021-07-07:
Sure can. Again in the README it says add:
--json_xxx=yyy
This will put "xxx":"yyy" into the generated json. You can do this multiple times.
You can have per meter unique settings assuming you have
/etc/wmbusmeters.d/MyMeterFloor1
you can add
json_floor=1
to that file, thus adding "floor":"1"
only to telegrams coming from that meter.

The manufacturer is not actually available right now in the telegram json.
No-one has had any need for that, but it is easy to add.

For historical, slightly random reasons:
The type is called media in the json.
The driver is called meter in the json.

closed 2021-07-08 by ptorrent