wmbusmeters/wmbusmeters #310: RevPi /dev/ttyConBridge

ptorrent, 2021-07-08:
Hello there !

Do you have some experiences with RevPi Mbus module ?

https://revolution.kunbus.com/tutorials/revpi-con-m-bus-overview/

Do you think this driver works with it ?

BIBOLV, 2021-07-08:
Looks like it uses RC1180-MBUS chip, but I could not find much of details, so if you have a device you can try to use `/dev/ttyConBridge:19200` and see if you can receive S1 telegrams. 
Configuration of module will not work as for that support should be implemented, details in [pages 15-19](url)

ptorrent, 2021-07-09:
Ok I was able to configure the device on T1,

Now I've some packets but seems hard to decode them , I will investigate...

So this driver is not supported by wmbusmeters?

[image]

BIBOLV, 2021-07-09:
Well if device spits out clear wmbus packets over serial then you can try to run ` wmbusmeters /dev/ttyConBridge:19200`
C mode is not mentioned, but better to confirm it with devise manufacturer docs.

weetmuts, 2021-07-10:
Wmbsmeters does support the rc1180 mbus in a usb-dongle with an serial2usb connection.
The RC1180-MBUS only supports T1 mode, as far as I know.

You can try `wmbusmeters rc1180:t1`
or `wmbusmeters /dev/ttyConBridge:rc1180:t1`

But if wmbusmeters does not recognize it as an RC1180, then there something else in between.
It says something that its connected to an m-bus module?

BIBOLV, 2021-07-11:
Yes, sorry, somehow missed that rc1180 IS supported.

weetmuts, 2021-08-08:
The device looks nice. But someone with a physical device must add support for this.
I do not have one.

Also it seems like to sends the wmbus frames over mbus. I have started to implement
plain mbus support as well, but there are so many question here about how the
device actually translates thea async push wmbus telegrams into sync pull m-bus data.

So support can probably be added, but it is not trivial.

weetmuts, 2022-06-18:
Plain mbus support is added to wmbusmeters. So we could give this device a try. 
@ptorrent Do you have this device?

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue to continue with device support implementation.

closed 2023-01-01 by BIBOLV