wmbusmeters/wmbusmeters #312: Can you help with this type of meter, the encryption key was given but no readings

spiri439, 2021-07-09:
(rtl433) dll_len adjusted to 48 from 44. Fix rtl_433? This warning will not be printed again.
(meter) test: meter detection could not find driver for id: 21221796 mfct: (BMT) BMETERS, Italy (0x9b4) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x13
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|3044B4099617222113067A860020052F2F|0C1315000000046D3A37A9270F050B000000002F2F2F2F2F2F2F2F2F2F2F3D89|+32
{"media":"warm water","meter":"auto","name":"test","id":"21221796","meter_info":"","timestamp":"2021-07-09T20:58:32Z","device":"rtl433[cmd_0]","rssi_dbm":999}

BIBOLV, 2021-07-10:
Duplicate of https://github.com/weetmuts/wmbusmeters/issues/308

weetmuts, 2021-07-10:
I have merged Bibos fix now! Please test!

spiri439, 2021-07-11:
worked,
telegram=|3044B4099617222113067ADB0020052F2F|0C1315000000046D3A29AB270F050B000000002F2F2F2F2F2F2F2F2F2F2FA0A2|+159
{"media":"warm water","meter":"hydrodigit","name":"test","id":"21221796","total_m3":0.015,"meter_datetime":"2021-07-11 09:58","timestamp":"2021-07-11T06:58:06Z","device":"rtl433[cmd_0]","rssi_dbm":999}

Thanks a lot

closed 2021-07-11 by weetmuts