wmbusmeters/wmbusmeters #313: UNIX Timestamp and own text in meterfiles selectfields command

Nikodemski2002, 2021-07-12:
Hi! Is there any way to make the timestamp write in UNIX format instead of human-readable data? And also is there a way to add my own text to every reading written to the meterfiles? So at the end the reading in the meterfiles instead of this:

`[meter_id];2.2;78.52;54.27;2021-07-12 14:27.39`

will look like this:

`[my_own_text];[meter_id];2.2;78.52;54.27;[UNIX_timestamp]`

Best regards.

BIBOLV, 2021-07-13:
Hi, while waiting for @weetmuts input you can try to add static fields as described in manual, test it with json and see if it works also with fields. If not, you can use `shell=somescript` and do post processing in script like adding fields and/or converting date to needed format. 
You could also describe in bit more details what you are trying to achieve, so we could try to help.

weetmuts, 2021-08-08:
Hi @Nikodemski2002 ! Pushed a fix. You can now do:
wmbusmeters --field_myextrastuff=my_own_text --format=fields --selectfields=myextrastuff,id,total_m3,timestamp_ut

Please test!

There is also timestamp_utc for explictly selecting UTC format for the timestamp in the fields output.
Symmetrically there is also timestamp_lt for local time, which is identical to the existing timestamp in the fields output.

The json output stays the same. Only the field/key "timestamp" exists and it is in the UTC format.

The option --field_xxx=yyy is identical to --json_xxx=yyy. However the --json is removed from the manual,
but will continue to work forever.