wmbusmeters/wmbusmeters #314: Snap wmbusmeters on a Raspberry Pi

bensaeba, 2021-07-12:
Hello,

As suggested, I'm posting my issue here: 

I have bought an Amber 8465 dongle et installed your program and I followed your instructions here : https://weetmuts.github.io/wmbusmeterswiki/SNAP.html
But now, I’m blocked here (https://github.com/weetmuts/wmbusmeters#build-from-source-and-run-as-a-daemon). It seems my dongle is not detected. I’m most probably doing something wrong but I don’t know what…
[wmbusmeters.log](https://github.com/weetmuts/wmbusmeters/files/6804242/wmbusmeters.log)

[wmbusmeters.conf.txt](https://github.com/weetmuts/wmbusmeters/files/6804245/wmbusmeters.conf.txt)
[wmbusmeters.log.txt](https://github.com/weetmuts/wmbusmeters/files/6804246/wmbusmeters.log.txt)

bensaeba, 2021-07-12:
So, I did as you suggested
Uninstall the snap. Snaps are still experimental, we are trying to solve their problems.
Terminate any remaining wmbusmeters: "sudo killall wmbusmeters"
"sudo killalll wmbusmetersd"
"sudo systemctl stop wmbusmeters.service"
git clone ...
./configure
make 
sudo make install
(You might have to install other packages to be able to build.)
No first try from command line:
wmbusmeters auto:t1
If it detects the amber dongle, then
you can do 
sudo systemctl enable wmbusmeters
sudo systemctl start wmbusmeters

Now you can monitor the program with "tail -f /var/log/wmbusmeters/wmbusmeters.log"

But I got this message when trying to "make"

checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for rtlsdr_get_device_count in -lrtlsdr... yes
checking how to run the C++ preprocessor... g++ -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for working ncursesw... yes
checking ncursesw/curses.h usability... yes
checking ncursesw/curses.h presence... yes
checking for ncursesw/curses.h... yes
configure: creating /home/pi/build/armv7l-unknown-linux-gnueabihf/config.status
config.status: creating /home/pi/build/armv7l-unknown-linux-gnueabihf/spec.gmk
config.status: creating /home/pi/build/armv7l-unknown-linux-gnueabihf/Makefile

$ make /home/pi/build/armv7l-unknown-linux-gnueabihf
make: rien à faire pour « /home/pi/build/armv7l-unknown-linux-gnueabihf ».

makefile seems to be empty
[Makefile.txt](https://github.com/weetmuts/wmbusmeters/files/6804533/Makefile.txt)

BIBOLV, 2021-07-13:
> Hello,
> 
> As suggested, I'm posting my issue here:
> 
> I have bought an Amber 8465 dongle et installed your program and I followed your instructions here : https://weetmuts.github.io/wmbusmeterswiki/SNAP.html
> But now, it seems my dongle is not detected. I’m most probably doing something wrong but I don’t know what…
> [wmbusmeters.log](https://github.com/weetmuts/wmbusmeters/files/6804242/wmbusmeters.log)
> 
> [wmbusmeters.conf.txt](https://github.com/weetmuts/wmbusmeters/files/6804245/wmbusmeters.conf.txt)
> [wmbusmeters.log.txt](https://github.com/weetmuts/wmbusmeters/files/6804246/wmbusmeters.log.txt)
> 
> Thanks in advance for your help.
> Regards
> (I'm using a raspberry pi with Pi OS)

Hi, Try snap from edge channel, it should work.

BIBOLV, 2021-07-13:
Try to go to directory where source is located and then run
`./configure; make; sudo make install`

bensaeba, 2021-07-15:
Hi Bibolv,

Thanks for the help. I'm a beginner on Linux, so I'm still trying to get all the basics.
I forgot to specify the directory where the source was located and was doing `/home/pi/wmbusmeters/configure` instead...
Now that I uninstalled snapd and build from source, everything is OK! :)
I'm getting all the meter of my building.

Next step: Reading only my 4 meters and integrating them in Domoticz

closed 2021-07-23 by bensaeba