wmbusmeters/wmbusmeters #315: q400 and axioma W1

sephiroth6, 2021-07-14:
Hello, I have the Qalcosonic w1 sensor, looks like can be managed like q400, indeed I received the total consumption, but I have some others values not visible in json/payload package sent over mqtt.
[debug-temp.txt](https://github.com/weetmuts/wmbusmeters/files/6817388/debug-temp.txt)

I would like to add temperature info, need to change something in meter_q400.cc, right?

Best regards

weetmuts, 2021-07-15:
It does indeed report with the same MFCT,Media,Version as the q400 driver detects.
But it transmits a lot more.... oh well. Should be easy to fix. 
As you say, simply add code to meter_q400.cc 
Start with the flow temperature. Look in meter_whe46x.cc and copy over all code referring
to flow_temperature_c_

The addPrint should have false,true instead of true,true. Since this is an existing
driver we should not add values on the default tabulated output, thus the first false.
However we can of course add values to the json, the second true.

sephiroth6, 2021-07-22:
Thank you for support!
I tried to edit the [file q400 ](https://pastebin.com/VDEqEA0V)
but I can't see any decode temperature inside json-payload.
I tried to uninstall and compile again everything, but no result.

Am I missing something?
Best regards

BIBOLV, 2021-07-23:
Try to remove `127` from line 44, to end up with `double flow_temperature_c_ {};`
Everything else looks like ok.

weetmuts, 2021-08-08:
Thanks! I have pushed an update to the q400 driver. Please test.

Do you happen to now how the flow vs forward _flow vs backward_flow relate?
Likewise what the historical set_forward_flow and set_backward_flow mean?

weetmuts, 2021-09-25:
Ping?