wmbusmeters/wmbusmeters #316: Apator AT-WMBUS-17-2

Merangle, 2021-07-22:
Hello
Thanks for the tool.

I've got a meter with an Apator-AT-WMBUS-17-2. 
Which driver can I use ? apator172? apator162 ?

When listening with this command "wmbusmeters  auto;c1", I can see many meters with different drivers (izar, mkradio3, unknown).
 When listening with this command "wmbusmeters --logtelegrams --format==json auto:c1 watermeter name_of_driver, serial_number NOKEY), I receive informations about all the meters identified but no one with the right value of the meter. 

Is there an issue with driver ?
Will the driver adaptor172 implemented ? 
Many thanks for your help

BIBOLV, 2021-07-23:
Hi,
From Apator currently only at-wmbus-08 and at-wmbus-16-2 are somewhat supported as Apator uses non-standard protocol. To check and see if support for at-wmbus-17-2 can be added you should log decrypted telegrams and post here.
Check serial number on module and try to find that serial in received telegrams. You should also have decryption key in order to decode telegrams, as they, most probably, are encrypted.

Merangle, 2021-07-24:
Hello 
Thanks for your answer. 

Please find attached two files:

- search.log with the result of the command = wmbusmeters --debug auto:c1 rtlwmbus to check what can be heard
- meters.log with the result of the command =   wmbusmeters --logtelegrams --logfile=/home/pi/meters.log auto:c1 watermeter auto 001143ca,001146d3 NOKEY to gather telegrams received from my two meters
- I have two meters with protocol apator-AT-Wmbus 17-2 identified with number 1131466 (1143ca in hexa) and 1132243 (1146d3 in hexa)

The question is how to get the encryption key ?
The radio module has been installed by a company which collect all meters for the landlord to invoice the water consumption. 
I'm would like to follow my consumption trough Home Assistant with  leakage notification. 

All your comments will be welcome. 
regards
[meters.log](https://github.com/weetmuts/wmbusmeters/files/6872753/meters.log)
[search.log](https://github.com/weetmuts/wmbusmeters/files/6872754/search.log)

BIBOLV, 2021-07-24:
You should ask company that installed them to provide key for each meter, without it nothing can be done.

weetmuts, 2021-08-08:
Hi @Merangle!

Your meter 1132243 (1146d3 in hexa) seems to output a readable telegram, that is not encrypted.
telegram=|1C441486D34611000411A0|0DAC04000040030000000005FF06C3980400|

However it is mostly proprietary data. But fortunately the telegram is short. :-)

Do this:
wmbusmeters --logtelegrams auto:t1 Apa apator162 001146d3 NOKEY
and then collect a lot of telegrams when you send water through the meter and manually read the actual
meter values when the log starts getting collected and when the loging is stopped.
Post the data here.

Then it should be possible to figure out which bytes represent the total_m3.

Please have a go at decoding the values yourself, if you can!

weetmuts, 2021-08-08:
I temporarily use the apator162 driver, but it produces the wrong values, as expected. With the information you collect we can create an apator172 driver.

Merangle, 2021-08-09:
Hello
thanks for the suggestion.
I won't be able to do the test before mid september.
I will let you know as soon as I am able to collect the data.
Best regards

Merangle, 2021-09-21:
Hello

Sorry för thé delay

Please find attached several files with the result of thé suggested query.
În thé name of thé file, you will find thé volume  meters-mmmlll.log
contains the result when thé meter read mmm m and lll liters

Ï hope this helps.
How can Ï help you more ?
best regards

weetmuts, 2022-02-26:
Sorry, I cannot see any attached files? Can be re-attach them or email them to me? Thanks!

akanegally, 2022-03-23:
Hello, I have the same water meter apator at-wmbus-17-2. 
Actually since the 15 of march, all residents have the same water meter.

I managed to capture all the telegrams I recieved throught this command (cf[ all_telegrams.log](https://www.amazon.fr/clouddrive/share/X3esewS3sJCBQJL2MbFZxUdo1xoj4M48KVfVPGtNt4F))
I hope it will helps.
wmbusmeters --debug --t1 auto

On my meter (cf [my_meter.jpg](https://www.amazon.fr/clouddrive/share/Rp80OYVayyr4ZXKlkdpTd5M17xFtcn6ENczHMmk1dDj)), the serial is 1353735.
So I think the id of my telegram is 0014A807.

 I have also captured all the telegram ([my_meter.log](https://www.amazon.fr/clouddrive/share/cLRgC5BqblpGeVRAIzv4S5UpoeZDW3iWtcZzEkHtA05)) with this command :
 ./wmbusmeters --debug auto:t1 watermeter apator162 0014a807 NOKEY

It doesn't seems to work. I think those telegrams are crypted.

I tried to ask for the encryption key but the hotline of my provider was clueless.
Is there a way to get my consuption from an apator172 ?

weetmuts, 2022-05-08:
Hi @akanegally ! Thank you for the log, the meters do not seem to be encrypted.
However it is a proprietary format, but there are so few bytes that this should be easy to figure out.
However, you need to collect a number of telegrams from your own meter. 

Are you using an rtlsdr? Can you remove the antenna, or  wrap parts of the antenna in aluminium foil
or some other metal. Place the rtlsdr/crippled antenna really close to your own meter. If you tweak it enough you 
should only receive your own meter telegrams!

Now collect telegrams over a few hours and take notice of the meter readings at the times you collect
and consume some water so that the readings go up.
So that we know approximately what the meter was sending when you received the telegram.

Use:
wmbusmeters --logtelegrams --t1 auto 2>&1 | tee thelog

Now thelog contains lines like these:
telegram=|_1c44148606ac14000411a03b0600000800030000000005ff027d01000|+1231

Post the log here. :-)

You can now start analysing this yourself if you like. The content that can containg the reading starts after a0
telegram=|_1c44148606ac14000411a0 >>>3b0600000800030000000005ff027d01000<<< |+1231
it should be pretty easy to see which values are increasing over time.

weetmuts, 2022-05-08:
If your meter is indeed 0014A807, then you can first try

wmbusmeters --logtelegrams --t1 auto 2>&1 | grep telegram  | grep 07A81400

and collect some telegrams, then use some water and see how the telegrams change.

akanegally, 2022-05-10:
Hi, I managed to capture new telegrams this sunday.
I tried to use water but my consumption went only from 4.056m3 to 4.320m3
You can find the log [here](https://www.amazon.fr/clouddrive/share/2naAzF63A8XUniMGdW66usKtI4lv1dKBqpwW4qPYPGx)

I'm not sure if it's enough so let me know. 

I will also try to capture some new data today.

akanegally, 2022-06-23:
Hello, I captured some telegram of my meter
it goes from 7033m3 to 7179m3
./wmbusmeters --logtelegrams --t1 auto 2>&1 | grep telegram | grep 07A81400 | tee mon_compteur_now.log
telegram=|_1C44148607A814000411A0BD5200000800030000000005FF05D83D0000|+39
telegram=|_1C44148607A814000411A0BD5200000800030000000005FF05D83D0000|+114
telegram=|_1C44148607A814000411A0BC5200000800030000000005FF05D83D0000|+140
telegram=|_1C44148607A814000411A0BE5200000800030000000005FF05D83D0000|+180
telegram=|_1C44148607A814000411A0C65200000800030000000005FF05D83D0000|+209
telegram=|_1C44148607A814000411A0D85200000800030000000005FF05D83D0000|+277
telegram=|_1C44148607A814000411A0DC5200000800030000000005FF05D83D0000|+317
telegram=|_1C44148607A814000411A0DD5200000800030000000005FF05D83D0000|+359
telegram=|_1C44148607A814000411A0DC5200000800030000000005FF05D83D0000|+415
telegram=|_1C44148607A814000411A0DF5200000800030000000005FF05D83D0000|+455
telegram=|_1C44148607A814000411A0EC5200000800030000000005FF05D83D0000|+495
telegram=|_1C44148607A814000411A0FD5200000800030000000005FF05D83D0000|+546
telegram=|_1C44148607A814000411A0025300000800030000000005FF05D83D0000|+586
telegram=|_1C44148607A814000411A0025300000800030000000005FF05D83D0000|+647
telegram=|_1C44148607A814000411A0025300000800030000000005FF05D83D0000|+756
telegram=|_1C44148607A814000411A0025300000800030000000005FF05D83D0000|+841
telegram=|_1C44148607A814000411A0035300000800030000000005FF05D83D0000|+870
telegram=|_1C44148607A814000411A0025300000800030000000005FF05D83D0000|+939
telegram=|_1C44148607A814000411A0035300000800030000000005FF05D83D0000|+984
telegram=|_1C44148607A814000411A0035300000800030000000005FF05D83D0000|+1068
telegram=|_1C44148607A814000411A0035300000800030000000005FF05D83D0000|+1125
telegram=|_1C44148607A814000411A0025300000800030000000005FF05D83D0000|+1199
telegram=|_1C44148607A814000411A0025300000800030000000005FF05D83D0000|+1264
telegram=|_1C44148607A814000411A0035300000800030000000005FF05D83D0000|+1319
telegram=|_1C44148607A814000411A0035300000800030000000005FF05D83D0000|+1371
telegram=|_1C44148607A814000411A0045300000800030000000005FF05D83D0000|+1411
telegram=|_1C44148607A814000411A0045300000800030000000005FF05D83D0000|+1469
telegram=|_1C44148607A814000411A0055300000800030000000005FF05D83D0000|+1518
telegram=|_1C44148607A814000411A0055300000800030000000005FF05D83D0000|+1598
telegram=|_1C44148607A814000411A0065300000800030000000005FF05D83D0000|+1703
telegram=|_1C44148607A814000411A0065300000800030000000005FF05D83D0000|+1786
telegram=|_1C44148607A814000411A0075300000800030000000005FF05D83D0000|+1826
telegram=|_1C44148607A814000411A0075300000800030000000005FF05D83D0000|+1909
telegram=|_1C44148607A814000411A0075300000800030000000005FF05D83D0000|+1959
telegram=|_1C44148607A814000411A0085300000800030000000005FF05D83D0000|+1986
telegram=|_1C44148607A814000411A0085300000800030000000005FF05D83D0000|+2066
telegram=|_1C44148607A814000411A0095300000800030000000005FF05D83D0000|+2154
telegram=|_1C44148607A814000411A00A5300000800030000000005FF05D83D0000|+2230
telegram=|_1C44148607A814000411A00B5300000800030000000005FF05D83D0000|+2327
telegram=|_1C44148607A814000411A00A5300000800030000000005FF05D83D0000|+2389
telegram=|_1C44148607A814000411A00B5300000800030000000005FF05D83D0000|+2405
telegram=|_1C44148607A814000411A00C5300000800030000000005FF05D83D0000|+2505
telegram=|_1C44148607A814000411A00D5300000800030000000005FF05D83D0000|+2598
telegram=|_1C44148607A814000411A00D5300000800030000000005FF05D83D0000|+2678
telegram=|_1C44148607A814000411A00E5300000800030000000005FF05D83D0000|+2757
telegram=|_1C44148607A814000411A00E5300000800030000000005FF05D83D0000|+2871
telegram=|_1C44148607A814000411A00F5300000800030000000005FF05D83D0000|+2981
telegram=|_1C44148607A814000411A0105300000800030000000005FF05D83D0000|+3067
telegram=|_1C44148607A814000411A0105300000800030000000005FF05D83D0000|+3155
telegram=|_1C44148607A814000411A0105300000800030000000005FF05D83D0000|+3213
telegram=|_1C44148607A814000411A0115300000800030000000005FF05D83D0000|+3253
telegram=|_1C44148607A814000411A0115300000800030000000005FF05D83D0000|+3351
telegram=|_1C44148607A814000411A0125300000800030000000005FF05D83D0000|+3386
telegram=|_1C44148607A814000411A0135300000800030000000005FF05D83D0000|+3451
telegram=|_1C44148607A814000411A0135300000800030000000005FF05D83D0000|+3544
telegram=|_1C44148607A814000411A0145300000800030000000005FF05D83D0000|+3636
telegram=|_1C44148607A814000411A0145300000800030000000005FF05D83D0000|+3711
telegram=|_1C44148607A814000411A0155300000800030000000005FF05D83D0000|+3751
telegram=|_1C44148607A814000411A0155300000800030000000005FF05D83D0000|+3803
telegram=|_1C44148607A814000411A0165300000800030000000005FF05D83D0000|+3888
telegram=|_1C44148607A814000411A0165300000800030000000005FF05D83D0000|+3952
telegram=|_1C44148607A814000411A0175300000800030000000005FF05D83D0000|+3988
telegram=|_1C44148607A814000411A0175300000800030000000005FF05D83D0000|+4070
telegram=|_1C44148607A814000411A0175300000800030000000005FF05D83D0000|+4161
telegram=|_1C44148607A814000411A0185300000800030000000005FF05D83D0000|+4201
telegram=|_1C44148607A814000411A0185300000800030000000005FF05D83D0000|+4241
telegram=|_1C44148607A814000411A0195300000800030000000005FF05D83D0000|+4257
[... 177 lines cut ...]
telegram=|_1C44148607A814000411A0975300000840030000000005FF05D83D0000|+21707
telegram=|_1C44148607A814000411A0975300000840030000000005FF05D83D0000|+21862
[... cut ...]
telegram=|_1C44148607A814000411A0215400000840030000000005FF05D83D0000|+23499
telegram=|_1C44148607A814000411A0215400000840030000000005FF05D83D0000|+23684
telegram=|_1C44148607A814000411A0215400000840030000000005FF05D83D0000|+23811
telegram=|_1C44148607A814000411A0215400000840030000000005FF05D83D0000|+23927
telegram=|_1C44148607A814000411A0215400000840030000000005FF05D83D0000|+24095
telegram=|_1C44148607A814000411A0215400000840030000000005FF05D83D0000|+24361
telegram=|_1C44148607A814000411A0215400000840030000000005FF05D83D0000|+24490

Tell me if it's enough.

weetmuts, 2022-06-24:
Many thanks! It seems like the first 8 hex bytes in reverse order count 1/3 of a m3.
Pushed new driver apator172 please test!

akanegally, 2022-06-24:
Hi, I just tried the new driver and it works perfecly

 ./wmbusmeters auto:t1 watermeter apator172 0014a807 NOKEY
Started auto rtlwmbus[00000001] listening on t1
watermeter      0014a807        7265 m3 2022-06-24 21:56.21

Congratulation!

Thank you for all the hard work you guys are doing.

weetmuts, 2022-06-25:
Yay! Glad to hear that it works!

closed 2022-06-25 by weetmuts

Emmanuel-FR, 2023-01-14:
Thank you so much, I was looking very hard to decode these signals and I discovered your work. The most difficult for me was to find the signal because I tried from Windows with my HackRF. Amazing what you can do with a little bit of signal processing !

Cheers