wmbusmeters/wmbusmeters #317: Sending several meters to Domoticz

bensaeba, 2021-07-26:
Hi,

As suggested, I'm creating a new issue on this subject that started here: https://github.com/weetmuts/wmbusmeters/issues/300#issuecomment-886239532

To summarize:
I got 4 water meter in my appartement (hot and  cold water for both kitchen and bathroom) and I managed to send my first meter to domoticz in idx337 (out of the 4 meters) following the tuto here: https://weetmuts.github.io/wmbusmeterswiki/DOMOTICZ.html

Now, I would like to send the 3 remaining meter to domoticz as well.

4 telegrams:
- 20b841be
- 20adee42
- 20adee41
- 20b841a9

I would like to send those 4 meters in 4 idx in domoticz

My Meter files are as follow:

```
name=bathroom
id=20adee41
key=
json_idx=337
```

```
name=Hotbathroom
id=20b841a9
key=
json_idx=338

```
and my config file:

```
loglevel=normal
device=amb8465
listento=c1,t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\",\"svalue\":\"$METER_TOTAL_M3\"}"
```

In my config file I tried several commands for the last line (shell=/usr...) 

`shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\":337,\"svalue\":\"$METER_TOTAL_M3\"}"`
==> the water consumption of the 4 meter is stored in idx337

`shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\",\"svalue\":\"$METER_TOTAL_M3\"}"`
==> domoticz log: Error: MQTT: MQTT: unknown idx received! (idx 0)

`shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"svalue\":\"$METER_TOTAL_M3\"}"`
==> domoticz log:  Error: MQTT: MQTT: Invalid data received!

`shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\":\"$METER_IDX\",\"svalue\":\"$METER_TOTAL_M3\"}"`
==> domoticz log:  Error: MQTT: MQTT: Invalid data received!

So, I'm requesting your help about this last command to be able to send my water consumption in idx337 to 340.
thanks.

PS: my system: raspeberry 4 with both wmbusmeters and domoticz installed on it.

BIBOLV, 2021-07-26:
Hi,
There are two ways how to solve your issue:
1. In main config in shell cmd change to `$METER_idx` or in each meter config change to `json_IDX=`. Mind the upper and lower case letters. You can actually observe correct env variables by enabling debug loglevel and inspect logfile on received telegrams.
2. Remove shell cmd from main config and add to each meters config with `\"idx\":337`, `\"idx\":338` etc.

bensaeba, 2021-07-27:
Hi @BIBOLV 

Thank you very much. I tried solution 1 and it works great! :)

My Meter files :

```
name=bathroom
id=20adee41
key=
json_idx=337
```

```
name=Hotbathroom
id=20b841a9
key=
json_idx=338
```

config file
```
loglevel=normal
device=auto:t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h localhost -t domoticz/in -m "{\"idx\":$METER_idx,\"svalue\":\"$METER_TOTAL_M3\"}"
```

closed 2021-08-08 by weetmuts