wmbusmeters/wmbusmeters #318: TPR6

mcike59295, 2021-08-01:
HELLO

IS THE ELSTER TPR6 868MHZ COMPATIBLE WITH WATER BECAUSE I ARE NOT RECEIVING ANY SIGNAL
THANK YOU

weetmuts, 2021-08-01:
We have not heard of the meter ELSTER TPR6 868MHZ so far.
Is it transmitting C1 or T1 or S1 telegrams?
Which dongle are you using to collect the radio telegrams?
Do you have a decryption key for the meter?

Assuming you have a working build of the latest wmbusmeters and your
radio dongle of choice installed (build rtl_wmbus as well if you need it for an rtl_sdr dongle.)
Then run:
wmbusmeters --logtelegrams auto:c1 | tee mytelegrams.log
and do this for an hour or so, then do t1 and then s1. 
Post the mytelegrams.log here.

mcike59295, 2021-08-02:
Hello
the three logs are empty.
according to a technical data are equivalent is the TRC600, which can be forced into 433mhz!
how to modify the frequency?
Cordially

weetmuts, 2021-08-02:
Please read the README. It is explained there.

mcike59295, 2021-08-03:
thank you
I abandon the protocol to use is wavenis!
otherwise I receive 433:
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2021-08-03 11:30:56.000;31;16;FFFFFFFF;0x6dffffffffffffffffffffffffffffffffffff96ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff69fffffff0ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff<0A>"

weetmuts, 2021-08-08:
The T1;0 means that the message was dectected but failed to decode.

Either this was a fluke bad reception, but if you always fail to decode, then
try tuning the rtl_sdr by specifying the rtlsdr tuning specifier ppm.
You can specify this through wmbusmeters like this: 
rtlwmbus(ppm=17):433M

You have to read up on which ppm value to use.
https://www.youtube.com/watch?v=pFVS3-NthKM

You can also try a better antenna or move closer to the meter.

weetmuts, 2021-09-25:
Ping? Any success?

weetmuts, 2023-01-30:
@Toniob Thanks for the link to the pdf. It does not look like it sends wmbus telegrams....
It talks mbus with the meter and the translates it into its own radio message.

@merbanan Have you heard about this radio format used by this meter before?

Frequency technology FHSS (15 radio channels)
(Frequency Hopping Spread Spectrum)
Frequency 868 MHz, ISM band
Type of transmission Bi-directional

1) I do not know how to decode the radio waves using rtlsdr.
2) Is it bi-directional always? Then it will be really hard triggering it to send because you have to send using their protocol as well. ie. rtlsdr is not enough.
3) If we can get the bits, then for sure it is probably mbus in there somewhere that can be decoded.

merbanan, 2023-01-30:
@Toniob -> https://github.com/merbanan/rtl_433/blob/master/docs/ANALYZE.md

merbanan, 2023-01-30:
@weetmuts I know about Elster and Honeywell but I have not heard anything about the wireless protocol.

Toniob, 2023-01-30:
It seems the protocol was meant to become a standard but I can’t find anything : https://www.criticalcomms.com.au/content/public-safety/product/wavenis-open-standard-alliance-284857067

merbanan, 2023-01-30:
https://beckassets.blob.core.windows.net/product/readingsample/368080/9781848002029_excerpt_001.pdf

merbanan, 2023-01-30:
https://pdf.directindustry.com/pdf/elster/elster-coronis-product-catalog-en/23338-249951.html

merbanan, 2023-01-30:
https://github.com/dpinte/jwavelib

Toniob, 2023-01-31:
[cu8.zip](https://github.com/weetmuts/wmbusmeters/files/10545059/cu8.zip)

I managed to get 2 signals yesterday. But I really don’t know what to do with them now. I don’t even know if that’s the data I’m looking for.

merbanan, 2023-01-31:
https://triq.org/iqs/

merbanan, 2023-01-31:
The signals look good. Now we need alot of them. Again https://github.com/merbanan/rtl_433/blob/master/docs/ANALYZE.md. And open a ticket in the issue tracker. We'll get there eventually.

merbanan, 2023-01-31:
The signal is FSK modulated.

closed 2023-05-03 by github-actions[bot]