wmbusmeters/wmbusmeters #319: Expected amb8465 on /dev/ttyACM0 but found unknown instead, ignoring it!

antcroissant, 2021-08-07:
Hi,...

i have a problem initialising my amb8465 based stick on the first start of the systemd unit.
A simple restart of the unit or a -HUP helps and the stick is then recognized correctly.

What can I do, to make it work on the first try?

```
$ tail -f /var/log/wmbusmeters/wmbusmetersd.log 

(wmbusmeters) logging started 2021-08-07 08:59:47 using makepkg_1.3.0-50-g2c83c51
[2021-08-07_08:59:53] Expected amb8465 on /dev/ttyACM0 but found unknown instead, ignoring it!
[2021-08-07_08:59:54] Started config cul on /dev/ttyACM1 listening on t1
[2021-08-07_08:59:56] Started config cul on /dev/ttyACM2 listening on c1
[2021-08-07_08:59:56] (wmbusmeters) waiting for telegrams
[2021-08-07_08:59:58] (memory) rss 2588672 peak 2.46 MiB

$ sudo killall -HUP wmbusmetersd

[2021-08-07_09:00:05] (wmbusmeters) shutting down
[2021-08-07_09:00:05] (wmbusmeters) HUP received, restarting and reloading config files.
(wmbusmeters) logging started 2021-08-07 09:00:05 using makepkg_1.3.0-50-g2c83c51
[2021-08-07_09:00:05] Started config amb8465[00000000] on /dev/ttyACM0 listening on c1,t1
[2021-08-07_09:00:07] Started config cul on /dev/ttyACM1 listening on t1
[2[2021-08-07_09:00:08] Started config cul on /dev/ttyACM2 listening on c1
[2021-08-07_09:00:08] (wmbusmeters) waiting for telegrams

$ grep device= /etc/wmbusmeters.conf

device=/dev/ttyACM0:amb8465:c1,t1
device=/dev/ttyACM1:cul:t1
device=/dev/ttyACM2:cul:c1

$ lsusb|grep -i atmel
Bus 001 Device 011: ID 03eb:204b Atmel Corp. LUFA USB to Serial Adapter Project
Bus 001 Device 007: ID 03eb:204b Atmel Corp. LUFA USB to Serial Adapter Project
Bus 001 Device 005: ID 03eb:204b Atmel Corp. LUFA USB to Serial Adapter Project
```

weetmuts, 2021-08-08:
Thanks! Can you run with --debug so that we can see the communication with the dongle?

antcroissant, 2021-08-08:
Sure, I attached a debug logfile. I removed the two CUL devices from my config and did a "systemctl start wmbusmeters.service" and after some time a "killall -HUP wmbusmetersd". Thanks for looking into this!

[debug.log](https://github.com/weetmuts/wmbusmeters/files/6951255/debug.log)

weetmuts, 2021-08-08:
Thanks! That was weird. I have pushed a fix with extra debug information and
if it fails to send (which is very weird since the tty could be opened.) then
it will retry to send the detection command again in 100ms, and repeat this 3 times and
only then give up.

Please test and post the debug here again!

antcroissant, 2021-08-08:
Sorry, does not work. Same as before, i have to send "killall -HUP wmbusmetersd" to get him going. I attached the new debug file...

[debug.log](https://github.com/weetmuts/wmbusmeters/files/6951416/debug.log).

weetmuts, 2021-08-08:
Thanks! It seems like the detection code is overwhelmed with valid telegrams does not find the expected response within the size/time limit. 

I have pushed a fix please test!

Now it will read up to three times and wait 100 ms between each read for more data until the expected response should appear.

antcroissant, 2021-08-09:
[https://github.com/weetmuts/wmbusmeters/commit/5793d7f5b75b03bc8b4ca98500f63c112d27cff1](https://github.com/weetmuts/wmbusmeters/commit/5793d7f5b75b03bc8b4ca98500f63c112d27cff1) Works like charm! Thank you very much!

closed 2021-08-09 by antcroissant