wmbusmeters/wmbusmeters #320: apator162 - would there be a more flexible way to point out the offset?

to-masz, 2021-08-09:
I have bought the brand new apator AT-WMBUS-16-2 radio module on my own and after its activation I am able to receive the telegrams using nanocul and wmbusmeters 👌 

However, I have yet another pattern in the telegrams. Maybe it is because it is brand new and was not configured by any company 🤔 

```
0F 3307D8 9509020040 B2D2B5F0F4 F341555942 5850 0C00 F00101000010 E700 0000 FFFFFFFFFFFFFFFFFFFFFFF877
... a while after
0F 4167D8 9529020040 B4C0C4F0F3 F341555942 EB76 0C00 F00101000010 0E13 0000 FFFFFFFFFFFFFFFFFFFFFF49F9
0F 4467D8 9529020040 B4C0C2F0F3 F341555942 1377 0C00 F00101000010 0E13 0000 FFFFFFFFFFFFFFFFFFFFFF30C8
0F 4567D8 9529020040 B6C0BCF0F3 F341555942 2177 0C00 F00101000010 0E13 0000 FFFFFFFFFFFFFFFFFFFFFFBEEB
... few days after
0F 3068F8 9529020040 C7C1B3F0F3 F341555942 53C2 0C00 F00101000010 5714 0000 FFFFFFFFFFFFFFFFFFFFFFE883
0F 3168F8 9529020040 C7C1B3F0F3 F341555942 61C2 0C00 F00101000010 5714 0000 FFFFFFFFFFFFFFFFFFFFFF88C2
0F 3368F8 9529020040 C6C1B3F0F3 F341555942 7BC2 0C00 F00101000010 5714 0000 FFFFFFFFFFFFFFFFFFFFFFAC2B
// running some water
0F 3568F8 9529020040 C6C1B2F0F3 F341555942 96C2 0C00 F00101000010 5C14 0000 FFFFFFFFFFFFFFFFFFFFFF699F
0F 3668F8 9529020040 B2D3B3F0F4 F341555942 A6C2 0C00 F00101000010 5C14 0000 FFFFFFFFFFFFFFFFFFFFFF472D
0F 4468F8 9529020040 B2D3B3F0F4 F341555942 2BC3 0C00 F00101000010 5C14 0000 FFFFFFFFFFFFFFFFFFFFFF3657
```

As you can see the offset is 32 (2f2f + 0f + 29 other bytes). 

Would it be possible to add a configuration option for `MeterInfo` that everyone could override the offset on their own in meter file (if needed)?

weetmuts, 2021-08-15:
It would be pretty easy. But the problem is that the offset moves around! :-/
They meters have some sort of internal protocol which has not been decoded.
Can you modify the detection code to see if you can add this new offset and example?

to-masz, 2021-08-18:
So I was trying to reach out to the company, but without a big success. The main thing I have learnt is that the custom protocol can be somehow defined by the waterworks operator. Therefore frame / telegram schema is somehow dynamic - but it should not change without radio module reconfiguration. 

I am super worried about adding new detection guess as I cannot find previous telegrams to test against them. We don't have tests for that here, do we? 

Therefore my idea of using the `extras` attribute of the meter info to explicitly provide the offset. 
```
name=MainWatermeter
driver=apator162 
id=01234567
key=00000000000000000000000000000000
extras=offset=32
```

It would require:
* Adding a support for `parseMeterConfig` to pass the `extras=` content into `MeterInfo` constructor instead of `""`. Also I would prefer having wrapping extras in quotes (`extras="offset=32 foo=bar"` as it looks better 😅 ) and with adding such code we have a possibility to do it.
* In apator driver calling `parseExtras` in constructor (?) and set new `offset` attribute
* If `offset` set then skip guessing and use it right away 

@weetmuts WDYT?

weetmuts, 2021-08-19:
There is a whole test script for the apators. :-) If you run `make test`
it will run `./tests/test_apas.sh build/wmbusmeters`
(or `make testd` to test build_debug/wmbusmeters if you have built `make DEBUG=true`
which is recommended for development.)

I think we have experienced that an apator162 has been working fine for months and then suddenly the offset changed!
So unfortunately it is not that super stable as one would hope.... but, in lack of better knowledge I approve
of adding a manual override.....

I have already anticipated sending extra parameters to a meter driver. :-)
It works in the same way that we can send extra parameters to a wmbus device:
for example `device=rtlwmbus(ppm=17)` to adjust the rtlwmbus device. 

There is already parsing support in MeterInfo::parse to grab the extras from a meter driver string, so
feeding wmbusmeters the driver:
`driver=apator162(offset=32)`
would store `offset=32` into MeterInfo::extras.

You can now add code to the Apator162 constructor that extracts the offset using 
parseExtras like wmbus_rtlwmbus is doing it, and store the override offset into the meter object.
Then skip the whole guessing game if you have an override. :-)

Then add code to testinternals.cc:
test_meters()
{
...
    testm("apator162(offset=162)", true,
          "apator162", // driver                                                                                                                
          "", // extras                                                                                                                      
          "", // bus                                                                                                                     
          "", // bps                                                                                                                     
          "", // linkmodes 
         "offset=32"); // extras
...
}

and add your new telegram with override to tests/test_apas.sh
MyTapWaterOverride 'apator162(offset=32)'  12345678 NOKEY
and to simulations/simulation_apas.txt

weetmuts, 2021-09-21:
Hi! @to-masz I have just improved the code for decoding the apator162. Now your hardcoded offset should not be needed any more since wmbusmeters somewhat understands the framing of the telegram content. The offset extra option will now be ignored by wmbusmeters. Please test the latest master to check that it produces the right values.

weetmuts, 2021-09-21:
You might want to run the latest wmbusmeters with --debug on some of your telegrams, you will something like:
(apator162) 11: 0F manufacturer specific data 86B4B89529020040C6C1B4F0F3F341555942FA701000F00101000010BC780000FFFFFFFFFFFFFFFFFFFFFF2483
(apator162) 11: *** 0F|86B4B8
(apator162) 15: *** 95|290200
(apator162) 19: *** 40|C6C1
(apator162) 1c: *** B4|F0F3F3
(apator162) 20: *** 41|5559
(apator162) 23: *** 42|FA701000
(apator162) 28: *** F0|01010000
(apator162) 2d: *** 10|bc780000 total consumption (30.908000 m3)

This means that we can now group the content of the apator data into registers. 
However right now only the register 0x10 is known. But if you have the time, 
perhaps you can take a look at the other registers and see if you can correlate 
them with something in your setup.

archont00, 2024-07-07:
> I have bought the brand new apator AT-WMBUS-16-2 radio module on my own and after its activation I am able to receive the telegrams using nanocul and wmbusmeters

@to-masz 
Hi Tomasz,

Would you mind sharing how you obtained the encryption key?

I have 6 water meters with AT-WMBUS-16-2 (HW: 3, FW: 2) and if it helps I may consider buying the vendor's reading device & licensed software for a one-off use - **only if it provides the keys**, obviously. (Assuming the EWM software is able to derive the key from the serial number of the meter.)

to-masz, 2024-07-28:
@archont00 by default it is all zeros