wmbusmeters/wmbusmeters #321: Add support for Honeywell TPR11 Merlin 169 WmBus

tonioa, 2021-08-09:
Hello,
Firstly thanks for your work on that.

I am trying to catch my French Water Meter reference  Honeywell TPR11 Merlin 169 WmBus
Seem the frequency is 169Mhz

i have try all of this command: (note rtl_sdr and rtl-wmbus is installed)
wmbusmeters auto:c1
wmbusmeters --debug --t1 rtlwmbus

but i didn't receive nothing do you know if it compatible ?
[https://ibb.co/2WQBDCZ](url)

Thanks for your work and your time

Anthony

weetmuts, 2021-08-09:
You need to listen to 169MHz. You can try: wmbusmeters rtlwmbus:169M
(Perhaps tune it using ppm, see the README.)

And you can try the rtl_wmbus command by itself:
rtl_sdr -f 169M -s 1600000 - 2>/dev/null | build/rtl_wmbus
See: https://github.com/xaelsouth/rtl-wmbus

tonioa, 2021-08-10:
Thanks for your reply.

I have the 2 commands i only catch this

T1;0;0;2021-08-09 19:12:25.000;14;14;00000000;0x

I will continue and try tuning with ppm

merbanan, 2021-08-10:
169MHz uses mode N which is not supported by rtl-wmbus. Record some signal samples and post them here. Then we can add support via rtl_433 and maybe rtl-wmbus in the future. You can record with rtl_433 with rtl_433 -f 169M -S unknown.

Info about mode N can be found here:
https://radiocrafts.com/technologies/wireless-m-bus-technology-overview/

closed 2021-09-25 by weetmuts

Cissoubaka, 2023-11-03:
Hi, I have this cold water meter, I wait until my supplier change my meter (HS, after 6 months, still at 0) and then I'll post some record.