wmbusmeters/wmbusmeters #322: Build fails with "wmbus.cc:4157:39: error: ‘FUZZING’ was not declared in this scope"

antcroissant, 2021-08-09:
Hi,..

current build fails with errors (at least on archlinux). For me the last working one is [https://github.com/weetmuts/wmbusmeters/commit/5793d7f5b75b03bc8b4ca98500f63c112d27cff1](https://github.com/weetmuts/wmbusmeters/commit/5793d7f5b75b03bc8b4ca98500f63c112d27cff1), next one [https://github.com/weetmuts/wmbusmeters/commit/838f89d576bd5b39fe0873b4b8a6ad54dc30033a](https://github.com/weetmuts/wmbusmeters/commit/838f89d576bd5b39fe0873b4b8a6ad54dc30033a) fails with:

```
[...]

src/wmbus.cc: In member function ‘bool Telegram::parseELL(std::vector<unsigned char>::iterator&)’:
src/wmbus.cc:989:37: error: ‘FUZZING’ was not declared in this scope
  989 |         if (ell_pl_crc != check && !FUZZING)
      |                                     ^~~~~~~
src/wmbus.cc: In function ‘bool trimCRCsFrameFormatA(std::vector<unsigned char>&)’:
src/wmbus.cc:4066:35: error: ‘FUZZING’ was not declared in this scope
 4066 |     if (calc_crc != check_crc && !FUZZING)
      |                                   ^~~~~~~
src/wmbus.cc:4080:39: error: ‘FUZZING’ was not declared in this scope
 4080 |         if (calc_crc != check_crc && !FUZZING)
      |                                       ^~~~~~~
src/wmbus.cc:4096:39: error: ‘FUZZING’ was not declared in this scope
 4096 |         if (calc_crc != check_crc && !FUZZING)
      |                                       ^~~~~~~
src/wmbus.cc: In function ‘bool trimCRCsFrameFormatB(std::vector<unsigned char>&)’:
src/wmbus.cc:4143:35: error: ‘FUZZING’ was not declared in this scope
 4143 |     if (calc_crc != check_crc && !FUZZING)
      |                                   ^~~~~~~
src/wmbus.cc:4157:39: error: ‘FUZZING’ was not declared in this scope
 4157 |         if (calc_crc != check_crc && !FUZZING)
      |                                       ^~~~~~~
make: *** [Makefile:106: build/wmbus.o] Error 1
make: *** Waiting for unfinished jobs....
```

weetmuts, 2021-08-09:
Can you check that the Makefile is updated? It should pass -DFUZZING=false
for your build.

antcroissant, 2021-08-09:
Yes, looks like it does.
```
[...]

$(info Building $(VERSION))

FUZZFLAGS ?= -DFUZZING=false
CXXFLAGS ?= $(DEBUG_FLAGS) $(FUZZFLAGS) -fPIC -std=c++11 -Wall -Werror=format-security
CXXFLAGS += -I$(BUILD)
LDFLAGS  ?= $(DEBUG_LDFLAGS)

[...]
```

weetmuts, 2021-08-09:
What does the make print when compiling, it should be something like this:
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Ibuild src/bus.cc -MMD -c -o build/bus.o

antcroissant, 2021-08-09:
I just did a manual build (./configure && make), which works fine.
e.g:
```
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Ibuild src/meter_mkradio3.cc -c -E > build/meter_mkradio3.o.src
```
But trying to build a package in a clean chroot on arch (makechrootpkg -c -n -r $CHROOT) gives me something else.
e.g.:
```
g++ -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -Ibuild src/aes.cc -c -E > build/aes.o.src
```
Maybe i have to check my build env...

antcroissant, 2021-08-09:
Ok, got it. Had to exclude the default arch specific buildflags in the PKGBUILD file. Sorry, my bad.

closed 2021-08-09 by antcroissant