wmbusmeters/wmbusmeters #326: No readings and huge CPU usage

bensaeba, 2021-08-14:
Hello

I'm facing a new issue.
Everything was working fine until yesterday. But I have to re install everything on my raspberry because I'm now booting on a ssd instead of a sd card.
So, I went in all the process of installing wmbusmeters (git clone, ..configure, make, make install, etc).
But, on the contrary of what it was doing until now I noticed some issues:
- log file was "corrupted". I was unable to open it. I had to delete it and recreate it
- When I re created all my meters, nothing appears in the reading folder. 
- stop the service through sudo systemctl stop wmbusmeters to a very long time
- CPU usage is around 30% (vs 1% usually) so temperature increase a lot! I have to stop the service for the CPU to slowdown.
- If I try something line: `wmbusmeters amb8465:t1 MyTapWater auto 20addec4 NOKEY `==> nothing
- doing wmbusmeters auto:t1 is working on the contrary e-g:
```
wmbusmeters auto:t1
Started auto amb8465[00000000] on /dev/ttyUSB0 listening on t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 20b841aa
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: amb8465[00000000]
                  rssi: -21 dBm
                driver: izar
Received telegram from: 20e536a0
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
```

I put my config file into debug. Here is what I got:
```

(wmbusmeters) logging started 2021-08-14 12:15:40 using 1.4.0-3-g8dd3e87
(wmbusmeters) logging started 2021-08-14 12:23:01 using 1.4.0-3-g8dd3e87
(wmbusmeters) version: 1.4.0-3-g8dd3e87
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 4 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 1
(serial) stopping manager
(serial) event loop stopped!
(wmbusmeters) logging started 2021-08-14 12:26:18 using 1.4.0-3-g8dd3e87
(wmbusmeters) version: 1.4.0-3-g8dd3e87
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) number of meters: 1
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 4 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 1

```
I don't know why it says /dev/ttyAMA0 
Because I got:
```
pi:~ $ ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 août  14 12:26 /dev/ttyUSB0
```

Status:
  ```
 Active: active (running) since Sat 2021-08-14 12:26:18 CEST; 1s ago
     Docs: https://github.com/weetmuts/wmbusmeters
           man:wmbusmeters(1)
  Process: 4979 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (
  Process: 4981 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbu
  Process: 4985 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status
  Process: 4986 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmet
  Process: 4987 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pi
 Main PID: 4988 (wmbusmetersd)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/wmbusmeters.service
           └─4988 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid

août 14 12:26:18 benpi wmbusmetersd[4988]: (config) "meterfiles" "/var/log/wmbus
août 14 12:26:18 benpi wmbusmetersd[4988]: (config) "meterfilesaction" "overwrit
août 14 12:26:18 benpi wmbusmetersd[4988]: (config) "logfile" "/var/log/wmbusmet
août 14 12:26:18 benpi wmbusmetersd[4988]: (config) "" ""
août 14 12:26:18 benpi wmbusmetersd[4988]: (config) loading meter file /etc/wmbu
août 14 12:26:18 benpi wmbusmetersd[4988]: (config) name=MyTapWater
août 14 12:26:18 benpi wmbusmetersd[4988]: (config) id=20adee41
août 14 12:26:18 benpi wmbusmetersd[4988]: (config) key=<notprinted>
lines 1-23
```

So, today I tried to uninstall and re install everything.
I created my meters and it worked fine (meter readings, etc) and I coudl re create my 4 meters and have their readings. As it was working I changed config file to send data through mqtt to domoticz. 
I don't know if it's related but by the time I installed mosquitto client, everything "crashed" again.

So, I went again in the process of uninstall and install but back to yesterday: I got no meter reading, huge CPU usage and stopping the service takes a lot of time (cf debug log).
By the way, I noticed that le log file is not created during the process on installation but right after starting the service for the first time.

weetmuts, 2021-08-14:
Strange. Can you add donotprobe=/dev/ttyAMA0
to prevent wmbusmeters from probing this tty. 

Can you reproduce the problem where:
wmbusmeters --debug --donotprobe=/dev/ttyAMA0 amb8465:t1 MyTapWater auto 20addec4 NOKEY
is notworking
but 
wmbusmeters --debug --donotprobe=/dev/ttyAMA0 auto:t1 MyTapWater auto 20addec4 NOKEY
is working?

If so, post the debug logs here.

bensaeba, 2021-08-14:
Hi,

I tried.
So, you will find the debug log here under (sorry it's very long) :

when doing:
`wmbusmeters --debug --donotprobe=/dev/ttyAMA0 amb8465:t1 MyTapWater auto 20addec4 NOKEY`

I got:
```
$ wmbusmeters --debug --donotprobe=/dev/ttyAMA0 amb8465:t1 MyTapWater auto 20addec4 NOKEY
(wmbusmeters) version: 1.4.0-3-g8dd3e87
(config) using device: amb8465:t1 
(config) number of meters: 1
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input(ICRNL|IXON) output(ONLCR|OPOST) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHO|ECHOCTL|ECHOE|ECHOK|ECHOKE|ICANON|IEXTEN|ISIG) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 1
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFF"
(amb8465) not enough data for decoding ConfigAMB8465!
(amb8465) reading response... 2
(serial) received binary "FFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF17"
(amb8465) proprely decoded ConfigAMB8465 response. Content: id=00000000 media=00 version=00 c_field=44 auto_rssi=00
(amb8465) found response at offset 0
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(amb8465) detect id=00000000 media=00 version=00 c_field=44 auto_rssi=00
(serial) EVENT thread interrupted
(amb8465) are you there? yes 00000000
(main) found specified device (amb8465:t1) that matches detected device (/dev/ttyUSB0:amb8465[00000000]:9600/0)
(main) opening amb8465:t1
Started config amb8465[00000000] on /dev/ttyUSB0 listening on t1
(amb8465) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (amb8465)
(amb8465) get config
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "FF"
(serial) received binary "8A7A00780080710200000000FFFFFA00"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00"
(amb8465) not enough bytes yet, partial command response 17 126.
(serial) received binary "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 32 126.
(serial) received binary "FF003200021400FFFFFFFFFF01000400"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF01000400"
(amb8465) not enough bytes yet, partial command response 48 126.
(serial) received binary "0000FFFFFF01440000000000000000"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000"
(amb8465) not enough bytes yet, partial command response 63 126.
(serial) received binary "FFFF0B040100FFFFFFFFFF00030000"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000"
(amb8465) not enough bytes yet, partial command response 78 126.
(serial) received binary "FFFFFFFFFFFFFF0000FFFFFFFFFFFFFF"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFF"
[... 730 lines cut ...]
```
and it seems to be working just fine again.
so the key here is:  donotprobe=/dev/ttyAMA0

I just do not know why, because between today and yesterday the only difference is that I'm booting on ssd en /dev/sda

and I got this when  `sudo udevadm info --query=all --name=ttyAMA0`

```
pi@benpi:~ $ sudo udevadm info --query=all --name=ttyAMA0
P: /devices/platform/soc/fe201000.serial/tty/ttyAMA0
N: ttyAMA0
L: 0
S: serial1
E: DEVPATH=/devices/platform/soc/fe201000.serial/tty/ttyAMA0
E: DEVNAME=/dev/ttyAMA0
E: MAJOR=204
E: MINOR=64
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=5113578
E: DEVLINKS=/dev/serial1
E: TAGS=:systemd:
```

I do not know what's on this...

weetmuts, 2021-08-15:
I think it is connected to bluetooth on the raspberry pi 3.
However, I have no idea why it causes the whole raspberry to go down the drain,
when you send a few random bytes to it....

Can you try to disable the bluetooth adapther and see if it stops bringin down the ship?
That would be valuable to know. Then if there is a way we can check if bluetooth is enabled
and we are on raspberry pi 3, then wmbusmeters could skip the /dev/ttyAMA0

bensaeba, 2021-08-16:
Hi,
I'm on Pi 4. I have no bluetooth adaptater and the buit-in bluetooth is always disable. In fact, I have never enable it.

So, I tried to enable it and make `wmbusmeters --debug auto:t1 MyTapWater auto 20addec4 NOKEY`
but the result is the same
```
wmbusmeters --debug auto:t1 MyTapWater auto 20addec4 NOKEY
(wmbusmeters) version: 1.4.0-3-g8dd3e87
(config) number of meters: 1
(main) device /dev/ttyACM0 not currently used, detect contents...
(serial) EVENT thread interrupted
(amb8465) could not open tty for detection
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyACM0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B3000000) input() output(ONLCR) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHOCTL|ECHOE|ECHOK|ECHOKE) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 7 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 1
```

weetmuts, 2021-08-19:
Can you try (without running wmbusmeters) to trigger the high cpu usage by just doing:
echo "sdsdfsdfsdfsdf....sdfsdfsfd"  > /dev/ttyAMA0
or
cat > /dev/ttyAMA0 # Type a lot of random characters
or 
cat < /dev/random > /dev/ttyAMA0 # break with ctrl-c pretty quick...

BIBOLV, 2021-08-19:
Can you confirm that same behavior is noticed when booting from SD? And how SSD is attached to PI? My question is might it be somehow related to implementation of SSD connection and sending data to serial port triggers some weird behavior.

bensaeba, 2021-08-23:
Hi,

By doing `echo "sdsdfsdfsdfsdf....sdfsdfsfd" > /dev/ttyAMA0`  I also got a high CPU usage ==> When doing `top` , bash command line is 100%.

My SSD is connected on the USB3 port of the Pi. By booting with the HDMI cable I noticed this message: 
`mmc1 : controller never released inhibit bit(s)`. I don't know if it's related. First time noticing this since I usually boot without HMDI and so far I have managed boot (even with this message)

weetmuts, 2021-08-24:
> By doing `echo "sdsdfsdfsdfsdf....sdfsdfsfd" > /dev/ttyAMA0` I also got a high CPU usage ==> When doing `top` , bash command line is 100%.

Can you try to disable bluetooth and to the test again, to see if you do not get high CPU usage?
https://di-marco.net/blog/it/2020-04-18-tips-disabling_bluetooth_on_raspberry_pi/

If the bluetooth is indeed the problem we will have to add detection of a bluetooth enabled ttyAMA0  and avoid
probing if this is the case.

bensaeba, 2021-09-05:
Hi @BIBOLV ,

I can tell that when I was booting on the SD card, I did not have this issue.
My SSD is connected on the USB3 port of the PI.
I will try to re-use my SD card and see what happens then

@weetmuts , sorry I did not find the time to try your solution during the past days. I will try this week.

weetmuts, 2021-09-25:
@bensaeba @BIBOLV Ok, it seems like the culprit is /dev/ttyAMA0 on some raspberry pis. Writing some random data to this serial stops the computer! Wee! Security! :-)

This will probably be a recurring problem. I have opened a new ticket
https://github.com/weetmuts/wmbusmeters/issues/345

Anything else here @bensaeba that is not yet resolved? Can we close this?

bensaeba, 2021-09-27:
Hi,
Sorry for the delay.
I tried what you suggested ia disabling bluethooth
Open` /boot/config.txt file Permalink`
`sudo nano /boot/config.txt`
Add below, save and close the file:
```
# Disable Bluetooth
dtoverlay=disable-bt
```
Disable related services 
```
sudo systemctl disable hciuart.service
sudo systemctl disable bluealsa.service
sudo systemctl disable bluetooth.service
```

and I comment #donotprobe=/dev/ttyAMA0 in config file
```
loglevel=debug
device=amb8465:t1
#donotprobe=/dev/ttyAMA0
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
```

Well, no high CPU usage and I got my meter readings.

However, if I try:

`wmbusmeters --debug auto:t1 MyTapWater auto 20addec4 NOKEY`

I got :
```
pi@benpi:~ $ wmbusmeters --debug auto:t1 MyTapWater auto 20addec4 NOKEY
(wmbusmeters) version: 1.4.0-3-g8dd3e87
(config) number of meters: 1
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(amb8465) could not open tty for detection
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(amb8465) could not open tty for detection
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 1521 exited
Notice! Wmbusmeters daemon (pid 1470) is running and it might hog any wmbus devices.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>1489
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 1522 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
^X^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 4 devices
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(serialtty) closed /dev/ttyUSB0 (detect cul)
(serial) closing 4 devices
```

And if I try:
`wmbusmeters --debug --donotprobe=/dev/ttyAMA0 auto:t1 MyTapWater auto 20addec4 NOKEY`

I got:

```
pi@benpi:~ $ wmbusmeters --debug --donotprobe=/dev/ttyAMA0 auto:t1 MyTapWater auto 20addec4 NOKEY
(wmbusmeters) version: 1.4.0-3-g8dd3e87
(config) number of meters: 1
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(amb8465) could not open tty for detection
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>>1470
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 1676 exited
Notice! Wmbusmeters daemon (pid 1470) is running and it might hog any wmbus devices.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>1672
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 1677 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
^C(serial) closing 4 devices
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(serialtty) closed /dev/ttyUSB0 (detect cul)
(serial) closing 4 devices
```