wmbusmeters/wmbusmeters #330: Wmbusmeters gets only one reading then it switches its status to failed

Nikodemski2002, 2021-08-29:
Hi. I have a problem after installing wmbusmeters on a new device. As in the title, I get only one reading the service status switches to failed. This is my config file:
```
loglevel=debug
device=/dev/ttyUSB0:im871a:c1
donotprobe=/dev/ttyUSB0
logtelegrams=false
format=fields
selectfields=id,volume_flow_m3h,t1_temperature_c,t2_temperature_c,timestamp
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
meterfilestimestamp=micros
logfile=/var/log/wmbusmeters/wmbusmeters.log
ignoreduplicates=false
```

This is what  I get after typing: systemctl status wmbusmeters
```
× wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/usr/lib/systemd/system/wmbusmeters.service; disabled; vendor preset: disabled)
     Active: failed (Result: signal) since Tue 2021-08-24 23:45:30 CEST; 4 days ago
       Docs: https://github.com/weetmuts/wmbusmeters
             man:wmbusmeters(1)
    Process: 1069 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 1070 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 1071 (code=killed, signal=ABRT)
        CPU: 240ms

Aug 24 23:43:40 4GEN-LUBAN wmbusmetersd[1071]: (config) "" ""
Aug 24 23:43:40 4GEN-LUBAN wmbusmetersd[1071]: (config) loading meter file /etc/wmbusmeters.d/79806477
Aug 24 23:43:40 4GEN-LUBAN wmbusmetersd[1071]: (config) name=79806477
Aug 24 23:43:40 4GEN-LUBAN wmbusmetersd[1071]: (config) type=multical803:c1
Aug 24 23:43:40 4GEN-LUBAN wmbusmetersd[1071]: (config) id=79806477
Aug 24 23:43:40 4GEN-LUBAN wmbusmetersd[1071]: (config) key=<notprinted>
Aug 24 23:43:40 4GEN-LUBAN wmbusmetersd[1071]: (cmdline) setting link modes to c1 for meter 79806477
Aug 24 23:43:40 4GEN-LUBAN wmbusmetersd[1071]: (wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log
Aug 24 23:45:30 4GEN-LUBAN systemd[
```
This what I get after typing wmbusmeters --debug --donotprobe=/dev/ttyUSB0 im871a:c1 79806477 multical803 79806477 [ENCRYPTION_KEY]
```
(wmbusmeters) version: 1.1.0
(config) using device: im871a:c1
(config) number of meters: 1
(main) configured "79806477" "multical803" "79806477" encrypted
(main) device /dev/ttyS0 not currently used, detect contents...
(serial) before config: /dev/ttyS0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) EVENT thread interrupted
(serialtty) opened /dev/ttyS0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyS0) sent "FF0A02008077"
(serial /dev/ttyS0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyS0 (detect amb8465)
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyS0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a) are you there? no.
(serialtty) closed /dev/ttyS0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (detect rc1180)
(serial /dev/ttyS0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyS0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyS0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (detect cul)
(serial /dev/ttyS0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyS0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyS0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyS0 (detect cul)
(cul) are you there? no
(main) ignoring /dev/ttyS0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyS1 not currently used, detect contents...
(serial) before config: /dev/ttyS1 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS1 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS1 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyS1) sent "FF0A02008077"
(serial /dev/ttyS1) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyS1 (detect amb8465)
(serial) EVENT thread interrupted
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS1 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS1 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyS1) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a) are you there? no.
(serialtty) closed /dev/ttyS1 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS1 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS1 fd 3 (detect rc1180)
(serial /dev/ttyS1) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyS1) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyS1 (detect rc1180)
(serial) EVENT thread interrupted
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS1 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS1 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS1 fd 3 (detect cul)
(serial /dev/ttyS1) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyS1) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyS1) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyS1 (detect cul)
(cul) are you there? no
(main) ignoring /dev/ttyS1, it does not respond as any of the supported wmbus devices.
(serial) EVENT thread interrupted
(main) device /dev/ttyS2 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS2 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS2 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyS2) sent "FF0A02008077"
(serial /dev/ttyS2) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
```
I don't have a problem with high CPU usage as in the #326  thread. Also, Linux kernel doesn't mention the wmbusmeters.service in its logs. I tried to solve this problem on my own by studying other issues, but I've failed. 

P.S.
The whole service works perfectly on old raspberry with raspbian lite OS, now I'm using a normal PC with Fedora 34.

Best regards.
Nikodem

weetmuts, 2021-08-29:
It says the daemon was killed by sigabort. Perhaps there is a bug where wmbusmeters trigger libc to kill it off....
Are you running the fedora package?
If so, can you try the latest github version?

weetmuts, 2021-09-25:
Ping?

closed 2022-02-26 by weetmuts