wmbusmeters/wmbusmeters #332: Does wmbusmeters detect 923 Mhz Kamstrup FlowIQ 2102 (Multical 21)?

fc0ta, 2021-08-31:
Hello, I would like to know if it is possible that wmbusmeters detect a 923Mhz Kamstrup FlowIQ 2102 Water Meter.
I have left the dongle 24 hours without any response.
I am using a rtl-sdr dongle.

The usage:  wmbusmeters rtlwmbus:923M:c1

Anyone knows if it doesn't detect any telegram because of the frequency?  

Thanks a lot, and any input would be of much help.

Regards

weetmuts, 2021-08-31:
I think it is better to ask this question over at: https://github.com/xaelsouth/rtl-wmbus
since he is the expert on radio reception. 

However, assuming that your meter (which is located in mexico or australia?) uses the same fsk protocol,
just a different carrier frequency, then you could try different base frequencies around 923. 

Like this:
rtl_sdr -f 923.1M -s 1600000 - 2>/dev/null | rtl_wmbus

Then try a lot of values around 923 MHz.

fc0ta, 2021-09-09:
Thanks for your answer Fredrik. I will try what you recommend.

closed 2021-09-25 by weetmuts