wmbusmeters/wmbusmeters #333: New Techem / Kamstrup heat meter mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x43) ver: 0x39

Boldfor, 2021-09-04:
**Setting:**
- **Hardware:** Raspi 3 with Nooelec NESDR Nano 2 TL-SDR-USB-Set (RTL2832U + R820T2)
- **OS:** Raspian OS 10
- **wmbusmeters:** 1.4.1-9-g4c1c677 on Docker/Portainer
- **openHAB** 3.1.0 Release Build on Docker/Portainer

**Situation:**
I have two meters in my flat (both of them appear very common meters here in Germany): One for cold water (Techem radio 4) and one for heating/warm water (Techem / Kamstrup vario 3 type 3.2.1). The radio 4 works brilliantly (feeds information via mqtt to Openhab). The vario 3, however, does not work.

I'm using the following **/etc/wmbusmeters.conf**
```
loglevel=normal
device=rtlwmbus(ppm=17)
listento=c1,t1,s1
logtelegrams=true
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
logfile=/wmbusmeters_data/logs/wmbusmeters.log
meterfilesnaming=name
meterfilestimestamp=day
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
ignoreduplicates=false
```

**Problem description:**
With no meters configured the respective meter is recorded the following in /logs/wmbusmeters.log:
```
Received telegram from: 67552079
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0x43)
                   ver: 0x39
                device: rtlwmbus[stx:1090:-9]
                  rssi: 92 dBm
                driver: unknown!
telegram=||37446850792055673943A2109F29145600C8961A000000A2C11A6B8C11081E67E831D6597D7957218A31DE4EBBF5B6C63D2E71CE61654924|+0
Received telegram from: 67552079
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0x43)
                   ver: 0x39
                device: rtlwmbus[stx:1090:-9]
                  rssi: 125 dBm
                driver: unknown!
telegram=||37446850792055673943A2109F29145600C8961A000000A2C11A6B8C11081E67E831D6597D7957218A31DE4EBBF5B6C63D2E71CE61654924|+0
```

Configuring this meter via the following config file in /wmbusmeters.d/warmwasserzaehler ...
```
name=warmwasserzaehler
id=67552079
```
... leads to the following entries in /logs/wmbusmeters.log...
```
Started config rtlwmbus[stx:1090:-9](ppm=17) listening on c1,s1,t1
(meter) warmwasserzaehler: meter detection could not find driver for id: 67552079 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x43) ver: 0x39
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```

... and the following readings in the folder /meter_readings/
`{"media":"heat","meter":"auto","name":"warmwasserzaehler","id":"67552079","meter_info":"","timestamp":"2021-09-04T22:58:30Z","device":"rtlwmbus[stx:1090:-9]","rssi_dbm":122}`

I stumbled upon this issue (https://github.com/weetmuts/wmbusmeters/issues/263), which appears to deal with a similar meter and problem, but could not find a solution in the thread.

Following the wmbusmeters documentation (HowToAddaNewMeter.txt), I am supposed to know the `<yourkey>`-entry already, correct? So what do I do to get the key?

Thanks a lot in advance.

Superb project btw!

weetmuts, 2021-09-09:
Thank you for your excellent description of the problem!

I put the next line (your telegram) into a file simulation_techem.txt: telegram=|37446850792055673943A2109F29145600C8961A000000A2C11A6B8C11081E67E831D6597D7957218A31DE4EBBF5B6C63D2E71CE61654924|
and ran:
wmbusmeters --debug simulation_techem.txt Vatten mkradio3 67552079 NOKEY

You can see that the debug output prints:
(mkradio4) 0a: a2 tpl-ci-field (Mfct specific)
this means that all bytes after the first A2 are proprietary and not conformant to the wmbus standard. :-(

I tried with the other drivers, mkradio3 mkradio4 and fhkvdataiii
but I do not know if either of the outputs seems sensible to you? Please test!

If one of the drivers work, then you can use that driver explicitly, and I can add the meter version/type to the
database so that the lookup is automatic in the future.

If none of the drivers work, then you have to collect telegrams using --logtelegrams First running water through the meter for a couple of minutes, then stop it for a couple of minutes and then start again.

This way, it should be possible to figure out which bytes are transmitting the volume consumed.

Boldfor, 2021-09-10:
So I managed to dig a bit deeper into the problem. **Spoiler:** It seems to work already, just the mapping and formatting seems a bit off, so this should be solvable. :)

**Disclaimer:** I have not yet had the time to do the "time-series logging" ('running water through the meter' and 'letting time expire with no water running through'), to gather more data to figure out which bytes are transmitting the volume consumed. But from what I can see, maybe this is not even needed anymore.

So I did compare the actual readings **on the meter itself** with the readings **extracted from a recent telegram**...
`
telegram=||37446850792055673943A2109F29145600C8C61A000000A5C41A6B8C11081E67E831D6597D7957218A31DE4EBBF5B6C63D2E71CE61654924|`

... by **running them as a simulation** via...

`/wmbusmeters/wmbusmeters --debug /wmbusmeters_data/simulation_warmwasserzaehler.txt Vatten mkradio3 67552079 NOKEY`

... which resulted in the **following output**...

```
(wmbusmeters) version: 1.4.1-9-g4c1c677
(config) using device: /wmbusmeters_data/simulation_warmwasserzaehler.txt:any 
(config) number of meters: 1
(lookup) with file "/wmbusmeters_data/simulation_warmwasserzaehler.txt"
(lookup) driver: simulation file
(main) opening /wmbusmeters_data/simulation_warmwasserzaehler.txt:any
Started config simulation on /wmbusmeters_data/simulation_warmwasserzaehler.txt listening on any
(simulation) in /wmbusmeters_data/simulation_warmwasserzaehler.txt
(main) regular reset of simulation  on /wmbusmeters_data/simulation_warmwasserzaehler.txt will happen every 82800 seconds
(main) added /wmbusmeters_data/simulation_warmwasserzaehler.txt to files
(wmbus) no alarm (expected activity) for simulation
(main) /wmbusmeters_data/simulation_warmwasserzaehler.txt already configured as simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "37446850792055673943A2109F29145600C8C61A000000A5C41A6B8C11081E67E831D6597D7957218A31DE4EBBF5B6C63D2E71CE61654924"
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 56
(wmbus) parseELL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(meter) Vatten: for me? 67552079 in 
(meter) Vatten: meter detection did not match the selected driver mkradio3! correct driver is: unknown!
(meter) Not printing this warning agin for id: 67552079 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x43) ver: 0x39
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Vatten: yes for me
(meter) created "Vatten" "mkradio3" "67552079" encrypted
(meter) used meter template Vatten  mkradio3 to match 67552079
(meter) started meter 1 (Vatten 67552079 mkradio3)
(wmbus) parseDLL @0 56
(wmbus) parseELL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(meter) Vatten: for me? 67552079 in 67552079
(meter) Vatten: meter detection did not match the selected driver mkradio3! correct driver is: unknown!
(meter) Not printing this warning agin for id: 67552079 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x43) ver: 0x39
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Vatten: yes for me
(meter) Vatten mkradio3 handling telegram from 67552079
(meter) Vatten 67552079 "37446850792055673943A2109F29145600C8C61A000000A5C41A6B8C11081E67E831D6597D7957218A31DE4EBBF5B6C63D2E71CE61654924"
(wmbus) parseDLL @0 56
(telegram) DLL L=37 C=44 (from meter SND_NR) M=5068 (TCH) A=67552079 VER=39 TYPE=43 (Heat meter) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(telegram) TPL CI=a2
telegram=|37446850792055673943A2|109F29145600C8C61A000000A5C41A6B8C11081E67E831D6597D7957218A31DE4EBBF5B6C63D2E71CE61654924|+0
(mkradio3) 00: 37 length (55 bytes)
(mkradio3) 01: 44 dll-c (from meter SND_NR)
(mkradio3) 02: 6850 dll-mfct (TCH)
(mkradio3) 04: 79205567 dll-id (67552079)
(mkradio3) 08: 39 dll-version
(mkradio3) 09: 43 dll-type (Heat meter)
(mkradio3) 0a: a2 tpl-ci-field (Mfct specific)
(mkradio3) 0c: * 299f previous date (2020-12-31T02:00:00Z)
[... 11 lines cut ...]

... which I **summed up in the table below.** Note: Running the simulation I noticed that the outputs via mkradio3 driver (and _partially_ also via mkradio 4 driver) were more or less ok (more on this below). The outputs via the fhkvdataiii driver, however, resulted in gibberish, so I assume this can be ignored for now.

Here's my take on the **comparison "meter reading" vs. "wmbusmeter extraction"**:
[image]

**Some thoughts on the individual line items:**
1. Not sure whether the meter just submits a date which does not name sense to me (it's not the current date nor the last due date), or whether it was extracted wrongly and therefore does not make sense. But in any case, it is not the "current date" as labelled. My take: This one can be ignored, and instead # 14 should be the current date.
2. The numbers itself look good, just the UoM (m³ instead of kWh or MWh) and the formating looks off. First I thought "it's just the UoM that's off" (kWh instead of MWh), but even then, 2289 **k**Wh are not 22,89 **M**Wh.
3. Looks great. :)
4. Exactly the same as with # 2: UoM and number formatting looks off.
5. Does not seem to be transmitted by the meter.
6. Does not seem to be transmitted by the meter.
[... cut ...]
12. Does not seem to be transmitted by the meter.
13. Here it gets interesting: This value is not accessible on the meter itself, but since it appears to be the consumption since the last due date, it's precisely # 2 minus # 4.
14. This is funny: This is the current date / time (the one that should be # 1), ..., which leads to the question what # 1 actually is?

So much for now. Maybe this helps you to adjust the database? Would be really nice because, as said, this seems to be a really common meter here in Germany!

Thanks a lot!

Boldfor, 2021-10-03:
Quick update, I had some time to work on this today: Since the extraction does work _somehow_ with it being run via the mkradio3 driver being defined in /wmbusmeters_data/etc/wmbusmeters.d

```
name=abcde
id=12345
driver=mkradio3
```

and the most important value in the extraction being "heat energy consumed" (#2 above), it doesn't really matter that the extraction is not perfect. I transformed it via mqtt to openHAB and applied a simple

```
(function(i) {
    return parseFloat(i) / 100;
})(input)
```
to /openhab_conf/transform/warmwasserzaehler.js and activated it under Things --> Channel --> Profile (for those also working with openHAB):
[image]

Thanks again for this great project! :)

TheGreyDiamond, 2021-11-20:
So, has anyone continued to look into this? I have the same meter in our basement and it would be super cool to be able to read it out. Just setting the driver in the meter file doesn't work. So...?

Boldfor, 2021-11-20:
> Just setting the driver in the meter file doesn't work. So...?
I am also hoping someone finds the time to fix this at the source. However, in the meantime, my workaround described above (“ignoring the gibberish and dividing the relevant value by 100”) works perfectly well, at least for the most relevant value.

henkiejan1, 2021-12-14:
> 

Did you have a simple step by step instruction how to get it work? Thank you! :)

Boldfor, 2021-12-20:
> Did you have a simple step by step instruction how to get it work? Thank you! :)

Can you specify which step you'd want me to detail further? There are a number of different topics/areas involved, and I'd like to know which one you'd like to understand further, so that I'm not describing something which doesn't even fit your environment.

TheGreyDiamond, 2021-12-23:
So, I tried it using this config:
```
name=KellerHeiz
id=<Id>
driver=mkradio3
´´´
however, that didn't work. It resulted in it this output:
```
[2021-12-23_14:05:30] (meter) KellerHeiz: meter detection did not match the selected driver mkradio3! correct driver is: unknown!
(meter) Not printing this warning again for id: <Id> mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x04) ver: 0x17
[2021-12-23_14:05:30] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2021-12-23_14:05:30] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 12 (KellerHeiz <Id> mkradio3)
[2021-12-23_14:05:30] (meter) newly created meter (KellerHeiz <Id> mkradio3) did not handle telegram!
```

Is this me being too stupid to properly configure it or what happened?

Boldfor, 2021-12-23:
> It resulted in it this output:

I'm not a pro at all on wmbusmeters, so take my comments with caution. However, from what I can see, you're in a very similar setting tha I was, and maybe also getting the same output than I did, so you should also be getting similar entries recorded in /logs/wmbusmeters.log, right?

With those, you should be able to extract at least the most relevant information from the meter ("heat energy consumed").

henkiejan1, 2022-05-01:
I have just tried again something with the same meter. The Techem vario3 type 3.2.1 but can't get the same values as on the meter it self. Value 1 is on the meter in GJ Value 2 also in GJ. But on the meter of @Boldfor it's in m3? Here is a little bit from the log. I have configured the meter as:

`name=vario3321
id=67270509
driver=mkradio3
`

And some log output:

`New Note 2

(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-05-01 11:13:58.000;9;8;67283216;0x374468501632286739c3a2109f2b261302286039000082000076d81dd9679d346eb96bb15efb6b647ed8d9d6130000000000000044005d70>
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
(serial) received ascii "T1;1;1;2022-05-01 11:14:02.000;15;11;67270509;0x374468500905276739c3a2109f2bfa0202281e3b000082000070c0414aad9d466618b1aab13eab657a95b1a50a00000070400a003c30134d<0A>>
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-05-01 11:14:02.000;15;11;67270509;0x374468500905276739c3a2109f2bfa0202281e3b000082000070c0414aad9d466618b1aab13eab657a95b1a50a00000070400a003c3013>
(rtlwmbus) received full frame
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 56
(wmbus) parseELL @10 46
(wmbus) parseNWL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(meter) vario3321: for me? 67270509 in 67270509
[2022-05-01_13:14:02] (meter) vario3321: meter detection did not match the selected driver mkradio3! correct driver is: unknown!
(meter) Not printing this warning again for id: 67270509 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0xc3) ver: 0x39
[2022-05-01_13:14:02] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-05-01_13:14:02] (meter) to add support for this unknown mfct,media,version combination
(meter) vario3321: yes for me
(meter) created "vario3321" "mkradio3" "67270509" not-encrypted
(meter) used meter template vario3321 67270509 mkradio3 to match 67270509
(wmbusmeters) started meter 1 (vario3321 67270509 mkradio3)
(wmbus) parseDLL @0 56
(wmbus) parseELL @10 46
(wmbus) parseNWL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(meter) vario3321: for me? 67270509 in 67270509
[2022-05-01_13:14:02] (meter) vario3321: meter detection did not match the selected driver mkradio3! correct driver is: unknown!
(meter) Not printing this warning again for id: 67270509 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0xc3) ver: 0x39
[2022-05-01_13:14:02] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-05-01_13:14:02] (meter) to add support for this unknown mfct,media,version combination
(meter) vario3321: yes for me
(meter) vario3321 mkradio3 handling telegram from 67270509
(meter) vario3321 67270509 "374468500905276739C3A2109F2BFA0202281E3B000082000070C0414AAD9D466618B1AAB13EAB657A95B1A50A00000070400A003C30134D"
(wmbus) parseDLL @0 56
(telegram) DLL L=37 C=44 (from meter SND_NR) M=5068 (TCH) A=67270509 VER=39 TYPE=c3 (Heat meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=15
(wmbus) parseELL @10 46
(wmbus) parseNWL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(wmbus) parseTPL @10 46
(telegram) TPL CI=a2
(mkradio3) 000   : 37 length (55 bytes)
(mkradio3) 001   : 44 dll-c (from meter SND_NR)
(mkradio3) 002   : 6850 dll-mfct (TCH)
(mkradio3) 004   : 09052767 dll-id (67270509)
(mkradio3) 008   : 39 dll-version
(mkradio3) 009   : c3 dll-type (Heat meter)
(mkradio3) 010   : a2 tpl-ci-field (Mfct specific)
(mkradio3) 011 C?: 109F2BFA0202281E3B000082000070C0414AAD9D466618B1AAB13EAB657A95B1A50A00000070400A003C30134D mfct specific
(mkradio3) 057 C!: 2b9f previous date (2021-12-31T02:00:00Z)
(mkradio3) 059 C!: fa02 prev consumption (76.200000 m3)
(mkradio3) 061 C!: 2802 current date (2022-04-00T02:00:00Z)
(mkradio3) 063 C!: 1e3b curr consumption (1513.400000 m3)
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-05-01 11:14:02.000;20;19;67270509;0x374468500905276739c3a2109f2bfa0202281e3b000082000070c0414aad9d466618b1aab13eab657a95b1a50a00000070400a003c3013>
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
---
serial) received ascii "T1;1;1;2022-05-01 11:14:35.000;13;16;67270509;0x374468500905276739c3a2109f2bfa0202281e3b000082000070c0414aad9d466618b1aab13eab657a95b1a50a00000070400a003c30134d<0A>>
[... 66 lines cut ...]
(mkradio3) 009   : c3 dll-type (Heat meter)
(mkradio3) 010   : a2 tpl-ci-field (Mfct specific)
(mkradio3) 011 C?: 109F2BFA0202281E3B000082000070C0414AAD9D466618B1AAB13EAB657A95B1A50A00000070400A003C30134D mfct specific
(mkradio3) 057 C!: 2b9f previous date (2021-12-31T02:00:00Z)
(mkradio3) 059 C!: fa02 prev consumption (76.200000 m3)
(mkradio3) 061 C!: 2802 current date (2022-04-00T02:00:00Z)
(mkradio3) 063 C!: 1e3b curr consumption (1513.400000 m3)
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-05-01 11:15:13.000;15;17;67270509;0x374468500905276739c3a2109f2bfa0202281e3b000082000070c0414aad9d466618b1aab13eab657a95b1a50a00000070400a003c3013>
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
`
And the readings on the meter itself:
```
1: 0147,034 GJ
2: 2021,12,31
3:0131,834 GJ
4:5413,772 M3
5:0058732 h
6:46,70 C
7:41,64 C
8:4,96 K
9:296 l/h
10: 1,6 kW
11: 0
```

Boldfor, 2023-03-09:
This issue is still relevant. From what I can see, a fix would be rather trivial, however I lack the knowledge to do myself.

Boldfor, 2023-03-21:
Hmm, pity.

Out of curiosity: In which region do you live in? In my region (Frankfurt am Main, Germany) my sensor sends signals unencrypted, just with the need to do some awkward conversion (hence the issue in the first place).

Boldfor, 2023-07-23:
Still the same: This issue is still relevant. From what I can see, a fix would be rather trivial, however I lack the knowledge to do myself.

Boldfor, 2023-09-23:
A fix would be rather trivial, however I lack the knowledge to do myself. The issue is still relevant.

TheGreyDiamond, 2023-09-25:
This is the exact reason I dislike automated actions on Github issues. It creates unnecessary spam.

Boldfor, 2023-09-26:
> This is the exact reason I dislike automated actions on Github issues. It creates unnecessary spam.
Even though I somehow agree, I believe it does make sense that the person who requests something has to confirm from time to time that it's still relevant. Otherwise there's the risk that a person has a random idea, opens an issue, completely forgets about it or moves on (and doesn't need it anymore) and then someone invests time to fix it for nothing.

sunflowerenergias, 2024-10-21:
For this counter model you can use the Compact5 driver and it seems to export the correct values.

Boldfor, 2024-11-04:
@sunflowerenergias , quick question: Do you have exactly this meter running with the compact5-driver?

I just change the driver but couldn't get it working (but also didn't have enough time and will try again tomorrow).

closed 2024-11-05 by Boldfor

henkiejan1, 2024-11-09:
Did not work for vario 3 type 3.2.1....

`(meter) Myheating: meter detection did not match the selected driver compact5! correct driver is: unknown!
(meter) Not printing this warning again for id: 67270509 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0xc3) ver: 0x39
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
Myheating	67270509	32327 kWh	3523 kWh	28804 kWh	2024-11-09 16:16:55
Myheating	67270509	32327 kWh	3523 kWh	28804 kWh	2024-11-09 16:20:57
`