wmbusmeters/wmbusmeters #334: Can not install Home Assistand wmbusmeters addon (The add-on could not be installed)

pawsli, 2021-09-06:
Hello!
I've tried to install wmbusmmeters HA addon but I can not do this. 
 I received error like this:
 [image]
English translation of polish text on screenshot is: "The add-on could not be installed" ;)

These are supervisor log screenshots:

[image]

[image]

And this is my system tab screenshot:
[image]

weetmuts, 2021-09-07:
Ping @to-masz @andriej

to-masz, 2021-09-07:
@pawsli do you run supervisor on pure debian 10? I will see if that can be related to HA core 2021-9

to-masz, 2021-09-07:
@pawsli from quick research on the Internet - you need to update your docker version on the host machine (running Debian 10) 😉 

@weetmuts @BIBOLV today HA owners build the addon image on their machines. Would it be possible to build it in the project CI? The only complexity is that there is an official builder https://github.com/home-assistant/builder and either we should use it or figure out what it does additionally to normal docker build action

pawsli, 2021-09-07:
> @pawsli do you run supervisor on pure debian 10? I will see if that can be related to HA core 2021-9

My host machine is running OMV 5.3.9-1 (Usul).

Thanks for the tip with the docker update. I'll check it in a few days because now I don't have direct access to the host machine.

andriej, 2021-09-07:
@pawsli just remember that non-Debian non-stable installation are out of support regarding Supervisor, so even if it would end that it's fault of Supervisor somehow, you won't get any support from HA's devs side. It might not matter, but it's worth to remember.

BIBOLV, 2021-09-10:
@to-masz, sure docker images can be built using CI. Preferably in github using actions, so should I create it or will you? I did a quick view of your docker file and currently I do not think that it could be combined in one, so better to build separate one for ha-addon and if it will make sense, then we will look at it later on.

weetmuts, 2021-09-25:
@to-masz When the HA addon user has installed the addon (and it built the source automatically)
will it stick to that built version until the user asks for an upgrade?

to-masz, 2021-09-25:
@weetmuts in general yes. 

When the `version` value will be increased in `ha-addon/config.json` all add-on users will see the update option. Once approved, the HA supervisor will download the new config (and as long as there is no `image` with docker url there - the HA supervisor will build the Dockerfile on its own; if image url is provided - it will just pull the image instead).

Adman, 2023-01-25:
@jsuchal Could you please give a hint on how to build that docker image? I tried running `docker build` in ha-addon folder, but it requires `BUILD_FROM` argument and I'm not sure what to pass there. Thanks!

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

closed 2023-11-04 by BIBOLV