wmbusmeters/wmbusmeters #336: AMB8465: Working great on my Laptop but not on my RPi 3 B+ / Docker

Boldfor, 2021-09-15:
**Setting:**
- **Hardware:** Raspi 3 with Nooelec NESDR Nano 2 TL-SDR-USB-Set (RTL2832U + R820T2)
- **OS:** Raspian OS 10
- **wmbusmeters:** 1.4.1-9-g4c1c677 on Docker/Portainer
- **openHAB** 3.1.0 Release Build on Docker/Portainer

**Situation:**
I'm running wmbusmeters on my Raspberry Pi with my NESDR Nano 2 without any problem (feeding data via mosquitto_pub to openHAB) with the following config file:

```
loglevel=debug
device=rtlwmbus(ppm=17)
listento=c1,t1,s1
logtelegrams=false
format=json
logfile=/wmbusmeters_data/logs/wmbusmeters.log
meterfilesnaming=name
meterfilestimestamp=day
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
ignoreduplicates=true
```

For two reasons I wanted to switch to an AMB8465 USB Stick. First, the Nooelec NESDR gets **freaking** hot (way too hot to touch / I measured above 60°C, so I would be surprised if the stick will last longer than a year or two). And second, the fact that one of the RPi ARM cores is constantly at 100% is not super critical, but also not that nice.

So I got an AMB8465, which works like a charm on my Linux Mint 20.2 laptop during first trials. The fact that it does work shows me that the stick is actually working (not broken or so).

Then I tried to replace my Nooelec NESDR at my Raspberry Pi  with the AMB8465. With the following config (same as above for the Nooelec, except for a change in the device line...

`device=amb8465`

... I just can't get it to work. The log reads the following.

```
(wmbusmeters) version: 1.4.1-9-g4c1c677
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) number of meters: 2
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(2,9,144,234,81,2) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(3,144,234,81,40) modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
```

Am I missing something simple? Not sure if it's a Raspberry-related problem (if so please feel free to close this issue) or related to wmbusmeters? Or maybe an amb8465-specific problem?

Thanks a lot! And thanks for all your efforts on this project!

weetmuts, 2021-09-16:
Can you check using dmesg which serial port the amb8465 is connected to when the dongle is plugged in?
The /dev/ttyAMA0 is weird on some raspberry pis, since send a small byte string to it causes the whole
pi to lock up.

Try: wmbusmeters --donotprobe=/dev/ttyAMA0 auto:t1,c1

Boldfor, 2021-09-16:
> Try: wmbusmeters --donotprobe=/dev/ttyAMA0 auto:t1,c1

Great, that trick convinced it to work. Thanks a lot! :)

Ended up with the following config file (the # lines are the ones for the Nooelec NESDR, for me to remember in case I have to switch back at some point in time)

```
loglevel=normal
device=auto:t1,c1
donotprobe=/dev/ttyAMA0
listento=c1,t1
logtelegrams=false
format=json
logfile=/wmbusmeters_data/logs/wmbusmeters.log
meterfilesnaming=name
meterfilestimestamp=day
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
ignoreduplicates=true
```

Btw: Interesting to see the difference in reception quality between the Nooelec NESDR (stable 90-100 db) and the amb8465 with its internal antenna (~40 db).

For the sake of completeness 1/2:
> Can you check using dmesg which serial port the amb8465 is connected to when the dongle is plugged in?
First line shows me disconnecting the NESDR Nano 2, line 2 - 10 shows me connecting the amb8465:
```
[967350.335983] usb 1-1.3: USB disconnect, device number 8
[967359.339132] usb 1-1.3: new full-speed USB device number 9 using dwc_otg
[967359.498283] usb 1-1.3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[967359.498310] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[967359.498326] usb 1-1.3: Product: AMB8465 AMBER USB Stick
[967359.498340] usb 1-1.3: Manufacturer: FTDI
[967359.498355] usb 1-1.3: SerialNumber: 2A0209EC
[967359.509861] ftdi_sio 1-1.3:1.0: FTDI USB Serial Device converter detected
[967359.510086] usb 1-1.3: Detected FT232RL
[967359.513042] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB0
```

For the sake of completeness 2/2: Below the log of the amb8465 working with the config file above.
```
(wmbusmeters) version: 1.4.1-9-g4c1c677
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) number of meters: 2
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(2,9,48,89,246,118,2) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(3,48,89,246,118,40) modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FF"
(amb8465) not enough data for decoding ConfigAMB8465!
(amb8465) reading response... 2
(serial) received binary "FFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF17"
(amb8465) proprely decoded ConfigAMB8465 response. Content: id=00000000 media=00 version=00 c_field=44 auto_rssi=00
(amb8465) found response at offset 0
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(amb8465) detect id=00000000 media=00 version=00 c_field=44 auto_rssi=00
(amb8465) are you there? yes 00000000
(main) opening c1,t1
Started auto amb8465[00000000] on /dev/ttyUSB0 listening on c1,t1
(amb8465) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(2,9,64,89,246,118,2) modem(DTR|RTS|CTS)
[... 1270 lines cut ...]
(amb8465) received full frame
(amb8465) rssi 30 (-59 dBm)
(wmbus) parseDLL @0 48
(wmbus) parseELL @10 38
(wmbus) parseAFL @10 38
(wmbus) parseTPL @10 38
(meter) kaltwasserzaehler: for me? 74302990 in 74302994
(meter) kaltwasserzaehler: not for me: not my id
(wmbus) parseDLL @0 48
(wmbus) parseELL @10 38
(wmbus) parseAFL @10 38
(wmbus) parseTPL @10 38
(meter) warmwasserzaehler: for me? 74302990 in 67552079
(meter) warmwasserzaehler: not for me: not my id
(meter) no meter handled 74302990 checking 2 templates.
(wmbus) parseDLL @0 48
(wmbus) parseELL @10 38
(wmbus) parseAFL @10 38
(wmbus) parseTPL @10 38
(meter) warmwasserzaehler: for me? 74302990 in 67552079
(meter) warmwasserzaehler: not for me: not my id
(meter) kaltwasserzaehler: for me? 74302990 in 74302994
(meter) kaltwasserzaehler: not for me: not my id
(wmbus) telegram from 74302990 ignored by all configured meters!
```

Boldfor, 2021-09-25:
Closing this issue, since Problem is solved.

closed 2021-09-25 by Boldfor

weetmuts, 2021-09-25:
Great! @Boldfor 

>Btw: Interesting to see the difference in reception quality between the Nooelec NESDR (stable 90-100 db) and the amb8465 with its internal antenna (~40 db).

So dBm is an absolute measurement using an exponential scale where 0 dBm is 1 milli Watt. Negative dBm values means fractions of milli Watt. Unfortunately we cannot calculate the proper dBm for the rtlsdr since those devices usually have an automatic gain/amplification which is unknown.

So currently the dBm value reported for the rtlsdrs are in the range ~100 and this is obviously not a valid
absolute dBm value, since it would be the equivalent of you receiving up to 10 MegaWatt on energy in that antenna.
:-)

But even between two dongles like amb8465 and im871a where I have tried to calculate the proper dBm
from the tables in the documentation, I would treat the values as not-quite comparable.
Use it as a relative measurement for your dongle to improve location/orientation of the dongle/antenna.

https://www.rapidtables.com/convert/power/dBm_to_Watt.html