wmbusmeters/wmbusmeters #337: Doesn't read the data?

djmalin, 2021-09-17:
Hello, I have such a problem, after turning on the device, I only see this and there is no data on wear?

- [ Received telegram from: 00231652
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Water meter (0x07)
                   ver: 0x05
                device: rtlwmbus[00000001]
                  rssi: 59 dBm
                driver: apator162
Received telegram from: 00225094
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Water meter (0x07)
                   ver: 0x05
                device: rtlwmbus[00000001]
                  rssi: 139 dBm
                driver: apator162] 

here is my configuration:

- [Home Assistant OS 6.4
- supervisor-2021.09.0
- core-2021.9.6
- rpi4
- chipset RTL2838 Realteka ]

weetmuts, 2021-09-25:
Can you look inside /config/wmbusmeters/wmbusmeters.log and post it here?

@to-masz

to-masz, 2021-09-25:
@djmalin 
1. I would like to ask you to fix the formatting of your config 😉 
2. I really recommend a tool like MQTT Explorer to debug where the data goes
3. What do you see in the add-on logs?

weetmuts, 2022-12-30:
Closing stale issue.

closed 2022-12-30 by weetmuts