wmbusmeters/wmbusmeters #338: Unknown tpl-ci-field bc  (Diehl HYDRUS 2.0)

jvpap, 2021-09-18:
Got f.e. the two telegrams:
19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A
19442423850734423919BCDCA2610B0013ECB4D17A1781DBA86A3DFFDF92A9

Can't be decoded with: **wmbusmeters --debug 9442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A MyCounter auto 42340785 02083680939E660F3A6BE7D5FBB42AB7**
(wmbusmeters) version: 1.4.1-25-g0788ad7
(config) using device: <19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A>:any 
(config) number of meters: 1
(lookup) with file/hex "19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A"
(lookup) driver: simulation hex
(main) opening <19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A>:any
Started config simulation listening on any
(simulation) in 
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(lookup) with file/hex "19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A"
(lookup) driver: simulation hex
(main) opening <19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A>:any
Started config simulation listening on any
(simulation) in 
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A"
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(wmbus) Unknown tpl-ci-field bc
(meter) MyCounter: for me? 42340785 in 
(meter) MyCounter: yes for me
(meter) MyCounter: meter detection could not find driver for id: 42340785 mfct: (HYD) Hydrometer (0x2324) type: A/D converter (0x19) ver: 0x39
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) created "MyCounter" "unknown" "42340785" encrypted
(meter) used meter template MyCounter  auto to match 42340785
(meter) started meter 1 (MyCounter 42340785 auto)
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(wmbus) Unknown tpl-ci-field bc
(meter) MyCounter: for me? 42340785 in 42340785
(meter) MyCounter: yes for me
(meter) MyCounter auto handling telegram from 42340785
(meter) MyCounter 42340785 "19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A"
(wmbus) parseDLL @0 31
(telegram) DLL L=19 C=44 (from meter SND_NR) M=2324 (HYD) A=42340785 VER=39 TYPE=19 (A/D converter) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(wmbus) Unknown tpl-ci-field bc
(meter) newly created meter (MyCounter 42340785 auto) did not handle telegram!
(wmbus) telegram from 42340785 ignored by all configured meters!
(serial) stopping manager
(simulation) from file "19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A"
(wmbus) skipping already handled telegram.
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted simulation
(wmbus) deleted simulation

Any Ideas what i can try?

weetmuts, 2021-09-20:
That was odd! According to EN 13757 0xbc means set baud rate to 4800 baud.....
Do you have a link to the product page of the meter?

Received telegram from: 42340785
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: A/D converter (0x19)
                   ver: 0x39
                driver: unknown!
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 2423 dll-mfct (HYD)
(wmbus) 04: 85073442 dll-id (42340785)
(wmbus) 08: 39 dll-version
(wmbus) 09: 19 dll-type (A/D converter)
(wmbus) 0a: bc tpl-ci-field (Set baud rate to 4800)

jvpap, 2021-09-20:
That's the page of the device https://www.diehl.com/metering/de/produkte-services/wasserzaehler/wasserzaehler-produkt/hydrus-2-0/462711/

jvpap, 2021-10-05:
Is there anything i can do to get some more infos from the device?

weetmuts, 2021-10-05:
Ok, so it is a water meter for sure? Then it is yet another proprietary hack that on the outside looks like wmbus, but on the inside is something completely different. 

If you can collect a lot of telegrams like this:
wmbusmeters --logtelegram <device>

first 10 telegrams with no water flowing. 
then 10 telegrams with water flowing.
then 10 telegrams again with no water.

and post the telegrams here.

weetmuts, 2022-01-14:
This is solved! It turns out that the telegrams you provided still contain the dll crc:s. These are usually removed by the underlying receiving hardware, like im871a/amb8465/rtlwmbus/rtl433 etc. the notable exception is nanocul. 

I have now added support for detecting the dll-crc:s and removing them when feeding wmbusmeters from the command line or from a simulation file. Try the latest wmbusmeters and this will work:

./build/wmbusmeters --format=json  19442423850734423919BCDCA2110B0013ECD01D864664E0A56B72F2A9B69A Vatten izar ANYID NOKEY
{"media":"water","meter":"izar","name":"Vatten","id":"19394234","prefix":"","serial_number":"000000","total_m3":69.882,"last_month_total_m3":68.479,"last_month_measure_date":"2021-09-01","remaining_battery_life_y":5.5,"current_alarms":"no_alarm","previous_alarms":"no_alarm","transmit_period_s":8,"manufacture_year":"0","timestamp":"2022-01-14T09:25:09Z"}

jvpap, 2022-01-14:
Wow, thats really great. Thank you very much for all your work and effort!

closed 2022-01-14 by jvpap