wmbusmeters/wmbusmeters #339: Apator / HA add on 2x

niklas007, 2021-09-18:


niklas007, 2021-09-18:
How to configure HA Add on so its possible to read 2pc watermeters 1cold and 1 hot ? "apator162"

weetmuts, 2021-09-25:
Can you try the latest version of wmbusmeters on github? It has much improved decoding of the apator telegrams!

niklas007, 2021-09-25:
I must try out when did u upgrade the version ?

weetmuts, 2021-09-27:
I pushed the code to github 5 days ago. Its not tagged as a new version yet though.

weetmuts, 2021-11-06:
@to-masz I forgot to update the ha-addon/config.json for the 1.5.0 release.
I intend to make a 1.5.1 release and essentially oncly change the version in ha-addon/config.json to 1.5.1
(from 0.2.19 if that is ok?)

Do you want to test the 1.5.0 release so that it looks good in docker?

niklas007, 2021-11-07:
Would be great, but I'm not. Running my ha in docker containers ?

Btw.

Would it be possible for u to do a nice HA IU graphic, like we're the water
consumpiotions daily, weekly,monthly would be shown. And it could show to
the last week to. To be compered. If usage is more or lover.
And a input on the hot water. We're i could convert the x kWh to euros from
the hot water boiler smart plug. And the same for water water price for
cold water and more.

Sorry my English , main language is swedish and second Finnish . So I'm
gladly convert the text to ex swedish if something u don't understand

Cheers 🙂

wajo666, 2022-01-01:
Hi, last reading was in 31.12.2021. after new year I got still this error

(rtlwmbus) checkRTLWMBusFrame "C1;0;1;2022-01-01 11:26:55.000;14;15;28718989;0x49449344898971283508780dff5f3500822b0000800007b06effff00000000bf2c74000020bf2c74000000008000800080008000800080008000800000000019007c005e09da3619824b<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "C1;0;1;2022-01-01 11:26:57.000;23;24;28719302;0x49449344029371283508780dff5f350002c300007e0007b06effff030000003e58e60400017e58e604010001000100010001000100010001000100000000000000e8002f046d1b0cc121<0A>C1;0;1;2022-01-01 11:26:57.000;35;33;28719302;0x49449344029371283508f01bfd7cd4020b0c0001f8001ec1bbfcfc0c000002fcb1cc080002fcb1cc0803000200020002000200020002000200020000000000a403a000bc11b46c330486<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;0;1;2022-01-01 11:26:57.000;23;24;28719302;0x49449344029371283508780dff5f350002c300007e0007b06effff030000003e58e60400017e58e604010001000100010001000100010001000100000000000000e8002f046d1b0cc121<0A>C1;0;1;2022-01-01 11:26:57.000;35;33;28719302;0x49449344029371283508f01bfd7cd4020b0c0001f8001ec1bbfcfc0c000002fcb1cc080002fcb1cc0803000200020002000200020002000200020000000000a403a000bc11b46c330486<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(rtlwmbus) checkRTLWMBusFrame "C1;0;1;2022-01-01 11:26:57.000;35;33;28719302;0x49449344029371283508f01bfd7cd4020b0c0001f8001ec1bbfcfc0c000002fcb1cc080002fcb1cc0803000200020002000200020002000200020000000000a403a000bc11b46c330486<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.

weetmuts, 2022-01-09:
@wajo666 The version 1.6 has been pushed to the ha-addon. So this problem should be fixed now.
Please confirm and then we can close this issue. Thanks!

wajo666, 2022-01-09:
Was fixed in other thread

closed 2022-01-13 by weetmuts