wmbusmeters/wmbusmeters #340: Please add Support

wiesel24, 2021-09-19:
Hi all,
sorry to bother you. I've tried to add my water meter into HA with wmbusmeters add-on but he is telling me in the log files that the Water meter has unknown driver?
It's a Diehl Water Meter so I don't why it isn't the izar driver? In the logs it shows me following lines:
(meter) Water: meter detection could not find driver for id: 1800**** mfct: (HYD) Hydrometer (0x2324) type: Water meter (0x07) ver: 0x85
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
Pictures of the meter
[image]
[image]
:
Maybe you can add them to the HA add-on.
Thank you in advance!
Regards 
Wiesel

wiesel24, 2021-09-19:
What I've maybe forgot to paste is the Telegram Message
Received telegram from: ********
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Water meter (0x07)
                   ver: 0x85
                device: rtlwmbus[00000001]
                  rssi: 29 dBm
                driver: unknown!
telegram=||19442423850798160018A2410100133EBBD44081053F243A82A3|+37

weetmuts, 2021-09-20:
Thanks for the report! Your telegram decodes nicely with the izar driver. 
I have added your meters particular mfct/type/version combo to the detection table for izar.

Please test!

wiesel24, 2021-09-20:
Wow! Thank you! You've some contribution Button? Buy me a coffee?
Regards

weetmuts, 2021-09-25:
@wiesel24 Thanks! I might add a sponsor button in future!

closed 2021-09-25 by weetmuts