wmbusmeters/wmbusmeters #341: counter ARROW Wan MVM 169

bend94, 2021-09-20:
Hi,

I would like to know if the wmbusmeters is able to read and decode -> Radio Module wM-Bus 169 Mhz
My counter is a ARROW Wan MVM 169

Best Regards

weetmuts, 2021-09-21:
Please join the discussion here: https://github.com/xaelsouth/rtl-wmbus/issues/26
Be ready to collect signal samples so that rtl_433 and rtl_wmbus can add support for N mode.
:-)

weetmuts, 2021-09-21:
The iM170A chip apparently supports N mode reception (169.40652 MHz to 169.43750 MHz)

I do not know if one can buy such a prepared USB dongle though or if there is some other
purchaseable/buildable board with an im170a chip. I supposed the antenna size needs to 
be different from the one in the im871a USB dongle (868.95MHz)

If you find one, then I can add support to wmbusmeters to configure N mode
relatively easy since wmbusmeters already can detect if you have plugged in an im871a or an im170a chip.

weetmuts, 2021-09-21:
Then after we have received a telegram from the meter, the question is what protocol it uses. 
After reading the spec: https://www.maddalena.it/en/products/communication-modules/arrow-wan-169/83

I cannot conclusively say if the protocol is fully wmbus compliant, so we shall see if it is trivial to decode
(when it is compliant) or anything from messy, tricky or impossible....

But hopefully it is using the wmbus standard in a mostly compliant way.

bend94, 2021-09-21:
> Please join the discussion here: xaelsouth/rtl-wmbus#26

Hi
i won't be available until next January, my house is under construction
we can keep in touch then

Best regards

merbanan, 2022-02-06:
@bend94 are you near your meter now? You need a rtl-sdr to receive this signal.

bend94, 2022-02-07:
Hi
I will be in my new house around April.
I will be able to catch some logs about my water counter in April for you
with pleasure.
Let s keep in touch, I am very interested to decode my water consumption
Best regards

bend94, 2022-03-31:
Hi

My home will be ready at the end of April
Are you still interested in capturing  433Mhz frames ?
Let me know

thanks

closed 2023-02-19 by github-actions[bot]

DarKOrange75, 2025-05-10:
Hi all, I also have this meter, anyone made progress on reading telegrams ?