wmbusmeters/wmbusmeters #342: Please add Support Gas meter Apator AT-WMBUS-G-01

ajakubek80, 2021-09-23:
Hello
I have a Gas meter with Apator AT-WMBUS-G-01
Meter reading 916,998 m3

I have manage to receive that kind of message:
```
wmbusmeters --debug rtl433:868.95M 

Received telegram from: 00043094
          manufacturer: (AMX) APATOR METRIX, Poland (0x5b8)
                  type: Gas meter (0x03)
                   ver: 0x01
                device: rtl433[00000001]
                  rssi: 999 dBm
                driver: unknown!
(wmbus) 00: 60 length (96 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: b805 dll-mfct (AMX)
(wmbus) 04: 94300400 dll-id (00043094)
(wmbus) 08: 01 dll-version
(wmbus) 09: 03 dll-type (Gas meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: 1d tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field (OK)
(wmbus) 0d: 5085 tpl-cfg 8550 (bidirectional AES_CBC_IV nb=5 cntn=0 ra=0 hc=0 )
telegram=||6044B8059430040001037A1D005085E2B670BCF1A5C87E0C1A51DA18924EF984613DA2A9CD39D8F4C7208326C76D42DBEADF80D574192B71BD7C4F56A7F1513151768A9DB804883B28CB085CA2D0F7438C361CB9E2734712ED9BFBB2A14EF55208|+97

rtl433 -f 868.95M

time      : 2021-09-09 23:55:22
model     : Wireless-MBus Mode      : T            Manufacturer: AMX         ID        : 43094
Version   : 1            Device Type: 0x03         Device Type String: Gas   Control   : 0x44          Data Length: 95           Data      : 5c44b8059430040001037a050050a51e25f5b27e04f2fca996fb9e13824e845ae8c99030bc7acfdc954a09789c7d0d1d18b4847d9fc2194a0484e51957c7ec182ada7604b994b4f2230593ab1b4653782d9ad874f93cb6e647825e885e30b6b5be
Integrity : CRC
Control Info: 0x7A       Access number: 0x05       Device Type: 0x00         Configuration Word: 0xA550 Payload Encrypted: 1
_ _

time      : 2021-09-15 13:30:43
model     : Wireless-MBus Mode      : T            Manufacturer: AMX         ID        : 43094
Version   : 1            Device Type: 0x03         Device Type String: Gas   Control   : 0x44          Data Length: 95           Data      : 5c44b8059430040001037a1a005085078c082ec80f4e66cc7dcd1c160d20e40101f07eb14de87e4aa422c57ac371a3b928c7c1703ad73e8ded3c4de34c26955739a93aee81a02096db1f7d387cc858f11f536dcdb94bee758bea7c472341690db0
Integrity : CRC
Control Info: 0x7A       Access number: 0x1A       Device Type: 0x00         Configuration Word: 0x8550 Payload Encrypted: 1
-----------

time      : 2021-09-15 13:45:59
model     : Wireless-MBus Mode      : T            Manufacturer: AMX         ID        : 43094
Version   : 1            Device Type: 0x03         Device Type String: Gas   Control   : 0x44          Data Length: 95           Data      : 5c44b8059430040001037a1b00508504776585883797e7d64b5f71faadb96e109f887d4a934f79bdb681197abfa74f4cfc57012237ee886a4b1ee47ff2fbd97ba671c2b80472e3c2d9cfd8fd470da03d65f3acd562ed0328785a413c260c191319
Integrity : CRC
Control Info: 0x7A       Access number: 0x1B       Device Type: 0x00         Configuration Word: 0x8550 Payload Encrypted: 1
_ 
```
Which driver can I use ?
Best regards!

weetmuts, 2021-09-23:
You can see from AES_CBC_IV in the log that he meter telegram is encrypted. Do you have the decryption key?

weetmuts, 2021-09-23:
If you have the decryption key build the current github master (just pushed a first version of the driver) and run:
wmbusmeters --logtelegrams rtl433:868.95M MyMeter unismart 00043094 <YOURKEY>
and post a telegram here!

Thanks!

ajakubek80, 2021-09-23:
Thank you for the quick reply.
I used a key from Apator water meters 
The meter reading is correct 923,393m3
```
wmbusmeters --logtelegrams rtl433:868.95M MyMeter unismart 00043094 00000000000000000000000000000000
Started config rtl433[00000001] listening on none using fq 868.95M
Warning! Desired link modes none cannot be set for device 00000001:rtl433[00000001]
(rtl433) dll_len adjusted to 21 from 17. Fix rtl_433? This warning will not be printed again.
telegram=|6044B8059430040001037A200050852F2F|0C7896221603046D082EB7290C943A40230900446D0026A1294C943A3211090001FD670002FD74E80C0DFD0C0655474734202001FD0B02017F14066D0F080EB729B42F2F2F2F2F2F2F2F2F2F2F2F51C8|+36
MyMeter 00043094        923.4 m3        2021-09-23 15:00.02
telegram=|6044B8059430040001037A200050852F2F|0C7896221603046D092EB7290C943A40230900446D0026A1294C943A3211090001FD670002FD74E80C0DFD0C0655474734202001FD0B02017F14066D0F090EB729A32F2F2F2F2F2F2F2F2F2F2F2F5554|+96
MyMeter 00043094        923.4 m3        2021-09-23 15:01.02
telegram=|6044B8059430040001037A200050852F2F|0C7896221603046D0A2EB7290C943A40230900446D0026A1294C943A3211090001FD670002FD74E80C0DFD0C0655474734202001FD0B02017F14066D0F0A0EB729682F2F2F2F2F2F2F2F2F2F2F2F6C50|+156
MyMeter 00043094        923.4 m3        2021-09-23 15:02.02
telegram=|6044B8059430040001037A200050852F2F|0C7896221603046D0B2EB7290C943A40230900446D0026A1294C943A3211090001FD670002FD74E80C0DFD0C0655474734202001FD0B02017F14066D0F0B0EB729E72F2F2F2F2F2F2F2F2F2F2F2FDAF8|+216
MyMeter 00043094        923.4 m3        2021-09-23 15:03.02
telegram=|6044B8059430040001037A210050A52F2F|0C7896221603046D0C2EB7290C943A40230900446D0026A1294C943A3211090001FD670002FD74E80C0DFD0C0655474734202001FD0B02017F14066D1D0C0EB729AC2F2F2F2F2F2F2F2F2F2F2F2F1A92|+290
MyMeter 00043094        923.4 m3        2021-09-23 15:04.16
telegram=|6044B8059430040001037A210050852F2F|0C7896221603046D0D2EB7290C943A40230900446D0026A1294C943A3211090001FD670002FD74E80C0DFD0C0655474734202001FD0B02017F14066D1D0D0EB729562F2F2F2F2F2F2F2F2F2F2F2FCEDF|+350
MyMeter 00043094        923.4 m3        2021-09-23 15:05.16
telegram=|6044B8059430040001037A210050852F2F|0C7896221603046D0E2EB7290C943A40230900446D0026A1294C943A3211090001FD670002FD74E80C0DFD0C0655474734202001FD0B02017F14066D1D0E0EB729D62F2F2F2F2F2F2F2F2F2F2F2FC06A|+410
MyMeter 00043094        923.4 m3        2021-09-23 15:06.16

weetmuts, 2021-09-25:
I have added code to decode most of the telegram. Some fields are reported but we do not know what they mean.
This the json output for the last of your telegrams above.

{"media":"gas","meter":"unismart","name":"MyMeter","id":"00043094","fabrication_no":"3162296","total_date_time":"2021-09-23 14:14","total_m3":923.4,"target_date_time":"2021-09-01 06:00","target_m3":911.32,"version":"4GGU","device_date_time":"2021-09-23 14:14","suppler_info":"00","status":"E80C","parameter_set":"02","status2":"14","timestamp":"2021-09-25T10:05:04Z"}

ajakubek80, 2021-09-25:
It looks like the decoding is ok. 
These are mostly the serial numbers of the meter.
Thank you

[image]

pswiatki, 2021-09-25:
Very interesting. This is my gas-meter (here in Polska):
[image]
Funny how I cannot really find information on this AT-WMBUS-G-01 add-on device. I know for a fact (and verified on my meter) there is a magnet on the 10-liter roller which produces 1 pulse per rotation on a reed switch when put sufficiently close to the counter. So, my guess is the interface's starting value must be set to match the physical reading on the meter upon initial installation.

ajakubek80, 2021-09-25:
> Very interesting. This is my gas-meter (here in Polska):
> 
> [image]
> 
> Funny how I cannot really find information on this AT-WMBUS-G-01 add-on device. I guess it is becasue this version must be certified for use near gas installation, whereas other version (which can easily be find on the internet) don't (e.g. for watermeters).
> I know for a fact (and verified that on my meter) there is a magnet on the 10-liter roller which produces 1 pulse per rotation on a reed switch when put sufficiently close to the counter. So, my guess is the interface's starting value must be set to match the physical reading on the meter upon initial installation.

Też nie znalazłem nic na temat nakładki AT-WMBUS-G-01 w internecie, ani na stronie Apatora. 
Taki zestaw został mi zamontowany w całości przez lokalną firmę gazową.

pswiatki, 2021-09-25:
Would you be so kind and translate your response to English. Perhaps others might be interested as well  ;)

weetmuts, 2021-09-26:
Thanks for the pictures! I fixed so that the version string matches what is printed on the meter, ie UGG4.
Then one of the unknowns is a byte hex 0x14 = 20 decimal. Which perhaps is the year 2020 (2000+20)
or it is the 20 before 20MUGG4. Or perhaps they are the same. This is reported as other_int instead of status2.

{"media":"gas","meter":"unismart","name":"Gasen","id":"00043094","fabrication_no":"3162296","total_date_time":"2021-09-15 13:18","total_m3":917,"target_date_time":"2021-09-01 06:00","target_m3":911.32,"version":"UGG4","device_date_time":"2021-09-15 13:18","suppler_info":"00","status":"F00C","parameter_set":"02","other_int":20,"timestamp":"2021-09-26T08:18:53Z"}

ajakubek80, 2021-09-26:
The most important are the total m3 
Thank you for your time. 
Have a great day.

closed 2021-11-28 by weetmuts

GieBek, 2022-10-27:
Hi. @weetmuts 
I have a problem with adding a meter.
When I try to add a meter in config I dont have decrytion of telegram from my Gas Meter.
My Config:
`MyMeter unismart 00021290 00000000000000000000000000000000`

My log: 
Received telegram from: 00021290
          manufacturer: (AMX) APATOR METRIX, Poland (0x5b8)
                  type: Gas meter (0x03) encrypted
                   ver: 0x01
                device: rtl433[00000001]
                  rssi: 999 dBm
                driver: unismart
(wmbus) 000   : 60 length (96 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : b805 dll-mfct (AMX)
(wmbus) 004   : 90120200 dll-id (00021290)
(wmbus) 008   : 01 dll-version
(wmbus) 009   : 03 dll-type (Gas meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : 7f tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 5085 tpl-cfg 8550 (bidirectional AES_CBC_IV nb=5 cntn=0 ra=0 hc=0 )
(wmbus) 015 CE: FB871F78F05132D9F830FA436C4F12DB63194A7E4AAA7D399D205D1ABDE323F32D8D6EB7FAE866B9BF0F70D6A62C3549A52398372C3F6DA9535C871E64931EFEA108524FCD64F936D297A5966EEED691 encrypted
(wmbus) telegram from 00021290 ignored by all configured meters!

Please can You help me ? What I made wrong??

weetmuts, 2022-10-27:
It seems like wmbusmeters does not read your config. The log shows the expected output when no meter is configured.
Can you provide info on how it is started?